esphome/issues #4154: Dashboard after some time shows every device offline.

Kosygor, 2023-02-11:
### The problem

Hello I am running ESPHome in HA addon. Ha is in VM on TrueNAS Scale.
By the fact that `manual_ip` block for some reason does not have option to give host, a name (unless opnsense does not resolve it right) I started to investigating whole mDNS thing. So I have checked router and everything and everything seems to work. 
I am running mDNS for getting status.
For ESP32 and ESP8266 devices I can:
- ping them by IP and `[device].local `from windows cmd. 
- flash them/see logs ( for ESP8266 devices this is random and can also break after while);
But all of them are shown as offline in dashboard.
Until I restart addon. Then I can see them* online for a while for about 15-30 seconds. 

*These that are not powered are of course still offline.  Similar thing happens on my DMP-L1 ( ESP8265) but this one even after restart does not show as online - needs further investigation - bu i can still .

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant OS 9.5 /  Home Assistant Core: 2023.2.3

### What platform are you using?

ESP8266

### Board

ESP-WROOM32 / ESP2866 

### Example YAML snippet

```yaml
esphome:
  name: atrea

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "k2ASdx6utooNM79TGz8TS/B+No6AngqgFQawfMYPv24="

ota:
  password: "2d11b142598625087ddb971a67d3e1b6"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Atrea Fallback Hotspot"
    password: "siOfztfGtRB4"

captive_portal:

web_server:
  port: 80
  version: 2
  include_internal: true
  ota: false

time:
  - platform: sntp

output:
  - platform: esp32_dac
    pin: GPIO25
    id: dac_output
fan:
  - platform: speed
    output: dac_output
    name: "Atrea"
```

### Anything in the logs that might be useful for us?

```txt
-----------------------------------------------------------
 Add-on: ESPHome
 ESPHome add-on for intelligently managing all your ESP8266/ESP32 devices
-----------------------------------------------------------
 Add-on version: 2022.12.8
 You are running the latest version of this add-on.
 System: Home Assistant OS 9.5  (amd64 / qemux86-64)
 Home Assistant Core: 2023.2.3
 Home Assistant Supervisor: 2023.01.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[20:02:56] INFO: Configuring timezone (Europe/Warsaw)...
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service esphome: starting
s6-rc: info: service esphome successfully started
s6-rc: info: service init-nginx successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service discovery: starting
[20:02:56] INFO: Waiting for ESPHome dashboard to come up...
[20:02:56] INFO: Starting ESPHome dashboard...
2023-02-11 20:02:57,145 INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
[20:02:57] INFO: Starting NGINX...
[20:02:57] INFO: Successfully send discovery information to Home Assistant.
s6-rc: info: service discovery successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
2023-02-11 20:03:00,324 INFO 200 GET /devices (0.0.0.0) 1.80ms
2023-02-11 20:03:05,340 INFO 304 GET /devices (0.0.0.0) 6.76ms
2023-02-11 20:03:10,336 INFO 304 GET /devices (0.0.0.0) 1.73ms
2023-02-11 20:03:15,366 INFO 304 GET /devices (0.0.0.0) 11.24ms
2023-02-11 20:03:20,369 INFO 304 GET /devices (0.0.0.0) 5.84ms
2023-02-11 20:03:25,371 INFO 304 GET /devices (0.0.0.0) 1.67ms
2023-02-11 20:03:30,395 INFO 304 GET /devices (0.0.0.0) 6.70ms # and about this time all of devices are going offline
```

TBail, 2023-03-09:
If have more or less a similar setup. The network is Unifi and the base system for the HA VM is proxmox.

And unfortunately the same problem.

HA Version is the latest

TBail, 2023-03-10:
I fixed it for me. Problem was that i enabled som Multicast enhancement for my iot-WLAN. After diabling the enhancements it worked like a charm. With mDNS oder with ping

bdraco, 2023-11-13:
We have re-written the zeroconf implementation in 2023.11.x. It should be much more stable https://github.com/esphome/esphome/pull/5681

However if you have Multicast enhancement enabled, this may still prevent it from working as expected.

closed 2023-11-13 by bdraco