esphome/issues #4161: esp reboots if a text_sensor component get's a null/empty message

pfichtner, 2023-02-13:
### The problem

I have a text_sensor using mqtt_subscribe platform (see below). 

If there is a message published without a payload (e.g. mosquitto_pub with parameter  -m '') the ESP crashes and reboots. the log statement cannot be retrieved so I guess it even wasn't executed. 
Don't know if this is `mqtt_subscribe` specific or if it's a general `text_sensor` problem. 

### Which version of ESPHome has the issue?

v2022.12.8

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

-

### What platform are you using?

ESP32-IDF

### Board

Shelly-Plus-1PM

### Component causing the issue

text_sensor

### Example YAML snippet

```yaml
text_sensor:
  - platform: mqtt_subscribe
    name: some_name
    on_value:
      - then:
        - logger.log: "Logger Info"
```

ssieb, 2023-02-13:
Can you get the serial logs to get the exception message?

ssieb, 2023-02-13:
Please do.  The different type shouldn't matter in this case.

pfichtner, 2023-02-14:
"Unfortunately" the code does not crash a d1_mini (log statement gets executed and no restart of the device), so perhaps the different type matters still
```
esphome:
  platform: ESP8266
  board: d1_mini
```

So I hope I soon will get another Shelly-Plus-1PM I can use for serial debug or there is anyone out there able to do so.

ssieb, 2025-07-11:
Unlikely.  That was for long messages.