esphome/issues #4162: Dallas / DS18b20 not working

dossidr, 2023-02-13:
### The problem

Hi Experts 

all checks and Options available by ESP Home wont getting any DS18B20 Sensor connected on GPIO23 or other.
An explizit declaration w/ address also wont woking as well...
My system is esp-idf : latest and ESPHome v2022.12.8

i hope fix comming up next days....
best regards 

### Which version of ESPHome has the issue?

2033.12.8

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

mini D1

randybb, 2023-02-13:
Lets test it again with Arduino and ESP32-IDF - with one of my dev devices and a different set of sensors. As expected, no issues.
```yaml
dallas:
  - pin: 23
 
sensor:
  - platform: dallas
    address: 0x8b0516b39804ff28
    name: "${name} Temp 1"
  - platform: dallas
    address: 0xa50516b3d32aff28
    name: "${name} Temp 2"
  - platform: dallas
    address: 0x8f0516b39d87ff28
    name: "${name} Temp 3"
  - platform: dallas
    address: 0x810516b3c69fff28
    name: "${name} Temp 4"
```
```
[23:31:06][D][dallas.sensor:082]:   Found sensors:
[23:31:06][D][dallas.sensor:084]:     0x8b0516b39804ff28
[23:31:06][D][dallas.sensor:084]:     0xa50516b3d32aff28
[23:31:06][D][dallas.sensor:084]:     0x8f0516b39d87ff28
[23:31:06][D][dallas.sensor:084]:     0x810516b3c69fff28
...
[23:31:39][D][dallas.sensor:143]: 'DEV 1 Temp 1': Got Temperature=22.8°C
[23:31:39][D][sensor:127]: 'DEV 1 Temp 1': Sending state 22.75000 °C with 1 decimals of accuracy
[23:31:39][D][dallas.sensor:143]: 'DEV 1 Temp 2': Got Temperature=23.0°C
[23:31:39][D][sensor:127]: 'DEV 1 Temp 2': Sending state 23.00000 °C with 1 decimals of accuracy
[23:31:39][D][dallas.sensor:143]: 'DEV 1 Temp 3': Got Temperature=22.8°C
[23:31:39][D][sensor:127]: 'DEV 1 Temp 3': Sending state 22.81250 °C with 1 decimals of accuracy
[23:31:39][D][dallas.sensor:143]: 'DEV 1 Temp 4': Got Temperature=23.0°C
[23:31:39][D][sensor:127]: 'DEV 1 Temp 4': Sending state 23.00000 °C with 1 decimals of accuracy
```

Do you have any special setting, like lowered MCU clock speed?

If yes, then it is related to #3909.

dossidr, 2023-02-14:
Hi Roman...

not possible....
see Logfile attached....
Arduino Env:
[DALLAS_ERROR_output_2023-02-14_11-47-18.log](https://github.com/esphome/issues/files/10731600/DALLAS_ERROR_output_2023-02-14_11-47-18.log)

ESP-IDF Env:
[DALLAS_ERROR_1_output_2023-02-14_11-47-18.txt](https://github.com/esphome/issues/files/10731699/DALLAS_ERROR_1_output_2023-02-14_11-47-18.txt)
 my *.ymal
[yaml.txt.txt](https://github.com/esphome/issues/files/10731729/yaml.txt.txt)

randybb, 2023-02-14:
esphome uses inverted id's - note the 28 on the end. Nevertheless it should detect these sensors.
Mine are connected to 3,3 V, with 4k7 resistor between VDD and DQ, removed that resistor (internal mcu pullup can handle that) or disconnected VDD (= Parasite-Powered mode) and it still works.

dossidr, 2023-02-14:
Hi Roman,

all attempts so far have been unsuccessful... I don't know why... I've tried everything... There are 4 sensors with a 4.7 kOhm resistor. On the Arduino, these are recognized immediately on the ESP32, absolutely no chance...
Thank you for your tip I overlooked the addresses.
I also defined the PIN as an extra input, but I still get errors...
[https://www.az-delivery.de/products/esp32-d1-mini
[image]
[D1_Mini_ESP32_-_pinout.pdf](https://github.com/esphome/issues/files/10735218/D1_Mini_ESP32_-_pinout.pdf)
](url)
`Using '/dev/ttyUSB0' as serial port.
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Connecting..........

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WD-V3 (revision 3)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 08:B6:1F:38:34:48
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 11.1s
Flash will be erased from 0x00001000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00048fff...
Compressed 15872 bytes to 10319...
Writing at 0x00001000... (100 %)Wrote 15872 bytes (10319 compressed) at 0x00001000 in 0.4 seconds (effective 327.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Writing at 0x00008000... (100 %)Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 614.7 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 905.5 kbit/s)...
Hash of data verified.
Compressed 232832 bytes to 126841...
Writing at 0x00044a5a... (100 %)Wrote 232832 bytes (126841 compressed) at 0x00010000 in 3.0 seconds (effective 612.3 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[18:28:32]ets Jul 29 2019 12:21:46
[18:28:32]
[18:28:32]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[18:28:32]configsip: 0, SPIWP:0xee
[18:28:32]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[18:28:32]mode:DIO, clock div:2
[18:28:32]load:0x3fff0018,len:4
[18:28:32]load:0x3fff001c,len:1044
[18:28:32]load:0x40078000,len:8896
[18:28:32]load:0x40080400,len:5828
[18:28:32]entry 0x400806ac
[18:28:32]I (123) cpu_start: Pro cpu up.
[... 68 lines cut ...]
[18:28:32][C][dallas.sensor:089]:     Device Class: 'temperature'
[18:28:32][C][dallas.sensor:089]:     State Class: 'measurement'
[18:28:32][C][dallas.sensor:089]:     Unit of Measurement: '°C'
[18:28:32][C][dallas.sensor:089]:     Accuracy Decimals: 1
[18:28:32][V][dallas.sensor:089]:     Unique ID: 'dallas-8d0316a35989ff28'
[18:28:32][C][dallas.sensor:097]:     Address: 0x8d0316a35989ff28
[18:28:32][C][dallas.sensor:098]:     Resolution: 12
[18:28:32][C][dallas.sensor:089]:   Device 'Temperature #4'
[18:28:32][C][dallas.sensor:089]:     Device Class: 'temperature'
[18:28:32][C][dallas.sensor:089]:     State Class: 'measurement'
[18:28:32][C][dallas.sensor:089]:     Unit of Measurement: '°C'
[18:28:32][C][dallas.sensor:089]:     Accuracy Decimals: 1
[18:28:32][V][dallas.sensor:089]:     Unique ID: 'dallas-310316a379b3ff28'
[18:28:32][C][dallas.sensor:097]:     Address: 0x310316a379b3ff28
[18:28:32][C][dallas.sensor:098]:     Resolution: 12
[18:29:18][E][dallas.sensor:112]: Requesting conversion failed
[18:29:18][V][sensor:076]: 'Temperature #1': Received new state nan
[18:29:18][D][sensor:127]: 'Temperature #1': Sending state nan °C with 1 decimals of accuracy
[18:29:18][V][sensor:076]: 'Temperature #2': Received new state nan
[18:29:18][D][sensor:127]: 'Temperature #2': Sending state nan °C with 1 decimals of accuracy
[18:29:18][V][sensor:076]: 'Temperature #3': Received new state nan
[18:29:18][D][sensor:127]: 'Temperature #3': Sending state nan °C with 1 decimals of accuracy
[18:29:18][V][sensor:076]: 'Temperature #4': Received new state nan
[18:29:18][D][sensor:127]: 'Temperature #4': Sending state nan °C with 1 decimals of accuracy
[18:30:18][E][dallas.sensor:112]: Requesting conversion failed`

dossidr, 2023-02-14:
okay one Step more positiv .... but one of the sensors deliverd an "scratch ... failure" 
Whats happend? Im getting the Board itself to an external 5V power supply... and after them working...
`Using '/dev/ttyUSB0' as serial port.
Connecting.......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.......
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WD-V3 (revision 3)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 08:B6:1F:38:34:48
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 11.1s
Flash will be erased from 0x00001000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00048fff...
Compressed 15872 bytes to 10319...
Writing at 0x00001000... (100 %)Wrote 15872 bytes (10319 compressed) at 0x00001000 in 0.4 seconds (effective 324.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Writing at 0x00008000... (100 %)Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 613.5 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 908.8 kbit/s)...
Hash of data verified.
Compressed 232832 bytes to 126841...
Writing at 0x00044a5a... (100 %)Wrote 232832 bytes (126841 compressed) at 0x00010000 in 3.0 seconds (effective 612.1 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[18:40:45]ets Jul 29 2019 12:21:46
[18:40:45]
[18:40:45]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[18:40:45]configsip: 0, SPIWP:0xee
[18:40:45]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[18:40:45]mode:DIO, clock div:2
[18:40:45]load:0x3fff0018,len:4
[18:40:45]load:0x3fff001c,len:1044
[18:40:45]load:0x40078000,len:8896
[18:40:45]load:0x40080400,len:5828
[18:40:45]entry 0x400806ac
[18:40:45]I (123) cpu_start: Pro cpu up.
[18:40:45]I (123) cpu_start: Starting app cpu, entry point is 0x40081adc
[18:40:45]I (0) cpu_start: App cpu up.
[18:40:45]I (137) cpu_start: Pro cpu start user code
[18:40:45]I (137) cpu_start: cpu freq: 160000000
[18:40:45]I (137) cpu_start: Application information:
[18:40:45]I (142) cpu_start: Project name:     dallas
[18:40:45]I (146) cpu_start: App version:      2022.11.5
[... 77 lines cut ...]
[18:40:46][D][sensor:127]: 'Temperature #4': Sending state nan °C with 1 decimals of accuracy
[18:41:45][D][dallas.sensor:143]: 'Temperature #1': Got Temperature=21.8°C
[18:41:45][V][sensor:076]: 'Temperature #1': Received new state 21.750000
[18:41:45][D][sensor:127]: 'Temperature #1': Sending state 21.75000 °C with 1 decimals of accuracy
[18:41:45][D][dallas.sensor:143]: 'Temperature #2': Got Temperature=21.3°C
[18:41:45][V][sensor:076]: 'Temperature #2': Received new state 21.312500
[18:41:45][D][sensor:127]: 'Temperature #2': Sending state 21.31250 °C with 1 decimals of accuracy
[18:41:45][D][dallas.sensor:143]: 'Temperature #3': Got Temperature=21.4°C
[18:41:45][V][sensor:076]: 'Temperature #3': Received new state 21.437500
[18:41:45][D][sensor:127]: 'Temperature #3': Sending state 21.43750 °C with 1 decimals of accuracy
[18:41:45][W][dallas.sensor:261]: 'Temperature #4' - Scratch pad checksum invalid!
[18:41:45][V][sensor:076]: 'Temperature #4': Received new state nan
[18:41:45][D][sensor:127]: 'Temperature #4': Sending state nan °C with 1 decimals of accuracy
[18:42:45][D][dallas.sensor:143]: 'Temperature #1': Got Temperature=21.8°C
[18:42:45][V][sensor:076]: 'Temperature #1': Received new state 21.750000
[18:42:45][D][sensor:127]: 'Temperature #1': Sending state 21.75000 °C with 1 decimals of accuracy
[18:42:45][D][dallas.sensor:143]: 'Temperature #2': Got Temperature=21.3°C
[18:42:45][V][sensor:076]: 'Temperature #2': Received new state 21.312500
[18:42:45][D][sensor:127]: 'Temperature #2': Sending state 21.31250 °C with 1 decimals of accuracy
[18:42:45][D][dallas.sensor:143]: 'Temperature #3': Got Temperature=21.4°C
[18:42:45][V][sensor:076]: 'Temperature #3': Received new state 21.437500
[18:42:45][D][sensor:127]: 'Temperature #3': Sending state 21.43750 °C with 1 decimals of accuracy
[18:42:45][W][dallas.sensor:261]: 'Temperature #4' - Scratch pad checksum invalid!
[18:42:45][V][sensor:076]: 'Temperature #4': Received new state nan
[18:42:45][D][sensor:127]: 'Temperature #4': Sending state nan °C with 1 decimals of accuracy`

dossidr, 2023-02-14:
So now the code works.. There is still a scratch error from time to time, but only very rarely... It is extremely important to supply the board and also the sensors with external 5 V... Because the power is only available via USB too low for the OneWire level... And then no connection can be established....

Thanks a lot for your support....

best regards denny

closed 2023-02-14 by dossidr