esphome/issues #4163: problem to send sms using ESP32+SIM800L

pilot1981, 2023-02-14:
### The problem

When I try to send sms using ESP32+SIM800L, sms don't arrive to recipient and I found an error in HA logs:

2023-02-13 21:52:01.709 WARNING (MainThread) [aioesphomeapi.connection] combinatoregsm @ 192.168.6.99: Connection error occurred: 
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 537, in send_message_await_response_complex
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 536, in send_message_await_response_complex
    async with async_timeout.timeout(timeout):
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 320, in _keep_alive_loop
    await self._ping()
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 645, in _ping
    await self.send_message_await_response(PingRequest(), PingResponse)
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 554, in send_message_await_response
    res = await self.send_message_await_response_complex(
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 539, in send_message_await_response_complex
    raise TimeoutAPIError(
aioesphomeapi.core.TimeoutAPIError: Timeout waiting for response for <class 'api_pb2.PingRequest'> after 10.0s

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.4

### What platform are you using?

ESP32

### Board

ESP32+SIM800L on the same board

### Component causing the issue

NA

### Example YAML snippet

```yaml
esphome:
  name: combinatoregsm

ota:
  safe_mode: true
  
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: verbose
  baud_rate: 0 # disable uart logger on esp 8266

# Enable Home Assistant API
api:
  services:
    - service: send_sms
      variables:
        recipient: string
        message: string
      then:
        - sim800l.send_sms:
            recipient: !lambda 'return recipient;'
            message: !lambda 'return message;'
    - service: dial
      variables:
        recipient: string
      then:
        - sim800l.dial:
            recipient: !lambda 'return recipient;'
    - service: connect
      then:
        - sim800l.connect
    - service: disconnect
      then:
        - sim800l.disconnect
    - service: send_ussd
      variables:
        ussdCode: string
      then:
        - sim800l.send_ussd:
            ussd: !lambda 'return ussdCode;'

text_sensor:
  - platform: template
    id: sms_sender
    name: "Sms Sender"
  - platform: template
    id: sms_message
    name: "Sms Message"
  - platform: template
    id: caller_id_text_sensor
    name: "Caller ID"
  - platform: template
    id: ussd_message
    name: "Ussd Code"

uart:
  baud_rate: 9600
  tx_pin: TX
  rx_pin: RX

sim800l:
  on_sms_received:
    - lambda: |-
        id(sms_sender).publish_state(sender);
        id(sms_message).publish_state(message);
  on_incoming_call:
    - lambda: |-
        id(caller_id_text_sensor).publish_state(caller_id);
  on_call_connected:
    - logger.log:
        format: Call connected
  on_call_disconnected:
    - logger.log:
        format: Call disconnected
  on_ussd_received:
    - lambda: |-
        id(ussd_message).publish_state(ussd);
```

### Anything in the logs that might be useful for us?

```txt
to try to send sms I call this service in this way:

service: ESPHome: combinatoregsm_send_sms
recipient: +39xxxxxxxxxxxxx
message: 'Hello word'
```

### Additional information

new board, never used

ssieb, 2023-02-14:
You should be using different pins for the `sim800l` uart connection.  If the device is rebooting, you need to get serial logs to find out why.

pilot1981, 2023-02-14:
> You should be using different pins for the `sim800l` uart connection. If the device is rebooting, you need to get serial logs to find out why.

I can attach device via USB to PC e download logs? I tried but I don't see other useful info...I have to change yaml configuration?

ssieb, 2023-02-14:
You have to change the sim800l uart to use different pins.  And remove the baud rate 0 option from the logger.  Then you can use something like "putty" to read the serial logs over the USB port.

pilot1981, 2023-02-14:
I buyed it here: https://www.amazon.it/dp/B081JPP3TD?psc=1&ref=ppx_yo2ov_dt_b_product_details

it see to refer it: https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800

could be useful find datasheet to understand pinout of SIM800L?

ssieb, 2023-02-14:
Do you actually have the sim800 connected to the ESP?

pilot1981, 2023-02-14:
> Do you actually have the sim800 connected to the ESP?

yes, ESP32 and SIM800L are on the same board

ssieb, 2023-02-14:
Ok, I see now.  You have the wrong pins.  The tx pin is 26 and the rx pin is 27.  Change the uart setting to that and remove the baud rate 0 from the logger.
https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/blob/master/schematic/LilyGo-SIM800L-IP5306-20190610.pdf

pilot1981, 2023-02-14:
ok, I changed yaml to this one:
``
esphome:
  name: combinatoregsm

ota:
  safe_mode: true
  
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: verbose
#  baud_rate: 0 # disable uart logger on esp 8266

# Enable Home Assistant API
api:
  services:
    - service: send_sms
      variables:
        recipient: string
        message: string
      then:
        - sim800l.send_sms:
            recipient: !lambda 'return recipient;'
            message: !lambda 'return message;'
    - service: dial
      variables:
        recipient: string
      then:
        - sim800l.dial:
            recipient: !lambda 'return recipient;'
    - service: connect
      then:
        - sim800l.connect
    - service: disconnect
      then:
        - sim800l.disconnect
    - service: send_ussd
      variables:
        ussdCode: string
      then:
        - sim800l.send_ussd:
            ussd: !lambda 'return ussdCode;'

text_sensor:
  - platform: template
    id: sms_sender
    name: "Sms Sender"
  - platform: template
    id: sms_message
    name: "Sms Message"
  - platform: template
    id: caller_id_text_sensor
    name: "Caller ID"
  - platform: template
    id: ussd_message
    name: "Ussd Code"

uart:
  baud_rate: 9600
  tx_pin: 26
  rx_pin: 27

sim800l:
  on_sms_received:
    - lambda: |-
        id(sms_sender).publish_state(sender);
        id(sms_message).publish_state(message);
  on_incoming_call:
    - lambda: |-
        id(caller_id_text_sensor).publish_state(caller_id);
  on_call_connected:
    - logger.log:
        format: Call connected
  on_call_disconnected:
    - logger.log:
        format: Call disconnected
  on_ussd_received:
    - lambda: |-
        id(ussd_message).publish_state(ussd);
``
I go to test it

pilot1981, 2023-02-14:
not solved, now I don't found any error in HA, but I didn't received SMS...I can increase verbose of log?

ssieb, 2023-02-14:
https://esphome.io/components/logger.html

pilot1981, 2023-02-14:
I already used this configuration:

logger:
  level: DEBUG

this isn't useful for my scope?

pilot1981, 2023-02-14:
In ESPHOME logs I don't see anything useful to troubleshoot...in HA logs don't any entry related to ESPHOME...

pilot1981, 2023-02-14:
I also tried:

logger:
level: verbose

but nothing changed in logs...

ssieb, 2023-02-14:
Continue this on discord.  Github issues are not for support.

closed 2023-02-22 by ssieb