esphome/issues #4168: ESP32_camera on esp32-s3-WROOM-1 module gives watchdog reset

lecrams, 2023-02-15:
### The problem

When I add the CAM32 component to my board yaml (see below) I get into a bootloop. Loggings can be found below.
Using the SSD1306 display which is also on the board without the camera definition has no problems

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.3

### What platform are you using?

ESP32

### Board

LilyGo T-camera-S3 with ESP32-S3 WROOM-1 module

### Component causing the issue

ESP32_CAM

### Example YAML snippet

```yaml
esphome:
  name: esp-lilycam

esp32:
  board: esp32-s3-devkitc-1
#  board: esp32cam 
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 5.1.1
  variant: esp32s3

# ESP32-CAM
esp32_camera:
  name: LilyCam
  external_clock:
    pin: GPIO38
    frequency: 20MHz
  i2c_pins:
    sda: GPIO5
    scl: GPIO4
  data_pins: [GPIO14, GPIO47, GPIO48, GPIO21, GPIO13, GPIO11, GPIO10, GPIO9]
  vsync_pin: GPIO8
  href_pin: GPIO18
  pixel_clock_pin: GPIO12
```

### Anything in the logs that might be useful for us?

```txt
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377508
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x43c
load:0x403b6000,len:0xbd0
load:0x403ba000,len:0x29c8
entry 0x403b61d8
E (183) psram: PSRAM ID read error: 0x00ffffff
E (5217) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (5217) task_wdt: - loopTask (CPU 1)
E (5217) task_wdt: Tasks currently running:
E (5217) task_wdt: CPU 0: IDLE
E (5217) task_wdt: CPU 1: IDLE
E (5217) task_wdt: Aborting.
```

lecrams, 2023-03-08:
issue solved

maxpersg, 2023-03-11:
> issue solved

can past the code for the yaml file? i'm getting similar issue as well. thanks1

trikerman, 2023-04-02:
> ```yaml
> # ESP32-CAM
> esp32_camera:
>   name: LilyCam
>   external_clock:
>     pin: GPIO38
>     frequency: 20MHz
>   i2c_pins:
>     sda: GPIO5
>     scl: GPIO4
>   data_pins: [GPIO14, GPIO47, GPIO48, GPIO21, GPIO13, GPIO11, GPIO10, GPIO9]
>   vsync_pin: GPIO8
>   href_pin: GPIO18
>   pixel_clock_pin: GPIO12
> ```

ok, yaml file please?

trikerman, 2023-04-02:
> issue solved

ok, yaml file please?

lecrams, 2023-04-03:
this specific  issue was solved but the yaml for the camera part is still not working. Seems that that the PMU is not init correctlly

webcharlyx, 2023-04-30:
PMU is [AXP2101](https://github.com/Xinyuan-LilyGO/LilyGo-Cam-ESP32S3#power-channel) -> Camera: ALDO1/ALDO2/ALDO4
[in this example](https://github.com/Xinyuan-LilyGO/LilyGo-Cam-ESP32S3/blob/master/examples/MinimalCameraExample/main.cpp#L42) they show how to adjust the three voltages.
with https://github.com/Xinyuan-LilyGO/LilyGo-Cam-ESP32S3/tree/master/lib/XPowersLib

I guess a new ESPhome component for AXP2101 is mandatory as first step.

joopdo, 2023-06-01:
This still seems an issue with v2023.5.5.

mrdis, 2023-06-01:
Hi, I managed to make it work by using the following YAML

```yaml
esphome:
  name: prova
  friendly_name: prova
  libraries:
    - XPowersLib=https://github.com/lewisxhe/XPowersLib.git
  includes:
    - AXP2101_component.h
  platformio_options:
    build_flags: "-DBOARD_HAS_PSRAM -UARDUINO_USB_CDC_ON_BOOT"
```

I had to create the file "AXP2101_component.h" in the /config/esphome folder with this content:

```cpp
#include "esphome.h"

#define XPOWERS_CHIP_AXP2101
#include "XPowersLib.h"

#define I2C_SDA                     (7)
#define I2C_SCL                     (6)

class AXP2101Component : public Component {
 public:
  void setup() override {
    // This will be called once to set up the component
    // think of it as the setup() call in Arduino
    ESP_LOGD("custom", "Custom component setup");
    
    if (!PMU.begin(Wire, AXP2101_SLAVE_ADDRESS, I2C_SDA, I2C_SCL)) {
        Serial.println("Failed to initialize power.....");
        while (1) {
            delay(5000);
        }
    }
    //Set the working voltage of the camera, please do not modify the parameters
    PMU.setALDO1Voltage(1800);  // CAM DVDD  1500~1800
    PMU.enableALDO1();
    PMU.setALDO2Voltage(2800);  // CAM DVDD 2500~2800
    PMU.enableALDO2();
    PMU.setALDO4Voltage(3000);  // CAM AVDD 2800~3000
    PMU.enableALDO4();

    // TS Pin detection must be disable, otherwise it cannot be charged
    PMU.disableTSPinMeasure();
    
  }
  private:
    XPowersPMU  PMU;
};
```

lecrams, 2023-06-02:
I assume that this code comes as addition to the 'normal' esp32 yaml like the vsync and datapin definitions?

joopdo, 2023-06-02:
@Marcel, that's right. For example, here's my config with only the password scrubbed. This enables both camera and screen:

````
esphome:
  name: camera-2
  friendly_name: "Camera 2 - ESP32 S3"
  libraries:
    - XPowersLib=https://github.com/lewisxhe/XPowersLib.git
    - "Wire"
  includes:
    - AXP2101_component.h   
  platformio_options:
    build_flags: "-DBOARD_HAS_PSRAM -UARDUINO_USB_CDC_ON_BOOT"

custom_component:
- lambda: |-
    auto axp2101 = new AXP2101Component();
    return {axp2101};
  components:
  - id: axp2101

esp32:
  board: esp32s3box
  framework:
    type: arduino

logger:

# Enable Home Assistant API
api:
  encryption:
    key: "<scrub>"

ota:
  password: "4f9be56231e85ec6151e54196825efc3"

wifi:
  ssid: !secret wifi_ssid_2
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-S3 Fallback Hotspot"
    password: "<scrub>"
captive_portal:

psram:
time:
  - platform: homeassistant
    id: homeassistant_time  

font:
  - file: "nanum.ttf"
    id: tnr1
    size: 20

web_server:
  version: 2

i2c:
  sda: GPIO07
  scl: GPIO06

esp32_camera:
  name: camera-2
  external_clock:
     pin: GPIO38
     frequency: 20MHz
  i2c_pins:
    sda: GPIO05
    scl: GPIO04
  data_pins: [GPIO14, GPIO47, GPIO48, GPIO21, GPIO13, GPIO11, GPIO10, GPIO09]
  vsync_pin: GPIO8
  href_pin: GPIO18
  pixel_clock_pin: GPIO12
  vertical_flip: false
  horizontal_mirror: false
  resolution: 1024x768
 # agc_gain_ceiling: 128x
 # aec2:  true
  
  #vertical_flip: false
  max_framerate: 2 fps
  idle_framerate: 0.05 fps
  # ...
esp32_camera_web_server:
  - port: 8080
    mode: stream
  - port: 8081
    mode: snapshot      

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    address: 0x3C
    rotation: 180
    lambda: |-
      it.strftime(64, 0, id(tnr1), TextAlign::TOP_CENTER,"%m-%d-%Y", id(homeassistant_time).now());
      it.strftime(64, 64, id(tnr1), TextAlign::BASELINE_CENTER, "%I:%M:%S%p", id(homeassistant_time).now());
````

lecrams, 2023-06-06:
Sorry for my late reaction. When I try to build the above code I get this error:

[image]

Any Idea what can be wrong? Can this be a version mismatch of the framework? WHat version are you usinjg?

lecrams, 2023-06-06:
Having exactly the same versions except for
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1

But this has nothing todo imo with the errors. Did also a clean build but exactly the same result.

lecrams, 2023-06-08:
It can now be compiled without any errors. I was using a local esp32_camera component.
I can run the FW however it is not stable. If I only power the board through the USB-c port It reset every know and then. 
In the loggings I also see that the camera module is not always found at restet

If I connect the serial port on the USB-C everything seems to be okay.

Any idea what can cause this behaviour?

martijnbuts, 2023-07-06:
> It can now be compiled without any errors. I was using a local faulty esp32_camera component. I can run the FW however it is not stable. If I only power the board through the USB-c port It reset every know and then. In the loggings I also see that the camera module is not always found at restet
> 
> If I connect the serial port on the USB-C everything seems to be okay.
> 
> Any idea what can cause this behaviour?

Can you share your yaml file?

lecrams, 2023-07-10:
Here is my YAML

esphome:
  name: esp-lily-cam
  platformio_options:
    build_flags: "-DBOARD_HAS_PSRAM -UARDUINO_USB_CDC_ON_BOOT"
    upload_speed: 921600
    monitor_speed: 115200
  friendly_name: "Camera 2"
  libraries:
    - XPowersLib=https://github.com/lewisxhe/XPowersLib.git
    - "Wire"
  includes:
    - AXP2101_component.h

custom_component:
  - lambda: |-
      auto axp2101 = new AXP2101Component();
      return {axp2101};
    components:
      - id: axp2101    

esp32:
  board: esp32s3box
  framework:
    type: arduino

wifi:
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.178.xx
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.178.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

  networks:
    - ssid: "XXXX"
      password: ""

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp Cam Fallback Hotspot"
    password: "YyzRO7QXiUpO"

captive_portal:

external_components:
  - source: my_components
    components: [esp32_camera, decibel_meter, audio_source, i2s_audio_source, i2s_audio_sink, sound_generator]

  - source: github://mrhatman/esphome@dev
    components: [ audio_source, decibel_meter, i2s_audio_source, i2s_audio_sink, sound_generator ]

  use all components from a local folder
  - source:
      type: local
      path: my_components

      
# Enable logging
# logger:
 # level: DEBUG
 # baud_rate: 115200
  
# Enable Home Assistant API
api:
 reboot_timeout: 0s

ota:

web_server:
  version: 2

esp32_camera_web_server:
  - port: 8080
    mode: stream
  - port: 8081
    mode: snapshot  

time:
  - platform: homeassistant
    id: esptime

font:
  - file: 'fonts/slkscr.ttf'
    id: font1
    size: 8

  - file: 'fonts/BebasNeue-Regular.ttf'
    id: font2
    size: 30

  - file: 'fonts/Arial.ttf'
    id: font3
    size: 14

i2c:
  sda: GPIO7
  scl: GPIO6
  frequency: 200kHz
  scan: True

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    address: 0x3C
    external_vcc: False
    lambda: |-
      // Print "My Smart Home" in top center.
      it.printf(64, 0, id(font1), TextAlign::TOP_CENTER, "My Smart Home");

      // Print time in HH:MM format
      it.strftime(0, 30, id(font2), TextAlign::BASELINE_LEFT, "%H:%M", id(esptime).now());

      // Print inside temperature (from homeassistant sensor)
      if (id(inside_temperature).has_state()) {
        it.printf(127, 10, id(font3), TextAlign::TOP_RIGHT , "%.1f°", id(inside_temperature).state);
      }

      // Print outside temperature (from homeassistant sensor)
      if (id(outside_temperature).has_state()) {
        it.printf(127, 40, id(font3), TextAlign::BASELINE_RIGHT , "%.1f°", id(outside_temperature).state);
      }

# ESP32-CAM
esp32_camera:
  name: LilyCam
  external_clock:
    pin: GPIO38
    frequency: 20MHz
  i2c_pins:
    sda: GPIO5
    scl: GPIO4
  data_pins: [GPIO14, GPIO47, GPIO48, GPIO21, GPIO13, GPIO11, GPIO10, GPIO9]
  vsync_pin: GPIO8
  href_pin: GPIO18
  pixel_clock_pin: GPIO12
  vertical_flip: false
  horizontal_mirror: false
  resolution: 1024x768

audio_source:
  - platform: i2s_audio_source
    id: i2s_mic
    name: "Smaller room I2S Mic"
    i2s_lrclk_pin: GPIO40
    i2s_din_pin: GPIO41
    i2s_bclk_pin: GPIO40 #clock for output I2S
    bits_per_sample: 16
    audio_frequency: 16000
    mode: mono

binary_sensor:
  - platform: status
    name: "ESP LilyCam status"

  - platform: gpio
    pin: GPIO17
    name: "PIR Sensor"
    device_class: motion    

sensor:
#  - platform: decibel_meter
#    sensitivity: -23
#    audio_source: i2s_mic
#    name: decibels

  - platform: wifi_signal
    name: "ESP LilyCam WiFi Signal Sensor"
    update_interval: 20s
  - platform: uptime
    name: "LilyCam Uptime"

  - platform: homeassistant
    id: inside_temperature
    entity_id: sensor.bme280_temperature
    internal: true

  - platform: homeassistant
    id: outside_temperature
    entity_id: sensor.temperature
    internal: true    

text_sensor:
  - platform: version
    name: "ESPHome Version"
  - platform: wifi_info
    ip_address:
      name: IP Address
    ssid:
      name: Connected SSID
    bssid:
      name: Connected BSSID
    mac_address:
      name: Mac Address

switch:
  - platform: restart
    name: "CAM Restart"

martijnbuts, 2023-07-24:
Thanks for the config, i added the axp2101 and now it works.
What type of microphone is used in the Lilicam? Does it work in ESPhome? (it would be handy if it can be used as a voice assistant in HA) 

If I understand your yaml correctly u use a external component.

stathisktm, 2023-09-20:
did anyone manage to power it from the 5 pin? 5v or 3.3v?

cybuletz, 2023-11-04:
> It is an I2S microphone. Check the schematic for details : https://github.com/Xinyuan-LilyGO/LilyGo-Cam-ESP32S3/blob/master/schematic/T-Camera-S3-V1.6.pdf

Hello, sorry for digging this up, but can you please share those components you are using in your YAML. I am trying to make the microphone work but I am getting the same as @martijnbuts : [sensor:094]: 'decibels': Sending state -inf with 0 decimals of accuracy

Thanks in advance!

lecrams, 2023-11-06:
can't recall where I got them but here is a link to my Gdrive where I stored it.
https://drive.google.com/file/d/1SVJS7dZHQ15pYIGGPqyrXlh7gkrgI3Vl/view?usp=drive_link

cybuletz, 2023-11-06:
Thanks!

În lun., 6 nov. 2023 la 10:08, Marcel Stumpel ***@***.***> a
scris: