esphome/issues #4169: ESP Home 2023.2.0 Bluetooth proxy update failing

yestrdave, 2023-02-15:
### The problem

Updated to ESP Home 2023.2.0 this morning then proceeded to update the Bluetooth proxies. I tried two firmware updates via wireless connection and then one with serial. This error is present with both methods

[image]

HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash

framework-espidf @ 3.40402.0 (4.4.2)
tool-cmake @ 3.16.4
tool-ninja @ 1.7.1
toolchain-esp32ulp @ 2.35.0-20220830
toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- Improv @ 1.2.3
*** [/data/house-garage-bluetooth-proxy/.pioenvs/house-garage-bluetooth-proxy/src/esphome/components/mdns/mdns_esp32_arduino.o] Source src/esphome/components/mdns/mdns_esp32_arduino.cpp' not found, needed by target /data/house-garage-bluetooth-proxy/.pioenvs/house-garage-bluetooth-proxy/src/esphome/components/mdns/mdns_esp32_arduino.o'.
========================== [FAILED] Took 6.05 seconds ==========================

### Which version of ESPHome has the issue?

2023.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.4

### What platform are you using?

ESP32

### Board

Atom Lite

### Component causing the issue

firmware update

### Example YAML snippet

```yaml
substitutions:
  name: "house-garage-bluetooth-proxy"
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false

Left out wifi and static ip
```

niemyjski, 2023-02-15:
I'm also seeing this on all of my atom lites

```
*** [/data/atom-lite-xx/.pioenvs/atom-lite-03/src/esphome/components/mdns/mdns_esp32_arduino.o] Source `src/esphome/components/mdns/mdns_esp32_arduino.cpp' not found, needed by target `/data/atom-lite-xx/.pioenvs/atom-lite-xx/src/esphome/components/mdns/mdns_esp32_arduino.o'.
========================== [FAILED] Took 7.17 seconds ==========================
```

jum0n, 2023-02-15:
Same issue as original post on mine:
`Compiling /data/esp32-bluetooth-proxy-2e9628/.pioenvs/esp32-bluetooth-proxy-2e9628/src/esphome/components/mdns/mdns_component.o
*** [/data/esp32-bluetooth-proxy-2e9628/.pioenvs/esp32-bluetooth-proxy-2e9628/src/esphome/components/mdns/mdns_esp32_arduino.o] Source `src/esphome/components/mdns/mdns_esp32_arduino.cpp' not found, needed by target `/data/esp32-bluetooth-proxy-2e9628/.pioenvs/esp32-bluetooth-proxy-2e9628/src/esphome/components/mdns/mdns_esp32_arduino.o'.
========================= [FAILED] Took 72.36 seconds =========================`

Mariusthvdb, 2023-02-15:
adding to this:

after having 3 failed devices, they remain in the update section to be updated .

Yet, the dashboard, which did show them with an update available, now shows 'online', without anything further. checking the log indeed reveals they are still on the former firmware.2022.12.8

earlyhost, 2023-02-15:
I got the same error, however cleaning the build files before installing the update did do the trick.

danielduerr, 2023-02-15:
I always do an `esphome clean` before every build, did not help in this case. I then completely removed the `.esphome` directory from my build env and that didn't fix it either. Are there other build dirs I should remove?

Mariusthvdb, 2023-02-15:
dont think we should have to delete anything in the file system when updating.... this is an issue and should be fixed properly.

let's just wait for a dev to see this

c0bra99, 2023-02-15:
Clicking the "Clean Build Files" menu option in the UI before installing/updating allows the latest version to be installed (had the same issue as the original post with mdns_esp32_arduino.o)

This should not be necessary, but is an easy workaround for updating a couple devices, but could be a pain to update many devices.

[image]

Electronlibre2012, 2023-02-15:
> I got the same error, however cleaning the build files before installing the update did do the trick.

Thanks! it works

jum0n, 2023-02-15:
I was playing around with one of my bluetooth proxies and decided to clean build files, works fine now.  Guessing we should do that at each major version just in case.  Lots of changes in this one.

jum0n, 2023-02-15:
> adding to this:
> 
> after having 3 failed devices, they remain in the update section to be updated .
> 
> Yet, the dashboard, which did show them with an update available, now shows 'online', without anything further. checking the log indeed reveals they are still on the former firmware.2022.12.8
> 
> and don show the updates being available anymore

I've seen this happen before but didn't check into it.  It did happen this time as well so I think it might be an issue that has been lurking since the versioning changes.

Mariusthvdb, 2023-02-15:
I managed to update the m5 stack by cleaning the build files, but my 

took 5 minute before returning and not having updated:

[image]

ssieb, 2023-02-15:
It's a known issue when using esp-idf that it doesn't always handle build dependencies correctly and you sometimes need to clean the build directory first.

Mariusthvdb, 2023-02-15:
> I successfully updated all five of my Atom Lites after cleaning the build files.
> 
> 
> 
> Thanks @earlyhost !
> 
> 
> 
> Looks like I can close this.. Any objections?

No objections but still would recommend having a dev have a look at this 'requirement'

jesserockz, 2023-02-15:
IIRC ESPHome is actually supposed to be doing an automatic clean build when there is a version change. I will have a look into that part.

austwhite, 2023-02-15:
I was just looking for this issue as I have it also on the prebuilt ESP Home BT Proxy.
Saved me hours of messing around.  Thank you!

robloh, 2023-02-15:
Cleaning the build files fixed it for me too

niemyjski, 2023-02-16:
Thanks for the fix!

> and don show the updates being available anymore

I've seen this for a long time anytime you run update all and the device that fails to update will be marked up-to-date. The ui never recovers.

Stooovie, 2023-03-24:
Cleaning the build files did NOT fix the issue for me. Unable to install anything beyond 2022.12.6

ssieb, 2023-03-24:
That's a completely different issue than this one.