esphome/issues #4170: CCS811 Set baseline in configuration is not used during measurement

Koen00, 2023-02-15:
### The problem

Configured baseline as mentioned in the [docs](https://esphome.io/components/sensor/ccs811.html) is not used or looks like it is not used in the deployed operation. 

The start log gives the output where the baseline is correctly given. But at a measurement, another value for baseline is given in the log output. After 'startup' of the sensor/esp there is also some time of no measurement, because of the self calibration/initialisation which will be avoided if there is a set baseline (like which is mentioned in the documentation). 

### Which version of ESPHome has the issue?

2023.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.01.1

### What platform are you using?

ESP32

### Board

featheresp32

### Component causing the issue

CCS811

### Example YAML snippet

```yaml
esp32:
  board: featheresp32
  framework:
    type: arduino
    version: recommended

sensor:
  - platform: ccs811
    id: CCSSensor
    baseline: 0x56C2
    eco2:
      name: "$node_name eCO2 Value"
      id: eco2
    tvoc:
      name: "$node_name Total Volatile Organic Compound"
      id: tvoc
    address: 0x5A
    temperature: Temperature
    humidity: humi
    update_interval: never

  - platform: tmp117
    name: $node_name Temperature
    update_interval: 60s
    address: 0x48
    id: Temperature

  - platform: dht
    pin: 15
    model: dht11
    humidity:
      name: "$node_name Humidity"
      internal: False
      id: humi
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
[15:26:15][C][ccs811:152]: CCS811
[15:26:15][C][ccs811:153]:   Address: 0x5A
[15:26:15][C][ccs811:154]:   Update Interval: never
[15:26:15][C][ccs811:155]:   CO2 Sensor 'esp32-02 eCO2 Value'
[15:26:15][C][ccs811:155]:     Device Class: 'carbon_dioxide'
[15:26:15][C][ccs811:155]:     State Class: 'measurement'
[15:26:15][C][ccs811:155]:     Unit of Measurement: 'ppm'
[15:26:15][C][ccs811:155]:     Accuracy Decimals: 0
[15:26:15][C][ccs811:155]:     Icon: 'mdi:molecule-co2'
[15:26:15][C][ccs811:156]:   TVOC Sensor 'esp32-02 Total Volatile Organic Compound'
[15:26:15][C][ccs811:156]:     Device Class: 'volatile_organic_compounds'
[15:26:15][C][ccs811:156]:     State Class: 'measurement'
[15:26:15][C][ccs811:156]:     Unit of Measurement: 'ppb'
[15:26:15][C][ccs811:156]:     Accuracy Decimals: 0
[15:26:15][C][ccs811:156]:     Icon: 'mdi:radiator'
[15:26:15][C][ccs811:159]:   Baseline: 56C2

[15:29:10][D][ccs811:114]: Got co2=1521 ppm, tvoc=186 ppb, baseline=0x81BB
[15:29:10][D][sensor:127]: 'esp32-02 eCO2 Value': Sending state 1521.00000 ppm with 0 decimals of accuracy
[15:29:10][D][sensor:127]: 'esp32-02 Total Volatile Organic Compound': Sending state 186.00000 ppb with 0 decimals of accuracy

after restart:

[15:39:42][D][ccs811:114]: Got co2=400 ppm, tvoc=0 ppb, baseline=0x6ABB
[15:39:42][D][sensor:127]: 'esp32-02 eCO2 Value': Sending state 400.00000 ppm with 0 decimals of accuracy
[15:39:42][D][sensor:127]: 'esp32-02 Total Volatile Organic Compound': Sending state 0.00000 ppb with 0 decimals of accuracy
[15:39:45][D][tmp117:021]: Got temperature=15.00°C
[15:39:45][D][sensor:127]: 'esp32-02 Temperature': Sending state 15.00000 °C with 1 decimals of accuracy
```

stas-sl, 2023-02-16:
Baseline is not fixed forever, it is being updated whenever the sensor gets exposed to fresh air (or fresher than it was at the time when previous baseline was logged). It also depends a lot on current env/temp/humidity. Actually I’m not sure if there is much sense to specify it in config, as you have to update it periodically to reflect current environment, or just keep it running for a day and let it figure out the baseline on its own.

Koen00, 2023-02-20:
okay. Is there a way to speed up the startup process? After a restart of the ESP, the CCS811 also needs lots of time to generate measurements. But it's not disconnected from the power supply. Ideally, I want to use a deep sleep and take an air quality measurement every 1-5 minutes. But with this speed, I don't think it's viable. 

Saving the latest baseline to the flash, not an option?

stas-sl, 2023-02-20:
Have you tried deep sleep? Is baseline being reset after waking up (with or without baseline specified in config)? I believe CCS811 should still keep it in memory, as the sensor is still powered in deep sleep mode. In fact it has 4 different modes of operation 0-3, each having different power consumption. If the sensor is powered on or after being idle (mode 0) for long time it has to be conditioned for at least 20 min according to the specification before accurate readings even if you restore the baseline. So I don’t think completely powering off or putting the sensor in idle mode 0 is a good idea. Instead probably I would use low power mode 3, in which measurements are taken each 60 seconds.

Storing in flash could be an option I think, however I’m not sure if it is required for deep sleep. But if you want to restore it after reset/power off, you could try storing it in flash.