esphome/issues #4171: ld2410: Component ld2410 cannot be loaded via YAML (no CONFIG_SCHEMA).

Electronlibre2012, 2023-02-15:
### The problem

I have 4 presence detection with Wemos D1 mini base on the custom_component and its working great.

I see that EspHome 2023.2.0 have now LD2410 native component, so i copy the config in a new device page but i get theses errors : 
```
INFO Reading configuration /config/esphome/capteur-presence-sejournew.yaml...
Failed config

ld2410: [source /config/esphome/capteur-presence-sejournew.yaml:41]
  
  Component ld2410 cannot be loaded via YAML (no CONFIG_SCHEMA).
  timeout: 150s
  max_move_distance: 6m
  max_still_distance: 0.75m
  g0_move_threshold: 10
  g0_still_threshold: 20
  g1_move_threshold: 10
  g1_still_threshold: 20
  g2_move_threshold: 20
  g2_still_threshold: 21
  g3_move_threshold: 30
  g3_still_threshold: 31
  g4_move_threshold: 40
  g4_still_threshold: 41
  g5_move_threshold: 50
  g5_still_threshold: 51
  g6_move_threshold: 60
  g6_still_threshold: 61
  g7_move_threshold: 70
  g7_still_threshold: 71
  g8_move_threshold: 80
  g8_still_threshold: 81
binary_sensor.ld2410: [source /config/esphome/capteur-presence-sejournew.yaml:69]
  
  Platform not found: 'binary_sensor.ld2410'.
  platform: ld2410
  has_target: 
    name: Presence
  has_moving_target: 
    name: Moving Target
  has_still_target: 
    name: Still Target
sensor.ld2410: [source /config/esphome/capteur-presence-sejournew.yaml:78]
  
  Platform not found: 'sensor.ld2410'.
  platform: ld2410
  moving_distance: 
    name: Moving Distance
  still_distance: 
    name: Still Distance
  moving_energy: 
    name: Move Energy
  still_energy: 
    name: Still Energy
  detection_distance: 
    name: Detection Distance
```

should i remove the custom_component to make the native work? i dont think so...

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.4

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

### Component causing the issue

ld2410

### Example YAML snippet

```yaml
esphome:
  name: capteur-presence-sejournew
  friendly_name: capteur-presence-sejournew

esp8266:
  board: d1_mini

# Enable logging
logger:
  level: WARN
  baud_rate: 0
  
# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pwd
  manual_ip:
    static_ip: 192.168.1.88
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1
  fast_connect: on
  power_save_mode: none

uart:
  id: uart1
  tx_pin: TX
  rx_pin: RX
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

ld2410:
  timeout: 150s
  max_move_distance : 6m
  max_still_distance: 0.75m
  g0_move_threshold: 10
  g0_still_threshold: 20
  g1_move_threshold: 10
  g1_still_threshold: 20
  g2_move_threshold: 20
  g2_still_threshold: 21
  g3_move_threshold: 30
  g3_still_threshold: 31
  g4_move_threshold: 40
  g4_still_threshold: 41
  g5_move_threshold: 50
  g5_still_threshold: 51
  g6_move_threshold: 60
  g6_still_threshold: 61
  g7_move_threshold: 70
  g7_still_threshold: 71
  g8_move_threshold: 80
  g8_still_threshold: 81

status_led:
  pin:
    number: GPIO2 #ESP8266 OnBroad LED
    inverted: false
      
binary_sensor:
  - platform: ld2410
    has_target:
      name: Presence
    has_moving_target:
      name: Moving Target
    has_still_target:
      name: Still Target
  - platform: gpio
    pin: D1
    name: "Sejour occupancy"
    id: sejour_occupancy
    device_class: occupancy
    on_press:
      then:
        - light.turn_on:
            id: led_presence_sejour
            brightness: 25%
            red: 100%
            green: 0%
            blue: 0%
    on_release:
      then:
        - light.turn_on:
            id: led_presence_sejour
            brightness: 25%
            red: 0%
            green: 100%
            blue: 0%

sensor:
  - platform: ld2410
    moving_distance:
      name : Moving Distance
    still_distance:
      name: Still Distance
    moving_energy:
      name: Move Energy
    still_energy:
      name: Still Energy
    detection_distance:
      name: Detection Distance

light:
  - platform: neopixelbus
    type: GRB
    variant: WS2812x
    pin: D2
    num_leds: 1
    name: "Led Presence Sejour"
    id: led_presence_sejour
    effects:
      - strobe:
      - addressable_rainbow:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:
      - addressable_flicker:
```

Electronlibre2012, 2023-02-17:
I reply to myself because i have no received some help...i found the solution : 

So for those having problems with ld2410 component in EspHome 2023.2.1, add this to your node did the trick :
```
external_components:
  - source: github://esphome/esphome@dev
    components: [ ld2410 ]
```
i dont understand why i am the only one who having trouble with LD2410 component in the last EspHome 2023.2.1 or 2023.2.2 of today, same problem…

I use the “standard” AddOn : EspHome, not “Beta” or “Dev” version, and it seems that LD2410 is, for the moment, only on the Dev version of the EspHome AddOn, that why the component LD2410 is not found on the basic version of EspHome :smile:

[image]

regevbr, 2023-03-03:
see https://github.com/esphome/esphome/pull/4434 which you can start using right away and is the next and enhanced version of the component

ssieb, 2023-03-21:
> should i remove the custom_component to make the native work? i dont think so...

Most likely you do have to do that.  You didn't include any info about that in your post.

closed 2023-03-21 by ssieb