esphome/issues #4172: Too many ble events to process

tergontor, 2023-02-15:
### The problem

Five esp32devkitv1 boards with Project esphome.bluetooth-proxy version 1.0 and espidf framework. Of those 5, three show continuous records in the log (20 to 25 per minute) with this information: “W][esp32_ble_tracker:121]: Too many BLE events to process. Some devices may not show up.”
After doing some research, I have found out that these registers are only output when I have the SPI bus in the code of the board (necessary to install the st7735s tft screen through which I see the data that I consider necessary such as temperature, humidity, co2, ...).

### Which version of ESPHome has the issue?

ESPHome version 2023.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.2.4

### What platform are you using?

ESP32-IDF

### Board

esp32 devkitv1

### Component causing the issue

bus spi

### Anything in the logs that might be useful for us?

```txt
[17:28:35][D][homeassistant.sensor:024]: 'sensor.rm4_pro_salon_temperature': Got state 19.22
[17:28:35][D][sensor:127]: 'temperatura_salon': Sending state 19.22000  with 1 decimals of accuracy
[17:28:35][D][homeassistant.sensor:024]: 'sensor.rm4_pro_salon_humidity': Got state 49.23
[17:28:35][D][sensor:127]: 'humedad_salon': Sending state 49.23000  with 1 decimals of accuracy
[17:28:38][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:28:39][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:28:43][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:28:44][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:28:53][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:28:54][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:28:54][D][homeassistant.sensor:024]: 'sensor.terraza_temperatura': Got state 13.40
[17:28:54][D][sensor:127]: 'terraza_temperatura': Sending state 13.40000  with 1 decimals of accuracy
[17:28:58][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:28:59][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:28:59][D][sensor:127]: 'CO2 Salón': Sending state 1625.00000 ppm with 0 decimals of accuracy
[17:28:59][D][sensor:127]: 'Temperatura Salón SCD41': Sending state 21.34598 °C with 2 decimals of accuracy
[17:28:59][D][sensor:127]: 'Humedad Salón SCD41': Sending state 43.94073 % with 2 decimals of accuracy
[17:29:00][D][internal_temperature:051]: Got temperature: 62.2°C
[17:29:00][D][sensor:127]: 'Temp del ESP32 scd41-salón': Sending state 62.22223 °C with 1 decimals of accuracy
[17:29:03][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:29:04][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:29:13][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:29:14][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:29:18][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:29:19][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:29:23][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:29:24][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:29:28][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:29:29][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:29:33][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:29:34][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:29:35][D][homeassistant.sensor:024]: 'sensor.rm4_pro_salon_humidity': Got state 49.19
[17:29:35][D][sensor:127]: 'humedad_salon': Sending state 49.19000  with 1 decimals of accuracy
[17:29:38][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:29:39][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:29:43][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:29:44][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:29:48][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[... 28 lines cut ...]
[17:30:48][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:30:49][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:30:53][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:30:54][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:30:54][D][homeassistant.sensor:024]: 'sensor.terraza_temperatura': Got state 13.30
[17:30:54][D][sensor:127]: 'terraza_temperatura': Sending state 13.30000  with 1 decimals of accuracy
[17:30:58][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:30:59][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:30:59][D][sensor:127]: 'CO2 Salón': Sending state 1620.00000 ppm with 0 decimals of accuracy
[17:30:59][D][sensor:127]: 'Temperatura Salón SCD41': Sending state 21.34331 °C with 2 decimals of accuracy
[17:30:59][D][sensor:127]: 'Humedad Salón SCD41': Sending state 43.88428 % with 2 decimals of accuracy
[17:31:00][D][internal_temperature:051]: Got temperature: 62.2°C
[17:31:00][D][sensor:127]: 'Temp del ESP32 scd41-salón': Sending state 62.22223 °C with 1 decimals of accuracy
[17:31:03][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:31:04][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:31:08][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:31:14][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:31:18][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:31:19][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:31:23][D][sensor:127]: 'cO2_salon_wifi_signal': Sending state -63.00000 dBm with 0 decimals of accuracy
[17:31:28][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:31:29][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:31:33][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
[17:31:34][W][esp32_ble_tracker:111]: Too many BLE events to process. Some devices may not show up.
```

tergontor, 2023-02-17:
I don't understand why this topic has been closed without any explanation

jonny190, 2024-10-20:
Same issue, Any updates?