esphome/issues #4174: undefined reference to `esp_eth_phy_new_jl1101'

kquinsland, 2023-02-15:
### The problem

After updating to the new `2023.02` release this morning, I attempted to update the firmware on my BTLE proxy device and got this failure during compilation:

> 
> /home/karl/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/remote-bt-01/src/esphome/components/ethernet/ethernet_component.cpp.o: in function `esphome::ethernet::EthernetComponent::setup()':
> /home/karl/Projects/esphome/hardware/build-root/remote-bt-01/src/esphome/components/ethernet/ethernet_component.cpp:100: undefined reference to `esp_eth_phy_new_jl1101'

### Which version of ESPHome has the issue?

2023.02

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

LILYGO TTGO T-Internet-POE ESP32-WROOM LAN8720A

### Component causing the issue

ethernet

### Example YAML snippet

```yaml
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
```

syssi, 2023-02-15:
Same here.

```
Linking .pioenvs/zb-gw03-coordinator/bootloader.elf
Building .pioenvs/zb-gw03-coordinator/bootloader.bin
Creating esp32 image...
Successfully created esp32 image.
Linking .pioenvs/zb-gw03-coordinator/firmware.elf
/home/runner/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/zb-gw03-coordinator/src/esphome/components/ethernet/ethernet_component.o:(.literal._ZN7esphome8ethernet17EthernetComponent5setupEv+0x44): undefined reference to `esp_eth_phy_new_jl1101'
/home/runner/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/zb-gw03-coordinator/src/esphome/components/ethernet/ethernet_component.o: in function `esphome::ethernet::EthernetComponent::setup()':
/home/runner/work/esphome-zb-gw03/esphome-zb-gw03/.esphome/build/zb-gw03-coordinator/src/esphome/components/ethernet/ethernet_component.cpp:57: undefined reference to `esp_eth_phy_new_jl1101'
collect2: error: ld returned 1 exit status
*** [.pioenvs/zb-gw03-coordinator/firmware.elf] Error 1
========================= [FAILED] Took 151.86 seconds =========================
Error: Process completed with exit code 1.
```

https://github.com/syssi/esphome-zb-gw03/actions/runs/4188369897/jobs/7259417281

Can be reproduced using this example configuration:

https://raw.githubusercontent.com/syssi/esphome-zb-gw03/main/coordinator-example.yaml

jesserockz, 2023-02-16:
@kquinsland Can you try a clean build? (But also try 2023.2.1 first as that should run a clean build automatically)

@syssi I am guessing it could be the same issue for you as you cache the `.esphome` directory in your CI runs. (Sitenote: Have you looked at the https://github.com/esphome/build-action ? )

syssi, 2023-02-16:
I clean build doesn't help. I've removed the cache of the CI pipeline manually and tried a local build:

```
$ cd /tmp
$ git clone git@github.com:syssi/esphome-zb-gw03.git 
$ cd esphome-zb-gw03
$ esphome run coordinator-example.yaml
[...]
Linking .pioenvs/zb-gw03-coordinator/bootloader.elf
Building .pioenvs/zb-gw03-coordinator/bootloader.bin
Creating esp32 image...
Successfully created esp32 image.
Linking .pioenvs/zb-gw03-coordinator/firmware.elf
/home/sebastian/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/zb-gw03-coordinator/src/esphome/components/ethernet/ethernet_component.o:(.literal._ZN7esphome8ethernet17EthernetComponent5setupEv+0x44): undefined reference to `esp_eth_phy_new_jl1101'
/home/sebastian/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/zb-gw03-coordinator/src/esphome/components/ethernet/ethernet_component.o: in function `esphome::ethernet::EthernetComponent::setup()':
/tmp/esphome-zb-gw03/.esphome/build/zb-gw03-coordinator/src/esphome/components/ethernet/ethernet_component.cpp:57: undefined reference to `esp_eth_phy_new_jl1101'
collect2: error: ld returned 1 exit status
*** [.pioenvs/zb-gw03-coordinator/firmware.elf] Error 1
$
```

kquinsland, 2023-02-16:
> @kquinsland Can you try a clean build? (But also try 2023.2.1 first as that should run a clean build automatically)

I failed to mention in the first post that I explicitly set the `build_path` to a new value before attempting to build.

Here's my shell history immediately after `pip install --upgrade esphome` and trying to build:

```shell
# This is a build attempt before cleaning....
Archiving .pioenvs/remote-bt-01/libFrameworkArduino.a
Indexing .pioenvs/remote-bt-01/libFrameworkArduino.a
Linking .pioenvs/remote-bt-01/firmware.elf
/home/karl/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/remote-bt-01/src/esphome/components/ethernet/ethernet_component.cpp.o:(.literal._ZN7esphome8ethernet17EthernetComponent5setupEv+0x44): undefined reference to `esp_eth_phy_new_jl1101'
/home/karl/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/remote-bt-01/src/esphome/components/ethernet/ethernet_component.cpp.o: in function `esphome::ethernet::EthernetComponent::setup()':
/home/karl/Projects/esphome/hardware/build-root/remote-bt-01/src/esphome/components/ethernet/ethernet_component.cpp:100: undefined reference to `esp_eth_phy_new_jl1101'
collect2: error: ld returned 1 exit status
*** [.pioenvs/remote-bt-01/firmware.elf] Error 1
======================================================================= [FAILED] Took 12.17 seconds =======================================================================
❯ rm -rf /home/karl/.platformio/packages/toolchain-xtensa-esp32*
❯ esphome clean projects/ha_bt_proxy.yaml
INFO Reading configuration projects/ha_bt_proxy.yaml...
INFO Deleting projects/../build-root/remote-bt-01/.pioenvs
INFO Deleting projects/../build-root/remote-bt-01/.piolibdeps
INFO Done!
```

Here's the build post `clean`:

```shell
❯ esphome -s version_hash $(git rev-parse --short HEAD) run projects/ha_bt_proxy.yaml
INFO Reading configuration projects/ha_bt_proxy.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing remote-bt-01 (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
INFO Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa-esp32@8.4.0+2021r2-patch3 has been installed!
INFO toolchain-xtensa-esp32@8.4.0+2021r2-patch3 has been installed!
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/remote-bt-01/src/esphome/components/api/api_connection.cpp.o
[... 138 lines cut ...]
Archiving .pioenvs/remote-bt-01/lib98e/libESPAsyncWebServer-esphome.a
Indexing .pioenvs/remote-bt-01/lib98e/libESPAsyncWebServer-esphome.a
Compiling .pioenvs/remote-bt-01/FrameworkArduino/esp32-hal-sigmadelta.c.o
Compiling .pioenvs/remote-bt-01/FrameworkArduino/esp32-hal-spi.c.o
Compiling .pioenvs/remote-bt-01/FrameworkArduino/esp32-hal-time.c.o
Compiling .pioenvs/remote-bt-01/FrameworkArduino/esp32-hal-timer.c.o
Compiling .pioenvs/remote-bt-01/FrameworkArduino/esp32-hal-tinyusb.c.o
Compiling .pioenvs/remote-bt-01/FrameworkArduino/esp32-hal-touch.c.o
Compiling .pioenvs/remote-bt-01/FrameworkArduino/esp32-hal-uart.c.o
Compiling .pioenvs/remote-bt-01/FrameworkArduino/firmware_msc_fat.c.o
Compiling .pioenvs/remote-bt-01/FrameworkArduino/libb64/cdecode.c.o
Compiling .pioenvs/remote-bt-01/FrameworkArduino/libb64/cencode.c.o
Compiling .pioenvs/remote-bt-01/FrameworkArduino/main.cpp.o
Compiling .pioenvs/remote-bt-01/FrameworkArduino/stdlib_noniso.c.o
Compiling .pioenvs/remote-bt-01/FrameworkArduino/wiring_pulse.c.o
Compiling .pioenvs/remote-bt-01/FrameworkArduino/wiring_shift.c.o
Archiving .pioenvs/remote-bt-01/libFrameworkArduino.a
Indexing .pioenvs/remote-bt-01/libFrameworkArduino.a
Linking .pioenvs/remote-bt-01/firmware.elf
[... cut ...]
/home/karl/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/remote-bt-01/src/esphome/components/ethernet/ethernet_component.cpp.o: in function `esphome::ethernet::EthernetComponent::setup()':
/home/karl/Projects/esphome/hardware/build-root/remote-bt-01/src/esphome/components/ethernet/ethernet_component.cpp:100: undefined reference to `esp_eth_phy_new_jl1101'
collect2: error: ld returned 1 exit status
*** [.pioenvs/remote-bt-01/firmware.elf] Error 1
```

nagyrobi, 2023-02-16:
Same here on WT32-ETH01, tried multiple devices.

Discord thread: https://discord.com/channels/429907082951524364/1075781680591286325

ssieb, 2023-02-16:
@kquinsland , what kind of system are you running it on?

kquinsland, 2023-02-17:
> @kquinsland , what kind of system are you running it on?

LILYGO POE ESP32 module.

ssieb, 2023-02-17:
I meant where are you running esphome?  What kind of system are you using for HA?

syssi, 2023-02-17:
In my case it's a simple python3.9 install (`pip3 install esphome`) on a amd64 based Debian Bullseye host.

StefanTT, 2023-02-17:
Same here. Native installation on Arch, running on the commandline.
Cleaning the build files does not help. Even tried deleting ~/.platformio

ssieb, 2023-02-17:
Ok, it turns out the pypi package is somehow missing that file.  So those of us that use the repo directly don't see it.
As a workaround, you can use the esphome repo as an external component.
```yaml
external_components:
  - source: github://esphome/esphome@dev
    components: [ ethernet ]
```

StefanTT, 2023-02-17:
Yep, that's it. Adding the external component fixes the problem for me. Thank you for investigating!

danielduerr, 2023-02-17:
@ssieb nice! I can confirm this works as well on a macOS install of `esphome` via `brew`.

kquinsland, 2023-02-17:
> I meant where are you running esphome? What kind of system are you using for HA?

Ah, I misunderstood.  

I run HA on a k8s cluster.  All my ESPHome builds take place inside of a container (CI/CD pipe) or my desktop... both linux based. 
For simplicity, I am showing things as they are on my desktop rather than from the CI/CD pipe which has some overhead.

In all cases, esphome is installed via `pip` in a `venv`

kpfleming, 2023-02-18:
Just ran into this myself (pip-installed esphome, building for wESP32) and will either put in this workaround or wait for 2023.2.3 :-)

UngluedChalice, 2023-02-18:
> pip-installed esphome

How do you do the workaround for this? The way the workaround looks to me as a super noob is a Home Assistant configuration file, but I have ESPHome installed via the command line on my Mac.

syssi, 2023-02-18:
In any case you have a YAML configuration with your ESPHome node configuration. To apply the workaround just extend this YAML file.

UngluedChalice, 2023-02-18:
Oh, I understand. Those lines go in the YAML file that I have for each ESP32 device. I was thinking there was some general configuration file for the ESPHome package somewhere. Sorry, just got this up and running a week ago so I'm still learning the basics. Thanks!

closed 2023-02-19 by jesserockz

kpfleming, 2023-02-19:
Confirmed, my configuration builds and works as expected with 2023.2.3. Thanks!

ssieb, 2023-03-02:
> Donde coloco esto?

You would put that in the device yaml, but you don't need to now.  Just update esphome to the latest version.

lexcorp, 2023-03-02:
> > Donde coloco esto?
> 
> Pondrías eso en el dispositivo yaml, pero no es necesario ahora. Simplemente actualice esphome a la última versión.

Tengo la última version d esphome, la última version de de homeassistant y no funciono...

Home Assistant 2023.3.0
Supervisor 2023.02.dev2801
Frontend 20230301.0 - latest

ESPHome
Current version: 2023.2.4

ssieb, 2023-03-02:
What happens?

lexcorp, 2023-03-02:
Intento instalar mi esp32, la hacer clic en instalar desde la interfaz de ESPHOME dentro de homeassistant:

INFO Reading configuration /config/esphome/tutum.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing tutum (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/tutum/.pioenvs/tutum/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/tutum/.pioenvs/tutum/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/tutum/.pioenvs/tutum/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/tutum/.pioenvs/tutum/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling /data/tutum/.pioenvs/tutum/src/esphome/components/api/api_pb2_service.cpp.o
*** [/data/tutum/.pioenvs/tutum/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [/data/tutum/.pioenvs/tutum/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/tutum/.pioenvs/tutum/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
========================== [FAILED] Took 5.18 seconds ==========================

kpfleming, 2023-03-02:
That is a different problem entirely.