esphome/issues #4175: Cannot access ESPHome UI since upgrade to 2023.2.0: 401: Unauthorized

ethzero, 2023-02-15:
### The problem

Allowed ESPHome to update to 2023.2.0 and unfortunately since then I cannot access it's management UI either from the Addon UI or Sidebar, now being presented with a "401: Unauthorized" error.

The ESPHome doesn't appear to throw any obvious start-up or run-time error and I frankly don't have the time/expertise to dig any further. I've tried uninstalling/re-installing but to no avail however the individual devices are prompting me for a firmware upgrade, FWIW.

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting
-----------------------------------------------------------
 Add-on: ESPHome
 ESPHome add-on for intelligently managing all your ESP8266/ESP32 devices
-----------------------------------------------------------
 Add-on version: 2023.2.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 9.5  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2023.2.4
 Home Assistant Supervisor: 2023.01.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[20:24:34] INFO: Configuring timezone (Europe/London)...
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service esphome: starting
s6-rc: info: service esphome successfully started
s6-rc: info: service init-nginx successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
[20:24:36] INFO: Waiting for ESPHome dashboard to come up...
s6-rc: info: service discovery: starting
[20:24:36] INFO: Starting ESPHome dashboard...
2023-02-15 20:24:37,431 INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
[20:24:37] INFO: Starting NGINX...
2023-02-15 20:24:38,523 INFO 200 GET /devices (0.0.0.0) 4.63ms
[20:24:38] INFO: Successfully send discovery information to Home Assistant.
s6-rc: info: service discovery successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
2023-02-15 20:29:39,146 INFO 200 GET /devices (0.0.0.0) 4.02ms
2023-02-15 20:34:39,147 INFO 200 GET /devices (0.0.0.0) 3.68ms
 
```

### Which version of ESPHome has the issue?

2023.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.4

### What platform are you using?

ESP8266

ethzero, 2023-02-15:
*sigh*. So apparently I needed to manually go in to Chrome's Dev Tools > Application > Storage > and "Clear site data", "including third-party cookies". Obviously this is quite the nasty workaround, but I can now re-access the ESPHome device management UI.

PS: I'd also tried prior rebooting HA a few times and disabling all Chrome extension to no avail.

jesserockz, 2023-02-16:
COnsidering the error you found was cookie related and you see
```
WARNING (MainThread) [supervisor.api.ingress] No valid ingress session None
```

This is an issue for Home Assistant / Supervisor, and not related to ESPHome exactly.

closed 2023-02-16 by jesserockz