esphome/issues #4176: pulse_meter stopped working after updating to 2023.2.0

Rvh91, 2023-02-16:
### The problem

My water meter that uses the `pulse_meter` platform has stopped working after updating to 2023.2.0. No pulses are ever reported in home assistant or in the logs of  the device.

### Which version of ESPHome has the issue?

2023.2.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

WT32-ETH01 ESP32

### Component causing the issue

pulse_meter

### Example YAML snippet

```yaml
I'm using the following YAML

  - platform: pulse_meter
    pin: GPIO39
    name: "Water Pulse Meter"
    unit_of_measurement: "liter/min"
    icon: "mdi:water"
    internal_filter: 700ms
    internal_filter_mode: pulse
    total:
      name: "Water Meter Total"
      unit_of_measurement: "m³"
      id: water_meter_total
      accuracy_decimals: 3
      device_class: water
      state_class: total_increasing
      filters:
        - multiply: 0.001
```
```

hofvanta2, 2023-02-20:
I have also a pulse_counter has stopped after 2023.2.0 ESP32 esp32dev board from az-delivery. Was working before without any issues.

Rvh91, 2023-02-20:
Today, I have resolved my issue by reducing the filter time from 700ms to 300ms. To me it is a bit unclear as to why this was needed, since i'm pretty the pulses from my watermeter are always more than 700ms apart. But perhaps it also works for you?

hofvanta2, 2023-02-20:
I using it to measure my energy via s0 signaling. It worked before and since 2023.2.x it doesn't work anymore.

jcamer, 2023-02-21:
I also use the pulse meter on my water meter and it too stopped reporting pulses after the update.  Is there a way to downgrade or revert the update?

Rvh91, 2023-02-21:
> I also use the pulse meter on my water meter and it too stopped reporting pulses after the update. Is there a way to downgrade or revert the update?

only if you have a backup i think. What kind of filtering did you use for the watermeter?

jcamer, 2023-02-21:
> > I also use the pulse meter on my water meter and it too stopped reporting pulses after the update. Is there a way to downgrade or revert the update?
> 
> only if you have a backup i think. What kind of filtering did you use for the watermeter?

No filtering.  I'm not sure why I didn't think of the backups.  I restored to version 2022.12.8 and then updated all devices to that version.  We'll see how it goes.

Here is my yaml for the meter:

 - platform: pulse_counter
    pin: 21
    id: water_usage
    name: "Water Usage Flow Rate (GPM)"
    unit_of_measurement: "Gallons/minute gpm"
  - platform: integration
    name: "Total House Water Usage"
    unit_of_measurement: "Gallons"
    sensor: water_usage
    time_unit: min

hofvanta2, 2023-02-21:
I went back to 2022.12.8 also (restore backup, rebooted, otherwise got errors while compiling) but didn't solve the issue.
Returned back to 2023.2.3. Will try new esp32 to see if this solves the issue to be sure that its not HW related, but other pulse_counter on other PIN is working normally. Only difference is the working PIN is connected to slow pulse NPN (water meter), while the not working is connected to energy kWh meter 1000imp/kWh for car charging. A internal filter: 13us (should be default for esp32) doesn't solve the issue.

jcamer, 2023-02-21:
> I went back to 2022.12.8 also (restore backup, rebooted, otherwise got errors while compiling) but didn't solve the issue. Returned back to 2023.2.3. Will try new esp32 to see if this solves the [...]

Same, no update after restoring from backup.  I also went back to 2023.2.3.  It's odd because over this last weekend it showed that overnight I used 67,000 gallons of water.  Hasn't reported anything since and nothing has changed other than updating esphome.  The timing to me just seems "off".  I have a dallas sensor on this same esp32 and it has worked fine the entire time.

[image]

jcamer, 2023-02-23:
I can't explain it, but things started working last night.  I DID install the latest update that showed up, maybe that had something to do with it.  2023.2.4

hofvanta2, 2023-02-27:
Same issue with HW replacement. Seems software related. Some binary has the same issues.