esphome/issues #4177: RGB lights crashing  after upgrade to 2023.2.1

alfwro13, 2023-02-16:
### The problem

After updating firmware in my ESPHome rgb lights the are crashing every couple of minutes with this error:

`INFO Reading configuration /config/esphome/rgb-light-office.yaml...
INFO Starting log output from 192.168.1.17 using esphome API
WARNING 192.168.1.17: Connection error occurred: Timeout waiting for response for <class 'api_pb2.ConnectRequest'> after 10.0s
WARNING Can't connect to ESPHome API for 192.168.1.17: Timeout waiting for response for <class 'api_pb2.ConnectRequest'> after 10.0s
INFO Trying to reconnect to 192.168.1.17 in the background
INFO Successfully connected to 192.168.1.17
[08:56:32][I][app:102]: ESPHome version 2023.2.1 compiled on Feb 16 2023, 08:22:53
[09:01:20][I][ota:113]: Boot seems successful, resetting boot loop counter.
WARNING 192.168.1.17: Connection error occurred: 
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 520, in send_message_await_response_complex
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 520, in send_message_await_response_complex
    await fut
  File "/usr/local/lib/python3.9/dist-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.9/dist-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 303, in _keep_alive_loop
    await self._ping()
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 628, in _ping
    await self.send_message_await_response(PingRequest(), PingResponse)
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 537, in send_message_await_response
    res = await self.send_message_await_response_complex(
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 522, in send_message_await_response_complex
    raise TimeoutAPIError(
aioesphomeapi.core.TimeoutAPIError: Timeout waiting for response for <class 'api_pb2.PingRequest'> after 10.0s
INFO Disconnected from ESPHome API for 192.168.1.17
WARNING Disconnected from API
INFO Successfully connected to 192.168.1.17`

Prior to the upgrade they were working without issues (on December firmware 2022.12)

### Which version of ESPHome has the issue?

v2023.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP8266

### Board

esp01_1m

### Example YAML snippet

```yaml
esphome:
  name: rgb-light-office

esp8266:
  board: esp01_1m

# Enable logging
logger:
  level: INFO

# Enable Home Assistant API
api:
<removed>

ota:
<removed>

wifi:
<removed>

captive_portal:

output:
  - platform: esp8266_pwm
    id: output_green
    pin: GPIO14
    max_power: 80%
    frequency: 10 kHz
  - platform: esp8266_pwm
    id: output_red
    pin: GPIO5
    max_power: 80%
    frequency: 10 kHz
  - platform: esp8266_pwm
    id: output_blue
    pin: GPIO12
    max_power: 80%
    frequency: 10 kHz
  - platform: esp8266_pwm
    id: output_white
    pin: GPIO15
    max_power: 80%
    frequency: 10 kHz

globals:
  - id: action_state
    type: int
    restore_value: no
    initial_value: '0'

binary_sensor:
  - platform: gpio
    pin:   GPIO13
    name: "RGBW Controller Button"
    internal: true
    filters:
      - invert:
      - delayed_on_off: 20ms
    on_press:
      then:
        - lambda: id(action_state) = (id(action_state) + 1) % 5;
        - if:
            condition:
              lambda: 'return id(action_state) == 0;'
            then:
              - light.turn_off:  office_rgbw_lightstrip
        - if:
            condition:
              lambda: 'return id(action_state) == 1;'
            then:
              - light.turn_on:
                  id: office_rgbw_lightstrip
                  brightness: 100%
                  color_brightness: 80%
                  red: 0%
                  green: 0%
                  blue: 0%
                  white: 80% 
        - if:
            condition:
              lambda: 'return id(action_state) == 2;'
            then:
              - light.turn_on:
                  id: office_rgbw_lightstrip
                  brightness: 100%
                  red: 0%
                  green: 100%
                  blue: 0%
                  white: 0%
        - if:
            condition:
              lambda: 'return id(action_state) == 3;'
            then:
              - light.turn_on:
                  id: office_rgbw_lightstrip
                  brightness: 100%
                  red: 100%
                  green: 0%
                  blue: 0%
                  white: 0% 
        - if:
            condition:
              lambda: 'return id(action_state) == 4;'
            then:
              - light.turn_on:
                  id: office_rgbw_lightstrip
                  brightness: 100%
                  red: 0%
                  green: 0%
                  blue: 100%
                  white: 0% 
light:
  - platform: rgbw
    name: "Office RGB Light Strip"
    id: office_rgbw_lightstrip
    red: output_red
    green: output_green
    blue: output_blue
    white: output_white
    restore_mode: ALWAYS_OFF
```

ldeluney, 2023-02-16:
I am seeing the exact same error after updating my Sonoff Basics (8266). Log attached. I attempted to reflash the device with the same version with no change. I have one basic currently in a bootloop.

> INFO Successfully connected to extension-cord.local
>WARNING extension-cord.local: Connection error occurred: 
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 520, in send_message_await_response_complex
    await fut
asyncio.exceptions.CancelledError
>
>During handling of the above exception, another exception occurred:
>Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 520, in send_message_await_response_complex
    await fut
  File "/usr/local/lib/python3.9/dist-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.9/dist-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
>
>The above exception was the direct cause of the following exception:...

ldeluney, 2023-02-18:
> > I have one basic currently in a bootloop.
> 
> How do you know it's a bootloop? That's just the logs disconnecting. Do you have an uptime sensor?

Well, to be fair, I have no idea exactly what it is doing. It loads (I think because the led starts blinking), and randomly I will see the fallback AP for less than a minute. If I push the button it will toggle the relay, but will reset after a boot cycle within two minutes. 

It almost looks like a race condition when connecting to the WiFi, causing it to crash and reboot.

JayNewstrom, 2023-02-20:
I'm seeing this issue as well. Here are my logs from the ESP32:

<details>
  <summary>Logs</summary>
  
```
INFO Successfully connected to office_desk.local
[16:37:50][W][api.connection:083]: Home Assistant 2023.2.5 (::FFFF:192.168.2.210): Connection closed
WARNING office_desk.local: Connection error occurred: 
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 520, in send_message_await_response_complex
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 520, in send_message_await_response_complex
    await fut
  File "/usr/local/lib/python3.9/dist-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.9/dist-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 303, in _keep_alive_loop
    await self._ping()
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 628, in _ping
    await self.send_message_await_response(PingRequest(), PingResponse)
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 537, in send_message_await_response
    res = await self.send_message_await_response_complex(
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 522, in send_message_await_response_complex
    raise TimeoutAPIError(
aioesphomeapi.core.TimeoutAPIError: Timeout waiting for response for <class 'api_pb2.PingRequest'> after 10.0s
INFO Disconnected from ESPHome API for office_desk.local
WARNING Disconnected from API
INFO Successfully connected to office_desk.local
WARNING office_desk.local: Connection error occurred: 
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 520, in send_message_await_response_complex
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 520, in send_message_await_response_complex
    await fut
  File "/usr/local/lib/python3.9/dist-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.9/dist-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 303, in _keep_alive_loop
    await self._ping()
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 628, in _ping
    await self.send_message_await_response(PingRequest(), PingResponse)
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 537, in send_message_await_response
    res = await self.send_message_await_response_complex(
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 522, in send_message_await_response_complex
    raise TimeoutAPIError(
aioesphomeapi.core.TimeoutAPIError: Timeout waiting for response for <class 'api_pb2.PingRequest'> after 10.0s
INFO Disconnected from ESPHome API for office_desk.local
[... 80 lines cut ...]
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 520, in send_message_await_response_complex
    await fut
  File "/usr/local/lib/python3.9/dist-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.9/dist-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 303, in _keep_alive_loop
    await self._ping()
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 628, in _ping
    await self.send_message_await_response(PingRequest(), PingResponse)
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 537, in send_message_await_response
    res = await self.send_message_await_response_complex(
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 522, in send_message_await_response_complex
    raise TimeoutAPIError(
aioesphomeapi.core.TimeoutAPIError: Timeout waiting for response for <class 'api_pb2.PingRequest'> after 10.0s
INFO Disconnected from ESPHome API for office_desk.local
WARNING Disconnected from API
INFO Successfully connected to office_desk.local
  ```
</details>

rognixon, 2023-02-23:
I'm also seeing this on an ESP32, for me it's an intermittent issue, my node can run OK for hours, then raise this error and lose connection for a minute or so.

- ESPHome v 2023.2.3
- Log trace is identical to those posted above.  
- Uptime and connection status sensors suggest that it's not rebooting, just losing connection then reconnecting.
- It's a newly developed project, so I don't know if this is connected to a particular upgrade or not.

rognixon, 2023-02-23:
> I'm also seeing this on an ESP32, for me it's an intermittent issue, my node can run OK for hours, then raise this error and lose connection for a minute or so.

Digging a little deeper, I'm seeing brief random disconnections on a lot of my nodes at the same times, and they coincide with periods of connectivity errors reported in the HA logs for lots of intergrations, not just ESPHome.  So I think maybe what I'm seeing is network or HA issues, rather than specifically ESPHome.  Maybe this set of logs in the ESPHome nodes is just what they report when they can't connect to HA.

andriej, 2023-02-23:
I'm seeing same set of errors:
```
Logger: aioesphomeapi.connection
Source: /usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py:577
First occurred: 12:36:55 (12 occurrences)
Last logged: 16:24:22

nodemcu_a @ 192.168.x.78: Connection error occurred:
sms-gateway @ 192.168.x.50: Connection error occurred:
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 537, in send_message_await_response_complex
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 536, in send_message_await_response_complex
    async with async_timeout.timeout(timeout):
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 320, in _keep_alive_loop
    await self._ping()
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 645, in _ping
    await self.send_message_await_response(PingRequest(), PingResponse)
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 554, in send_message_await_response
    res = await self.send_message_await_response_complex(
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 539, in send_message_await_response_complex
    raise TimeoutAPIError(
aioesphomeapi.core.TimeoutAPIError: Timeout waiting for response for <class 'api_pb2.PingRequest'> after 10.0s
```

One of thesse nodes operates on -78 dBm sgignal which is not the point of disconnection for other nodes.
The other one sits at around -45/-50 which is perfect signal and still I'm experiencing kind-of dropouts and even reboots without any clues in log file...
[image]

sweetpants, 2023-03-04:
I think this is caused by the api connection being dropped. My TAG reader (ESP32) can run for hours and all of a sudden it looses connecting with HA. Also if logging over WiFi, it's all verys slow. In the ESP log i see this when doing nothing

[D][sensor:127]: 'Tag Reader WiFi Channel': Sending state 13.00000  with 0 decimals of accuracy
[D][sensor:127]: 'Tag Reader WiFi Channel': Sending state 13.00000  with 0 decimals of accuracy
**[W][api.connection:083]: Home Assistant 2023.3.1 (::FFFF:192.168.24.250): Connection closed**
[D][api:102]: Accepted ::FFFF:192.168.24.250
**[D][api.connection:918]: Home Assistant 2023.3.1 (::FFFF:192.168.24.250): Connected successfully**
[D][time:045]: Synchronized time: 2023-03-04 16:40:48
[D][sensor:127]: 'Tag Reader WiFi Channel': Sending state 13.00000  with 0 decimals of accuracy
[D][sensor:127]: 'Tag Reader WiFi Channel': Sending state 13.00000  with 0 decimals of accuracy
[D][pn532:283]: Mifare ultralight
[D][pn532:162]: Found new tag '04-FD-D0-62-12-68-80'
[D][pn532:295]: Waiting to read next tag
[D][rtttl:038]: Playing song beep
[D][ledc.output:051]: Calculating resolution bit-depth for frequency 988.000000
[D][ledc.output:056]: Resolution calculated as 16
[D][rtttl:094]: Playback finished
[D][pn532:283]: Mifare ultralight
[D][pn532:162]: Found new tag '04-FD-D0-62-12-68-80'
[D][pn532:295]: Waiting to read next tag
[D][rtttl:038]: Playing song beep
[D][ledc.output:051]: Calculating resolution bit-depth for frequency 988.000000
[D][ledc.output:056]: Resolution calculated as 16
[D][rtttl:094]: Playback finished
[D][pn532:283]: Mifare ultralight
[D][pn532:162]: Found new tag '04-FD-D0-62-12-68-80'
[D][pn532:295]: Waiting to read next tag
[D][rtttl:038]: Playing song beep
[D][ledc.output:051]: Calculating resolution bit-depth for frequency 988.000000
[D][ledc.output:056]: Resolution calculated as 16
[D][rtttl:094]: Playback finished
[D][pn532:283]: Mifare ultralight
[D][pn532:162]: Found new tag '04-FD-D0-62-12-68-80'
[D][pn532:295]: Waiting to read next tag
[D][rtttl:038]: Playing song beep
[D][ledc.output:051]: Calculating resolution bit-depth for frequency 988.000000
[D][ledc.output:056]: Resolution calculated as 16
[D][rtttl:094]: Playback finished
[D][sensor:127]: 'Tag Reader WiFi Sensor': Sending state -41.00000 dBm with 0 decimals of accuracy
[D][sensor:127]: 'Tag Reader WiFi Channel': Sending state 13.00000  with 0 decimals of accuracy
[I][ota:113]: Boot seems successful, resetting boot loop counter.
[D][esp32.preferences:113]: Saving 1 preferences to flash...
[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[D][sensor:127]: 'Tag Reader WiFi Channel': Sending state 13.00000  with 0 decimals of accuracy
[D][sensor:127]: 'Tag Reader WiFi Channel': Sending state 13.00000  with 0 decimals of accuracy
[D][sensor:127]: 'Tag Reader WiFi Channel': Sending state 13.00000  with 0 decimals of accuracy
[D][sensor:127]: 'Tag Reader WiFi Channel': Sending state 13.00000  with 0 decimals of accuracy
**[W][api.connection:083]: Home Assistant 2023.3.1 (::FFFF:192.168.24.250): Connection closed**
[D][api:102]: Accepted ::FFFF:192.168.24.250
**[D][api.connection:918]: Home Assistant 2023.3.1 (::FFFF:192.168.24.250): Connected successfully**
[D][time:045]: Synchronized time: 2023-03-04 16:47:50
[D][sensor:127]: 'Tag Reader WiFi Sensor': Sending state -43.00000 dBm with 0 decimals of accuracy
[D][sensor:127]: 'Tag Reader WiFi Channel': Sending state 13.00000  with 0 decimals of accuracy
[D][sensor:127]: 'Tag Reader WiFi Channel': Sending state 13.00000  with 0 decimals of accuracy
**[W][api.connection:083]: Home Assistant 2023.3.1 (::FFFF:192.168.24.250): Connection closed**
[D][api:102]: Accepted ::FFFF:192.168.24.250
**[D][api.connection:918]: Home Assistant 2023.3.1 (::FFFF:192.168.24.250): Connected successfully**
[D][time:045]: Synchronized time: 2023-03-04 16:50:12
[D][sensor:127]: 'Tag Reader WiFi Channel': Sending state 13.00000  with 0 decimals of accuracy
[D][sntp:078]: Synchronized time: 2023-03-04 16:51:19
[D][sensor:127]: 'Tag Reader WiFi Channel': Sending state 13.00000  with 0 decimals of accuracy
[D][sensor:127]: 'Tag Reader WiFi Channel': Sending state 13.00000  with 0 decimals of accuracy
[D][sensor:127]: 'Tag Reader WiFi Sensor': Sending state -44.00000 dBm with 0 decimals of accuracy

Also when sometimes scanning a TAG, you have to hold the TAG for several seconds on the scanner. In the ESP log, you see the scanning directly, but is HA log (AppDaemon event) it can arrive seonds later

I used to have the RGB config in there, but removed it for the time beeing.

Just my 2cents, Harry

sweetpants, 2023-03-05:
Hmmm, because of a partial power outage last night, a few of my systems needed a reboot. Since the reboot the tag-reader responds much better. I will monitor this for a while to see if i have api timouts again.

sweetpants, 2023-03-05:
Oop, too soon, still have timeout errors

[D][rtttl:094]: Playback finished
[D][pn532:280]: Mifare classic
[D][pn532:162]: Found new tag 'F3-EC-5F-EF'
[D][pn532:166]:   NDEF formatted records:
[D][pn532:168]:      -
[D][pn532:295]: Waiting to read next tag
**[W][api.connection:083]: Home Assistant 2023.3.1 (::FFFF:192.168.24.250): Connection closed**
[D][api:102]: Accepted ::FFFF:192.168.24.250
**[D][api.connection:918]: Home Assistant 2023.3.1 (::FFFF:192.168.24.250): Connected successfully
[W][api.connection:083]: Home Assistant 2023.3.1 (::FFFF:192.168.24.250): Connection closed**
[D][api:102]: Accepted ::FFFF:192.168.24.250
**[D][api.connection:918]: Home Assistant 2023.3.1 (::FFFF:192.168.24.250): Connected successfully**
[D][time:045]: Synchronized time: 2023-03-05 08:02:18
[D][pn532:280]: Mifare classic
[D][pn532:162]: Found new tag 'F3-EC-5F-EF'
[D][pn532:166]:   NDEF formatted records:
[D][pn532:168]:      -
[D][pn532:295]: Waiting to read next tag
[D][rtttl:038]: Playing song two short
[D][ledc.output:051]: Calculating resolution bit-depth for frequency 1319.000000
[D][ledc.output:056]: Resolution calculated as 15
[D][ledc.output:051]: Calculating resolution bit-depth for frequency 1319.000000
[D][ledc.output:056]: Resolution calculated as 15
[D][rtttl:094]: Playback finished

kiselmanj, 2023-03-05:
I used a wifi signal amplifier and everything worked, apparently it refuses to work with a weak signal, you need a very strong wifi signal. I had the exact same problem before after an update.

andriej, 2023-03-05:
I have issues with units almost near to AP