esphome/issues #4183: VL53L0X hangs while waiting for interrupt

troelde, 2023-02-16:
### The problem

The sensor works fine for distances within its max range. But with every configuration I have tried including those here https://www.esphome.io/components/sensor/vl53l0x.html, the sensor often runs into a state waiting for interrupt often when distance is out of range. The log says: Warning in log: "_update called before prior reading complete - initiated:0 waiting_for_interrupt:1"_.  It is impossible to recover without rebooting device. 

The sensor state is set to unknown, the same as if distance is out of range. Therefore impossible to detect if sensor actually has stopped working or distance out of range. 

### Which version of ESPHome has the issue?

2023.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini compatible

### Component causing the issue

VL53L0X Time Of Flight Distance Sensor

### Anything in the logs that might be useful for us?

```txt
From log:
[18:25:05][I][app:102]: ESPHome version 2023.2.1 compiled on Feb 16 2023, 18:21:20
[18:25:05][C][wifi:504]: WiFi:
[18:25:05][C][wifi:362]:   Local MAC: 48:55:19:C8:9D:50
[18:25:05][C][wifi:363]:   SSID: 'Telenor6617tog'[redacted]
[18:25:05][C][wifi:364]:   IP Address: 10.0.0.6
[18:25:05][C][wifi:365]:   BSSID: 90:F8:91:ED:62:80[redacted]
[18:25:05][C][wifi:367]:   Hostname: 'garasjeport-20c-distanse'
[18:25:05][C][wifi:369]:   Signal strength: -53 dB ▂▄▆█
[18:25:05][C][wifi:373]:   Channel: 1
[18:25:05][C][wifi:374]:   Subnet: 255.255.255.0
[18:25:05][C][wifi:375]:   Gateway: 10.0.0.138
[18:25:05][C][wifi:376]:   DNS1: 148.122.164.253
[18:25:05][C][wifi:377]:   DNS2: 148.122.16.253
[18:25:06][C][logger:293]: Logger:
[18:25:06][C][logger:294]:   Level: DEBUG
[18:25:06][C][logger:295]:   Log Baud Rate: 115200
[18:25:06][C][logger:296]:   Hardware UART: UART0
[18:25:06][C][i2c.arduino:052]: I2C Bus:
[18:25:06][C][i2c.arduino:053]:   SDA Pin: GPIO4
[18:25:06][C][i2c.arduino:054]:   SCL Pin: GPIO5
[18:25:06][C][i2c.arduino:055]:   Frequency: 50000 Hz
[18:25:06][C][i2c.arduino:058]:   Recovery: bus successfully recovered
[18:25:06][I][i2c.arduino:068]: Results from i2c bus scan:
[18:25:06][I][i2c.arduino:074]: Found i2c device at address 0x29
[18:25:06][C][template.cover:071]: Template Cover 'Garasjeport 20C'
[18:25:06][C][template.cover:071]:   Device Class: 'garage'
[18:25:06][C][vl53l0x:024]: VL53L0X 'Distansemåler garasjeport 20C'
[18:25:06][C][vl53l0x:024]:   State Class: 'measurement'
[18:25:06][C][vl53l0x:024]:   Unit of Measurement: 'm'
[18:25:06][C][vl53l0x:024]:   Accuracy Decimals: 2
[18:25:06][C][vl53l0x:024]:   Icon: 'mdi:arrow-expand-vertical'
[18:25:06][C][vl53l0x:025]:   Update Interval: 0.5s
[18:25:06][C][vl53l0x:026]:   Address: 0x29
[18:25:06][C][vl53l0x:030]:   Timeout: 200us
[18:25:06][C][restart:076]: Restart Switch 'Omstart garasjeport 20C'
[18:25:06][C][restart:078]:   Icon: 'mdi:restart'
[18:25:06][C][restart:098]:   Restore Mode: restore defaults to OFF
[18:25:06][C][captive_portal:088]: Captive Portal:
[18:25:06][C][mdns:108]: mDNS:
[18:25:06][C][mdns:109]:   Hostname: garasjeport-20c-distanse
[18:25:06][C][ota:093]: Over-The-Air Updates:
[18:25:06][C][ota:094]:   Address: garasjeport-20c-distanse.local:8266
[18:25:06][C][ota:097]:   Using Password.
[18:25:06][D][sensor:126]: 'Distansemåler garasjeport 20C': Sending state nan m with 2 decimals of accuracy
[18:25:06][W][vl53l0x:265]: Distansemåler garasjeport 20C - update called before prior reading complete - initiated:0 waiting_for_interrupt:1
[18:25:06][C][api:138]: API Server:
[18:25:06][C][api:139]:   Address: garasjeport-20c-distanse.local:6053
[18:25:06][C][api:141]:   Using noise encryption: YES
[18:25:06][D][sensor:126]: 'Distansemåler garasjeport 20C': Sending state nan m with 2 decimals of accuracy
[18:25:06][W][vl53l0x:265]: Distansemåler garasjeport 20C - update called before prior reading complete - initiated:0 waiting_for_interrupt:1
[18:25:07][D][api:102]: Accepted 10.0.0.38
[18:25:07][W][api.connection:083]: Home Assistant 2023.2.5 (10.0.0.38): Connection closed
[18:25:07][D][sensor:126]: 'Distansemåler garasjeport 20C': Sending state nan m with 2 decimals of accuracy
[18:25:07][W][vl53l0x:265]: Distansemåler garasjeport 20C - update called before prior reading complete - initiated:0 waiting_for_interrupt:1
[18:25:07][D][api.connection:918]: Home Assistant 2023.2.5 (10.0.0.38): Connected successfully
[18:25:07][D][sensor:126]: 'Distansemåler garasjeport 20C': Sending state nan m with 2 decimals of accuracy
[18:25:07][W][vl53l0x:265]: Distansemåler garasjeport 20C - update called before prior reading complete - initiated:0 waiting_for_interrupt:1
[18:25:07][D][api:102]: Accepted 10.0.0.38
[18:25:09][W][api.connection:083]: Home Assistant 2023.2.5 (10.0.0.38): Connection closed
[18:25:09][D][sensor:126]: 'Distansemåler garasjeport 20C': Sending state nan m with 2 decimals of accuracy
[18:25:09][W][vl53l0x:265]: Distansemåler garasjeport 20C - update called before prior reading complete - initiated:0 waiting_for_interrupt:1
```

pruwait, 2023-05-08:
same problem

[vl53l0x:265]: Sensor 1 - update called before prior reading complete - initiated:0 waiting_for_interrupt:1

wr420, 2023-05-13:
Also having the same problem.  Is there a way it can ignore the bad readings and continue once the condition clears?

`ccuracy
[13:25:19][W][vl53l0x:266]: VL53L0x Bottom - update called before prior reading complete - initiated:0 waiting_for_interrupt:1
[13:25:19][D][sensor:110]: 'VL53L0x Bottom': Sending state nan m with 2 decimals of accuracy
[13:25:19][W][vl53l0x:266]: VL53L0x Bottom - update called before prior reading complete - initiated:0 waiting_for_interrupt:1
[13:25:19][D][sensor:110]: 'VL53L0x Bottom': Sending state nan m with 2 decimals of accuracy
`

thanks !

momonator25, 2024-01-28:
> With the help of user 'ssieb' on Discord:
> 
> on_message: level: WARN then: - lambda: if (std::string(tag) == "vl53l0x") id(restart_button).press();
> 
> It's crude, but does the trick on some of my applications.

A few month later and still the same issue with the sensor, but restarting on Warn message does the job.
I can exclude the fault on the sensor side because I ran the sensor for over a year before now i switched to ESPHome.

Gunthervc, 2024-05-26:
After all this time they not able to fix it. 
installed espeasy and all runs fine.

grambas, 2024-11-05:
I have the same issue. After restarting the device, I get some good readings, but then this error appears. I observed that if I disconnect the SLC pin, the same error also occurs. After reconnecting the SLC pin, the error persists until I restart the device.

schneich, 2025-01-13:
I just have set together a D1mini and a VL53L0X and I run into the same issue.
ESPHome Version 2024.12.2

```
[23:45:06][D][sensor:093]: 'VL53L0x Distanz': Sending state nan m with 2 decimals of accuracy
[23:45:06][W][vl53l0x:265]: VL53L0x Distanz - update called before prior reading complete - initiated:0 waiting_for_interrupt:1
```
Has this been fixed? O what version will this be included?

thank you

s1godfrey, 2025-01-21:
> With the help of user 'ssieb' on Discord:
> 
> on_message: level: WARN then: - lambda: if (std::string(tag) == "vl53l0x") id(restart_button).press();
> 
I would like to extend the method of resetting the esp device, but do not know how to parse the log messages:
-----------------------
1. [12:15:37][W][component:157]: Component vl53l0x.sensor set Warning flag: unspecified
2. [12:15:37][W][vl53l0x:265]: Doorway Sensor Dist - update called before prior reading complete - initiated:0 waiting_for_interrupt:1
3. [12:15:42][W][component:170]: Component vl53l0x.sensor cleared Warning flag
-------------------------
Objective: Set alarm (binary_sensor) to ON when message 1 occurs. Set alarm to OFF when message 3 occurs.

pseudo code:
if component vl53l0x.sensor set
  -start script alarm_timer; if >2s = reset
 
if component vl53l0x.sensor cleared
  -stop script alarm_timer

Please help if anyone has knowledge of parsing 'specific text' using message (const char *), level (int) and tag (const char *) for the on_message function.

s1godfrey, 2025-05-25:
I agree completely, however, at my limited level, the best I can do to resolve the problem (patch) is to use ingenuity to the best of my ability to overcome obstacles.  

I also acknowledge and am thankful for those core programmers whom have given their time, mostly with little to no reward, to develop the remarkable tools tinkerers like myself may often taken for granted.

TekFan, 2025-12-27:
This doesn't work unfortunately.
I have this pin hooked up and defined for years, because I have 2 sensors on the same bus and the esphome driver has to disable one sensor while configuring a different address for the other one and vice-versa.

Even with this configuration, both sensors end up crashing one after the other every day around the same time with a variation of about +/- half an hour.
Which is strange by itself, because whatever the time the sensors have been reset previously, they always crash around the same time window of the day. (around 10am in my case)
This would indicate some external influence occurring at that time, but believe me, I searched for months without finding anything happening at that moment either in HA, the Esphome device or anything around.

This behavior is still a big mystery for me.

What we would need is a method to reset the I2C driver and the sensors programmatically like it happens at boot.
Right now, like everyone, I have an automation in HA that reboot the esp when the sensors go unavailable but this also resets other sensors managed by the same ESP device which is not desirable.
A way to reset the VL530x sensors specifically would be much better in this context, but having a fix would be even better...

vks007, 2026-08-19:
For me the issue started when i joined a piece of dupont wire to an existing one to make its length longer. I removed the extra wire and the issue went away. Tried it a couple of times and I was able to reproduce it when the lenght was 2 dupont wires (20+20 cms). 
I am using an esp + VL53L0X sensor (3.3V)

TekFan, 2026-08-21:
Well, it's clear from the specs that an I2C bus should be as short as possible, in the centimeters range.
Even though I knew that when I designed my project, I still took a chance and my 2 sensors are on a 1 meter cable, but it is shielded, terminated at both sides and works at the slowest speed IC2 supports.
It's obviously still a recipe for trouble, but what I can't figure out is why it works flawlessly for 24H without any error and then crashes everyday around the same time...

TekFan, 2026-08-24:
Good news !
When I wrote my previous post I noticed the mention from @touste about his PR in a fork of the official component that implements a reset on sensor timeout.
So I gave it a try and, lo and behold, it works !!!
My node didn't crash for the last 3 days, yeah...
@touste, thank you so much, you should PR this to the official component if not already.