esphome/issues #4184: sgp30 platform generates an error on 'ppb' unit

Calimerorulez, 2023-02-16:
### The problem

`Entity sensor.airgradient_pro_tvoc (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'ppb' which is not a valid unit for the device class ('volatile_organic_compounds') it is using; expected one of ['µg/m³']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22`

### Which version of ESPHome has the issue?

2023.2.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

### Component causing the issue

sgp30

### Example YAML snippet

```yaml
- platform: sgp30
    eco2:
      id: eco2
      name: "${upper_devicename} eCO2"
      accuracy_decimals: 1
    tvoc:
      id: tvoc
      name: "${upper_devicename} TVOC"
      accuracy_decimals: 1
    store_baseline: yes
    address: 0x58
    update_interval: 1s
```

bdraco, 2023-06-21:
duplicate of https://github.com/esphome/issues/issues/4012

jamesmyatt, 2023-06-21:
Closed by https://github.com/esphome/esphome/pull/4982

closed 2023-06-22 by jesserockz