esphome/issues #4186: PCA9685 not working if i use more than one

neutrino220, 2023-02-16:
### The problem

Since esphome 2023.02 only one of my PCA9685 is working.
Both are detected at I2C but only the second one is working.
If i change the address, the other one works.
I use it for RGB Lights

### Which version of ESPHome has the issue?

2023.02

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

PCA 9686 / Light

### Example YAML snippet

```yaml
pca9685:
    - id: 'controller0'
      frequency: 1000
      address: 0x40
    
    - id: 'controller1'
      frequency: 1000
      address: 0x42

# Individual outputs
output:
  - platform: pca9685
    pca9685_id: 'controller0'
    id: "RGB0_0_R"
    channel: 0

  - platform: pca9685
    pca9685_id: 'controller0'
    id: "RGB0_0_G"
    channel: 1

  - platform: pca9685
    pca9685_id: 'controller0'
    id: "RGB0_0_B"
    channel: 2

  - platform: pca9685
    pca9685_id: 'controller0'
    id: "RGB0_1_R"
    channel: 3

  - platform: pca9685
    pca9685_id: 'controller0'
    id: "RGB0_1_G"
    channel: 4

  - platform: pca9685
    pca9685_id: 'controller0'
    id: "RGB0_1_B"
    channel: 5

  - platform: pca9685
    pca9685_id: 'controller0'
    id: "RGB0_2_R"
    channel: 6

  - platform: pca9685
    pca9685_id: 'controller0'
    id: "RGB0_2_G"
    channel: 7

  - platform: pca9685
    pca9685_id: 'controller0'
    id: "RGB0_2_B"
    channel: 8

  - platform: pca9685
    pca9685_id: 'controller0'
    id: "RGB0_3_R"
    channel: 9

  - platform: pca9685
    pca9685_id: 'controller0'
    id: "RGB0_3_G"
    channel: 10

  - platform: pca9685
    pca9685_id: 'controller0'
    id: "RGB0_3_B"
    channel: 11
  

  - platform: pca9685
    pca9685_id: 'controller0'
    id: "RGB0_4_R"
    channel: 12

  - platform: pca9685
    pca9685_id: 'controller0'
    id: "RGB0_4_G"
    channel: 13

  - platform: pca9685
    pca9685_id: 'controller0'
    id: "RGB0_4_B"
    channel: 14

################################################

  - platform: pca9685
    pca9685_id: 'controller1'
    id: "RGB1_0_R"
    channel: 0

  - platform: pca9685
    pca9685_id: 'controller1'
    id: "RGB1_0_G"
    channel: 1

  - platform: pca9685
    pca9685_id: 'controller1'
    id: "RGB1_0_B"
    channel: 2

  - platform: pca9685
    pca9685_id: 'controller1'
    id: "RGB1_1_R"
    channel: 3

  - platform: pca9685
    pca9685_id: 'controller1'
    id: "RGB1_1_G"
    channel: 4

  - platform: pca9685
    pca9685_id: 'controller1'
    id: "RGB1_1_B"
    channel: 5

  - platform: pca9685
    pca9685_id: 'controller1'
    id: "RGB1_2_R"
    channel: 6

  - platform: pca9685
    pca9685_id: 'controller1'
    id: "RGB1_2_G"
    channel: 7

  - platform: pca9685
    pca9685_id: 'controller1'
    id: "RGB1_2_B"
    channel: 8

  - platform: pca9685
    pca9685_id: 'controller1'
    id: "RGB1_3_R"
    channel: 9

  - platform: pca9685
    pca9685_id: 'controller1'
    id: "RGB1_3_G"
    channel: 10

  - platform: pca9685
    pca9685_id: 'controller1'
    id: "RGB1_3_B"
    channel: 11
  

  - platform: pca9685
    pca9685_id: 'controller1'
    id: "RGB1_4_R"
    channel: 12

  - platform: pca9685
    pca9685_id: 'controller1'
    id: "RGB1_4_G"
    channel: 13

  - platform: pca9685
    pca9685_id: 'controller1'
    id: "RGB1_4_B"
    channel: 14

light:
  - platform: rgb
    name: "RGB0.0"
    id: RGB0_0
    restore_mode: RESTORE_DEFAULT_OFF
    default_transition_length: 5s
    red: RGB0_0_R
    green: RGB0_0_G
    blue: RGB0_0_B
    
  - platform: rgb
    name: "RGB0.1"
    id: RGB0_1
    restore_mode: RESTORE_DEFAULT_OFF
    default_transition_length: 5s
    red: RGB0_1_R
    green: RGB0_1_G
    blue: RGB0_1_B

  - platform: rgb
    name: "RGB0.2"
    id: RGB0_2
    restore_mode: RESTORE_DEFAULT_OFF
    default_transition_length: 5s
    red: RGB0_2_R
    green: RGB0_2_G
    blue: RGB0_2_B
  
  - platform: rgb
    name: "RGB0.3"
    id: RGB0_3
    restore_mode: RESTORE_DEFAULT_OFF
    default_transition_length: 5s
    red: RGB0_3_R
    green: RGB0_3_G
    blue: RGB0_3_B

  - platform: rgb
    name: "RGB0.4"
    id: RGB0_4
    restore_mode: RESTORE_DEFAULT_OFF
    default_transition_length: 5s
    red: RGB0_4_R
    green: RGB0_4_G
    blue: RGB0_4_B

################################################

  - platform: rgb
    name: "RGB1.0"
    id: RGB1_0
    restore_mode: RESTORE_DEFAULT_OFF
    default_transition_length: 5s
    red: RGB1_0_R
    green: RGB1_0_G
    blue: RGB1_0_B
    
  - platform: rgb
    name: "RGB1.1"
    id: RGB1_1
    restore_mode: RESTORE_DEFAULT_OFF
    default_transition_length: 5s
    red: RGB1_1_R
    green: RGB1_1_G
    blue: RGB1_1_B

  - platform: rgb
    name: "RGB1.2"
    id: RGB1_2
    restore_mode: RESTORE_DEFAULT_OFF
    default_transition_length: 5s
    red: RGB1_2_R
    green: RGB1_2_G
    blue: RGB1_2_B
  
  - platform: rgb
    name: "RGB1.3"
    id: RGB1_3
    restore_mode: RESTORE_DEFAULT_OFF
    default_transition_length: 5s
    red: RGB1_3_R
    green: RGB1_3_G
    blue: RGB1_3_B

  - platform: rgb
    name: "RGB1.4"
    id: RGB1_4
    restore_mode: RESTORE_DEFAULT_OFF
    default_transition_length: 5s
    red: RGB1_4_R
    green: RGB1_4_G
    blue: RGB1_4_B
```

anhorbc, 2023-02-18:
I just noticed the same thing.

[17:11:08][I][i2c.arduino:068]: Results from i2c bus scan:
[17:11:08][I][i2c.arduino:074]: Found i2c device at address 0x20
[17:11:08][I][i2c.arduino:074]: Found i2c device at address 0x40
[17:11:08][I][i2c.arduino:074]: Found i2c device at address 0x41
[17:11:08][C][pca9685:089]: PCA9685:
[17:11:08][C][pca9685:090]:   Mode: 0x0C
[17:11:08][C][pca9685:094]:   EXTCLK: disabled
[17:11:08][C][pca9685:095]:   Frequency: 1000 Hz
[17:11:08][C][pca9685:089]: PCA9685:
[17:11:08][C][pca9685:090]:   Mode: 0x0C
[17:11:08][C][pca9685:094]:   EXTCLK: disabled
[17:11:08][C][pca9685:095]:   Frequency: 1000 Hz
[17:11:08][E][pca9685:098]: Setting up PCA9685 failed!

anhorbc, 2023-02-24:
@standahabich, could you have a look at this? It seems like it broke with your update in 2023.2.
See also: [https://github.com/esphome/issues/issues/4219](url)

omar7c1, 2023-04-17:
Has anyone found a way to fix the error?

standahabich, 2023-04-18:
ohh, ok. The software reset will take place twice. I understand. So, do I remove the software reset, with the fact that to turn off extclk, a power cycle will have to be performed?

omar7c1, 2023-05-19:
> Any news about this problem? Just trying since two hours why it was working with only one pcs9685 and with two PCA9685s it isn't anymore...

It is not fixed yet but if you add this in your code it works
## FIX  BUG  https://github.com/esphome/issues/issues/4219
external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome.git
      ref: 2022.12.8
    components: [ pca9685 ]

TheEris, 2023-06-08:
any updates on this ?

total1ty, 2023-07-06:
It seems to have been fixed 2 weeks ago, see https://github.com/esphome/esphome/commit/b346ad8080b3a9b29a3920c4ecb56c2dc69321bb

I've tried it and removed the workaround code for the newest esphome version 2023.6.4, and it works fine.

closed 2023-08-07 by ssieb