esphome/issues #4187: Cannot download compile toolchain for NODEMCU-32S

ShawnLin529, 2023-02-17:
### The problem

When we try to flash ESPHome into the NODEMCU-32S device (by `$esphome --dashboard run esphome_config/esp32-1.yaml --device OTA`), we gets to following notes:

INFO Reading configuration esphome_config/esp32-1.yaml...                                                               
INFO Generating C++ source...                                                                                           
INFO Compiling app...                                                                                                   
Processing \033[36m\033[1mesp32-1\033[0m (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32 @ 5.2.0)                                                                                                                     
\033[1m------------------------------------------------------------------------------------------------------------------------\033[0m                                                                                                          
Tool Manager: Installing \033[36mespressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3\033[0m                          
INFO Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3                                                  
Tool Manager: \033[33mWarning! Package Mirror: Got the unrecognized status code '404' when downloaded https://dl.registry.ns3.platformio.org/tools/53/b2/05540868edc4f775b6364726f7b9ef3abd58512eb00bb2e0b92ccccadc01/toolchain-xtensa-esp32-linux_armv6l-8.4.0+2021r2-patch3.tar.gz\033[0m                                                                             
WARNING Warning! Package Mirror: Got the unrecognized status code '404' when downloaded https://dl.registry.ns3.platformio.org/tools/53/b2/05540868edc4f775b6364726f7b9ef3abd58512eb00bb2e0b92ccccadc01/toolchain-xtensa-esp32-linux_armv6l-8.4.0+2021r2-patch3.tar.gz                                                                                                  
Tool Manager: \033[33mLooking for another mirror...\033[0m                                                              
WARNING Looking for another mirror... 

### Which version of ESPHome has the issue?

2023.2.1

### What type of installation are you using?

pip

### What platform are you using?

ESP32

ssieb, 2023-02-17:
This isn't an esphome problem.  If anything, it would be platformio.  What is your config?

CyanoFresh, 2023-03-08:
probably same here:
```
INFO Reading configuration /config/room1-esp32.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing room1-esp32 (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif32 @ 5.2.0
INFO Installing platformio/espressif32 @ 5.2.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%          
Platform Manager: espressif32@5.2.0 has been installed!
INFO espressif32@5.2.0 has been installed!
Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
INFO Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: Warning! Package Mirror: [('/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/strip', '/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/bin/strip', "[Errno 2] No such file or directory: '/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/strip'"), ('/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/ld', '/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/bin/ld', "[Errno 2] No such file or directory: '/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/ld'"), ('/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/readelf', '/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/bin/readelf', "[Errno 2] No such file or directory: '/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/readelf'"), ('/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/ld.bfd', '/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/bin/ld.bfd', "[Errno 2] No such file or directory: '/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/ld.bfd'"), ('/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/nm', '/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/bin/nm', "[Errno 2] No such file or directory: '/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/nm'"), ('/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/ar', '/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/bin/ar', "[Errno 2] No such file or directory: '/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/ar'"), ('/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/as', '/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/bin/as', "[Errno 2] No such file or directory: '/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/as'"), ('/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/objcopy', '/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/bin/objcopy', "[Errno 2] No such file or directory: '/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/objcopy'"), ('/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/ranlib', '/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/bin/ranlib', "[Errno 2] No such file or directory: '/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/ranlib'")]
WARNING Warning! Package Mirror: [('/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/strip', '/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/bin/strip', "[Errno 2] No such file or directory: '/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/strip'"), ('/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/ld', '/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/bin/ld', "[Errno 2] No such file or directory: '/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/ld'"), ('/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/readelf', '/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/bin/readelf', "[Errno 2] No such file or directory: '/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/readelf'"), ('/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/ld.bfd', '/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/bin/ld.bfd', "[Errno 2] No such file or directory: '/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/ld.bfd'"), ('/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/nm', '/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/bin/nm', "[Errno 2] No such file or directory: '/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/nm'"), ('/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/ar', '/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/bin/ar', "[Errno 2] No such file or directory: '/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/ar'"), ('/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-esp32-elf/bin/as', '/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/bin/as', "[Errno 2] No such file or directory: '/cache/platformio/cache/tmp/pkg-installing-8bhr5wnk/xtensa-es [... cut ...]
  scan_parameters:
    interval: 1100ms
    window: 1100ms

binary_sensor:
  - platform: ble_presence
    mac_address: --
[... 17 lines cut ...]
  encryption:
    key: $encryption_key

logger:

mdns:

ota:
  password: !secret manage_password

wifi:
  ssid: ---
  password: !secret wifi_password
  fast_connect: true
  
  manual_ip:
    static_ip: 192.168.1.${device_dhcp}
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    
  ap:
    ssid: $device_name
    password: !secret manage_password

```