esphome/issues #4191: esp32_camera not working with 10/20MHZ, only 8MHZ

mannipulator, 2023-02-17:
### The problem

Hello all,
First of all: I'm not sure if this is a bug or a feature request, but since the software generates malfunctions because of the default clock setting, I classified it as a bug, I hope that's right. 

it's about the esp32_camera class, with the ESP32-CAM-MB board it doesn't work, it has an extremely bad performance. After some research I found a GiHub post where users have found a solution, it is a hardware problem where probably interferences are generated that occur at a clock frequency above 10MHZ. One user added a ground plane with aluminum foil, and another user reduced the clock frequency to 8MHZ in Arduino Studio, eliminating the interference.
This is the post: https://github.com/espressif/arduino-esp32/issues/4655#issuecomment-1374509367
 
I have been able to reconstruct this using the Esp32CameraWebserver example in the Arduino Studio as well, at 10MHZ and 20MHZ the camera is virtually useless, once you reduce the clock frequency to 8MHZ the board works flawlessly.

The problem is that ESPHome only accepts 10 MHZ and 20 MHZ in the esp32_camera class, at 8 MHZ this error message appears:

 esp32_camera: [source /config/esphome/esp-cam.yaml:35]
   name: espcam
   external_clock: 
     pin: GPIO0
    
    Unknown value '8000000.0', did you mean '20000000.0', '10000000.0'?.
    frequency: 8 MHZ

My suggestion for this is to allow the clock frequency 8 MHZ as a valid value.
I hope I could help with this

### Which version of ESPHome has the issue?

2023.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP32

### Board

esp32cam

### Component causing the issue

esp32_camera

### Anything in the logs that might be useful for us?

```txt
esp32_camera: [source /config/esphome/esp-cam.yaml:35]
   name: espcam
   external_clock: 
     pin: GPIO0
    
    Unknown value '8000000.0', did you mean '20000000.0', '10000000.0'?.
    frequency: 8 MHZ
```

### Additional information

8 MHZ Clock suggestion 
https://github.com/espressif/arduino-esp32/issues/4655#issuecomment-1374509367

Alufoil-hack:
https://www.reddit.com/r/esp32/comments/r9g5jc/fixing_ymmv_the_poor_frame_rate_on_the_esp32cam/

berendhaan, 2023-03-24:
I've made a PR, https://github.com/esphome/esphome/pull/4619, lowering the range in the configuration.

diplix, 2023-03-25:
this helped make all my troubles with my 3 esp cameras disappear. wifi reliabilty or throughput have greatly improved and moreover, the home assistant API disconnects that happend almost all the time with lower RSSIs  quality have also disappeared.
the `8MHz` frequency setting should actually be the new defauklt value instead of the `20MHz`.

nvtkaszpir, 2023-05-12:
Please note that with `board: esp-wrover-kit` under v2023.4.4 camera frequency is by default 20MHz - with 640x480 it gets around 8FPS.
Lowering frequency to 8MHz makes it to perform worse, it drops to around 5FPS.

closed 2023-11-27 by mannipulator