esphome/issues #4194: Component esp32_touch not working in ESP32-S2

melwinek, 2023-02-17:
### The problem

As you can see in the logs below. Cannot compile correctly. In ESP32 it works fine. In ESP32-S2 it does not work. In the ESP-IDF framework it also does not work, not only in Arduino.
By the way, I found another bug. Not all touch pins from ESP32-S2 can be used for esp_touch. Please see the attachment. Only those that are in ESP32 can be used.
[image]

### Which version of ESPHome has the issue?

v2022.12.8

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

Wemos ESP32 S2 Mini

### Component causing the issue

esp32_touch

### Example YAML snippet

```yaml
wifi:
  ssid: !secret wifi_iot_ssid
  password: !secret wifi_iot_password
  manual_ip:
    static_ip: 192.168.3.33
    gateway: 192.168.3.49
    subnet: 255.255.255.0

api:
  encryption:
    key: !secret api_encryption_key

ota:
  password: !secret esphome_api_password
    
esphome:
  name: test
  platformio_options:
    board_build.extra_flags:
      - "-DARDUINO_USB_CDC_ON_BOOT=0"
        
esp32:
  board: lolin_s2_mini
  framework:
    type: arduino

esp32_touch:
  setup_mode: true

binary_sensor: 
- platform: esp32_touch
  name: "PIN1"
  pin: GPIO12
  threshold: 1000
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/test.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing test (board: lolin_s2_mini; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling .pioenvs/test/src/esphome/components/esp32_touch/esp32_touch.cpp.o
Compiling .pioenvs/test/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/test/src/esphome/components/mdns/mdns_esp_idf.cpp.o
Compiling .pioenvs/test/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling .pioenvs/test/src/esphome/components/network/util.cpp.o
Compiling .pioenvs/test/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling .pioenvs/test/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling .pioenvs/test/src/esphome/components/ota/ota_backend_arduino_rp2040.cpp.o
Compiling .pioenvs/test/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling .pioenvs/test/src/esphome/components/ota/ota_component.cpp.o
src/esphome/components/esp32_touch/esp32_touch.cpp: In member function 'virtual void esphome::esp32_touch::ESP32TouchComponent::setup()':
src/esphome/components/esp32_touch/esp32_touch.cpp:18:5: error: 'touch_pad_filter_start' was not declared in this scope
     touch_pad_filter_start(this->iir_filter_);
     ^~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/esp32_touch/esp32_touch.cpp:18:5: note: suggested alternative: 'touch_pad_filter_enable'
     touch_pad_filter_start(this->iir_filter_);
     ^~~~~~~~~~~~~~~~~~~~~~
     touch_pad_filter_enable
src/esphome/components/esp32_touch/esp32_touch.cpp:26:47: error: too many arguments to function 'esp_err_t touch_pad_config(touch_pad_t)'
     touch_pad_config(child->get_touch_pad(), 0);
                                               ^
In file included from src/esphome/components/esp32_touch/esp32_touch.h:12,
                 from src/esphome/components/esp32_touch/esp32_touch.cpp:3:
/config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s2/include/driver/esp32s2/include/driver/touch_sensor.h:156:11: note: declared here
 esp_err_t touch_pad_config(touch_pad_t touch_num);
           ^~~~~~~~~~~~~~~~
src/esphome/components/esp32_touch/esp32_touch.cpp: In member function 'virtual void esphome::esp32_touch::ESP32TouchComponent::loop()':
src/esphome/components/esp32_touch/esp32_touch.cpp:117:7: error: 'touch_pad_read_filtered' was not declared in this scope
       touch_pad_read_filtered(child->get_touch_pad(), &value);
       ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/esp32_touch/esp32_touch.cpp:117:7: note: suggested alternative: 'touch_pad_read_raw_data'
       touch_pad_read_filtered(child->get_touch_pad(), &value);
       ^~~~~~~~~~~~~~~~~~~~~~~
       touch_pad_read_raw_data
src/esphome/components/esp32_touch/esp32_touch.cpp:119:7: error: 'touch_pad_read' was not declared in this scope
       touch_pad_read(child->get_touch_pad(), &value);
       ^~~~~~~~~~~~~~
src/esphome/components/esp32_touch/esp32_touch.cpp:119:7: note: suggested alternative: 'touch_pad_reset'
       touch_pad_read(child->get_touch_pad(), &value);
       ^~~~~~~~~~~~~~
       touch_pad_reset
src/esphome/components/esp32_touch/esp32_touch.cpp: In member function 'virtual void esphome::esp32_touch::ESP32TouchComponent::on_shutdown()':
src/esphome/components/esp32_touch/esp32_touch.cpp:142:5: error: 'touch_pad_filter_stop' was not declared in this scope
     touch_pad_filter_stop();
     ^~~~~~~~~~~~~~~~~~~~~
src/esphome/components/esp32_touch/esp32_touch.cpp:142:5: note: suggested alternative: 'touch_pad_fsm_stop'
     touch_pad_filter_stop();
     ^~~~~~~~~~~~~~~~~~~~~
     touch_pad_fsm_stop
src/esphome/components/esp32_touch/esp32_touch.cpp:143:5: error: 'touch_pad_filter_delete' was not declared in this scope
     touch_pad_filter_delete();
     ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/esp32_touch/esp32_touch.cpp:143:5: note: suggested alternative: 'touch_pad_filter_disable'
     touch_pad_filter_delete();
     ^~~~~~~~~~~~~~~~~~~~~~~
     touch_pad_filter_disable
src/esphome/components/esp32_touch/esp32_touch.cpp:155:77: error: too many arguments to function 'esp_err_t touch_pad_config(touch_pad_t)'
       touch_pad_config(child->get_touch_pad(), child->get_wakeup_threshold());
                                                                             ^
In file included from src/esphome/components/esp32_touch/esp32_touch.h:12,
                 from src/esphome/components/esp32_touch/esp32_touch.cpp:3:
/config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s2/include/driver/esp32s2/include/driver/touch_sensor.h:156:11: note: declared here
 esp_err_t touch_pad_config(touch_pad_t touch_num);
           ^~~~~~~~~~~~~~~~
*** [.pioenvs/test/src/esphome/components/esp32_touch/esp32_touch.cpp.o] Error 1
========================== [FAILED] Took 4.62 seconds ==========================
```

TLCary, 2023-02-17:
Same.  This error was first reported at the Home Assistant/ESHome forum back in October.

melwinek, 2023-02-17:
ESP32-S2 not ESP32.
[image]

TLCary, 2023-02-17:
Already tried GPIO14.   Since the error is a compile error because it can't find what it needs in the current scope... seems like an include is missing.  I just don't know how to resolve it.

TLCary, 2023-02-18:
As of release (v12.1.1.3) S2 & S3 support the re-written touch driver:  SOC_TOUCH_VERSION_2 .  The default threshold for this driver is 40000.  It's higher than the old driver or what is reflected in the ESPHome ESP32 Touch Pad Documentation. 
It looks like the SOC_TOUCH_VERSION_2 isn't implemented in ESPHome.  I see Arends implemented it as an option in Tasmota last October.

melwinek, 2023-02-18:
Exactly.  It looks like you wrote.  In EspEasy, driver v2 is also made.  So we are waiting for ESPHome :(

closed 2023-03-13 by jesserockz

melwinek, 2023-03-13:
It seems that the touch pins from the ESP32-S2 are already accepting correctly. e.g. GPIO1.
The touch_sensor_v2 implementation is still missing.

LukassviheL, 2023-03-24:
Fixing this would make my day 😄 For me it's critical to have touch sensor capabilities when I'm trying to detect water leakage.