esphome/issues #4195: display: breaks wifi on M5Stack Core (Black model)

jonsenior79, 2023-02-17:
### The problem

I'm trying to use an M5Stack to read the temperature from two Xiaomi Mijia sensors. I can create a minimal yaml which successfully reads the temperatures and relays them to an instance of HomeAssistant. But ideally I'd like the M5Stack to also display information on the screen so that I don't need a smartphone or computer to see what's going on. When I try to add the display though (information sourced from "https://github.com/koenvervloesem/M5Stack-Air-Quality-ESPHome") the wifi stops working and the device seems to go into some sort of reset loop.

### Which version of ESPHome has the issue?

2023.2.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

m5stack-core-esp32

### Component causing the issue

display

### Example YAML snippet

```yaml
esphome:
  name: thermostat

esp32:
  board: m5stack-core-esp32
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "XXXXX"
  password: "XXXXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Thermostat Fallback Hotspot"
    password: "uvSC9hKsa86v"

captive_portal:

esp32_ble_tracker:

sensor:
  - platform: pvvx_mithermometer
    mac_address: "XXXXX"
    temperature:
      name: "Upstairs Temperature"
    humidity:
      name: "Upstairs Humidity"
    battery_level:
      name: "Upstairs Battery-Level"
    battery_voltage:
      name: "Upstairs Battery-Voltage"
    signal_strength:
      name: "Upstairs Signal"

  - platform: pvvx_mithermometer
    mac_address: "XXXXX"
    temperature:
      name: "Downstairs Temperature"
    humidity:
      name: "Downstairs Humidity"
    battery_level:
      name: "Downstairs Battery-Level"
    battery_voltage:
      name: "Downstairs Battery-Voltage"
    signal_strength:
      name: "Downstairs Signal"

spi:
  clk_pin: 18
  mosi_pin: 23
  miso_pin: 19

#font:
#  - file: "fonts/Roboto-Medium.ttf"
#    id: font_roboto_medium22
#    size: 22
#    glyphs: '!"%()+,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ #abcdefghijklmnopqrstuvwxyz/³µ'

display:
  - platform: ili9341
    id: m5stack_display
    model: M5Stack
    cs_pin: 14
    dc_pin: 27
    led_pin: 32
    reset_pin: 33
    rotation: 0
#    lambda: |-
#      Color RED(1,0,0);
#      Color BLUE(0,0,1);
#      Color WHITE(1,1,1);
#      it.rectangle(0,  0, it.get_width(), it.get_height(), BLUE);
#      it.rectangle(0, 22, it.get_width(), it.get_height(), BLUE);
#      it.print(it.get_width() / 2, 11, id(font_roboto_medium22), RED, #TextAlign::CENTER, "Particulate matter");

# Button to toggle the display backlight
#binary_sensor:
#  - platform: gpio
#    id: M5_BtnB
#    pin:
#      number: 38
#      inverted: true
#    on_click:
#      then:
#        - switch.toggle: backlight

# GPIO pin of the display backlight
#switch:
#  - platform: gpio
#    pin: 32
#    name: "Backlight"
#    id: backlight
#    restore_mode: ALWAYS_ON
```

### Anything in the logs that might be useful for us?

```txt
[17:43:05][I][logger:258]: Log initialized
[17:43:05][C][ota:469]: There have been 1 suspected unsuccessful boot attempts.
[17:43:05][D][esp32.preferences:113]: Saving 1 preferences to flash...
[17:43:05][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[17:43:05][I][app:029]: Running through setup()...
[17:43:05][C][spi:023]: Setting up SPI bus...
[17:43:05][C][esp32_ble:026]: Setting up BLE...
[17:43:06][D][esp32_ble:042]: BLE setup complete
[17:43:06][D][esp32_ble_tracker:238]: Starting scan...
[17:43:06][C][wifi:037]: Setting up WiFi...
[17:43:06][C][wifi:038]:   Local MAC: A4:CF:12:6D:B7:84
[17:43:06][D][esp-idf:000]: E (2627) wifi:
[17:43:06][D][esp-idf:000]: Expected to init 4 rx buffer, actual is 0
[17:43:06][D][esp-idf:000]: 
[17:43:06]
[17:43:06][  1162][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
[17:43:06][W][wifi_esp32:057]: Setting WiFi mode failed!
[17:43:06][D][wifi:386]: Starting scan...
[17:43:06][D][esp-idf:000]: E (2665) wifi:
[17:43:06][D][esp-idf:000]: Expected to init 4 rx buffer, actual is 0
[17:43:06][D][esp-idf:000]: 
[17:43:06]
[17:43:06][  1200][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
[17:43:06][W][wifi_esp32:057]: Setting WiFi mode failed!
[17:43:30][D][pvvx_mithermometer:120]: Got PVVX MiThermometer (A4:C1:38:30:9E:E6):
[17:43:30][D][pvvx_mithermometer:123]:   Temperature: 21.35 °C
[17:43:30][D][pvvx_mithermometer:126]:   Humidity: 63.14 %
[17:43:30][D][pvvx_mithermometer:129]:   Battery Level: 87 %
[17:43:30][D][pvvx_mithermometer:132]:   Battery Voltage: 2.990 V
[17:43:30][D][sensor:127]: 'Upstairs Temperature': Sending state 21.35000 °C with 2 decimals of accuracy
[17:43:30][D][sensor:127]: 'Upstairs Humidity': Sending state 63.14000 % with 2 decimals of accuracy
[17:43:30][D][sensor:127]: 'Upstairs Battery-Level': Sending state 87.00000 % with 0 decimals of accuracy
[17:43:30][D][sensor:127]: 'Upstairs Battery-Voltage': Sending state 2.99000 V with 3 decimals of accuracy
[17:43:30][D][sensor:127]: 'Upstairs Signal': Sending state -68.00000 dBm with 0 decimals of accuracy
[17:43:36][E][wifi:394]: Scan timeout!
[17:43:41][D][wifi:386]: Starting scan...
[17:43:41][D][esp-idf:000]: E (37676) wifi:
[17:43:41][D][esp-idf:000]: Expected to init 4 rx buffer, actual is 0
[17:43:41][D][esp-idf:000]: 
[17:43:41]
[17:43:41][ 36211][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
[17:43:41][W][wifi_esp32:057]: Setting WiFi mode failed!
[17:44:06][I][wifi:141]: Starting fallback AP!
[17:44:06][D][esp-idf:000]: E (62638) wifi:
[17:44:06][D][esp-idf:000]: Expected to init 4 rx buffer, actual is 0
[17:44:06][D][esp-idf:000]: 
[17:44:06]
[17:44:06][ 61173][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
[17:44:06][W][wifi_esp32:057]: Setting WiFi mode failed!
[17:44:06][C][wifi:210]: Setting up AP...
[17:44:06][C][wifi:212]:   AP SSID: 'Thermostat Fallback Hotspot'
[17:44:06][C][wifi:213]:   AP Password: 'uvSC9hKsa86v'
[17:44:06][D][esp-idf:000]: E (62686) wifi:
[17:44:06][D][esp-idf:000]: Expected to init 4 rx buffer, actual is 0
[17:44:06][D][esp-idf:000]: 
[17:44:06]
[17:44:06][ 61221][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
[17:44:06][W][wifi_esp32:057]: Setting WiFi mode failed!
[17:44:06][C][wifi:222]:   IP Address: 192.168.4.1
[17:44:06][ 61232][E][AsyncTCP.cpp:1290] begin(): failed to start task
[17:44:06][I][wifi:141]: Starting fallback AP!
[17:44:06][ 61260][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
[17:44:06][W][wifi_esp32:057]: Setting WiFi mode failed!
[17:44:06][C][wifi:210]: Setting up AP...
[17:44:06][C][wifi:212]:   AP SSID: 'Thermostat Fallback Hotspot'
[17:44:06][C][wifi:213]:   AP Password: 'uvSC9hKsa86v'
[17:44:06][ 61297][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
[17:44:06][W][wifi_esp32:057]: Setting WiFi mode failed!
[17:44:06][C][wifi:222]:   IP Address: 192.168.4.1
[17:44:06][I][wifi:141]: Starting fallback AP!
[17:44:06][ 61325][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
[17:44:06][W][wifi_esp32:057]: Setting WiFi mode failed!
```

jonsenior79, 2023-02-17:
Also... Nothing gets displayed on the screen. So it's not just the wifi breaking. I've also tried TFT 2.4R instead of M5Stack in case it was a driver chip issue, but neither seems to work. If I enable the backlight control via button B, I can turn on and off the backlight.

nielsnl68, 2023-02-18:
Not sure about the Xiaomi Mijia sensors, i hope someone else can look into that.

But i can help try with the display? First did it work before or is this the first time testing this?
Anyway i'm have created a PR for a new version for the ili9xxx series of displays and the M5 Core uses on of does as you already knows.

I have checked the code that M5 has released and found some differences. And made some modifications to see of your display works for that. Sadly i do not own a M5Core myself so it is a bit guess work. 

could you try:
```yaml
external_components:
  - source: github://pr#4275
    components: [ili9xxx]
    refresh: always

display:
  - platform: ili9xxx
    id: m5stack_display
    model: M5Core
    backlight_pin: 32
    ...

```

jonsenior79, 2023-02-18:
I tried it and got the following:

`INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[22:06:53] clock div:1
[22:06:53]load:0x3fff0030,len:1344
[22:06:53]load:0x40078000,len:13864
[22:06:53]load:0x40080400,len:3608
[22:06:53]entry 0x400805f0
[22:06:53]E (338) psram: PSRAM ID read error: 0xffffffff
[22:06:53][I][logger:258]: Log initialized
[22:06:53][C][ota:469]: There have been 0 suspected unsuccessful boot attempts.
[22:06:53][D][esp32.preferences:113]: Saving 1 preferences to flash...
[22:06:53][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[22:06:53][I][app:029]: Running through setup()...
[22:06:53][C][spi:023]: Setting up SPI bus...
[22:06:54][E][display:100]: Could not allocate buffer for display!
[22:06:54][C][esp32_ble:026]: Setting up BLE...
[22:06:54]
[22:06:54]Brownout detector was triggered
[22:06:54]
[22:06:54]ets Jun  8 2016 00:22:57
[22:06:54]
[22:06:54]rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
[22:06:54]configsip: 0, SPIWP:0xee
[22:06:54]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:06:54]mode:DIO, clock div:1
[22:06:54]load:0x3fff0030,len:1344
[22:06:54]load:0x40078000,len:13864
[22:06:54]load:0x40080400,len:3608
[22:06:54]entry 0x400805f0
[22:06:54]E (338) psram: PSRAM ID read error: 0xffffffff
[22:06:54][I][logger:258]: Log initialized
[22:06:54][C][ota:469]: There have been 1 suspected unsuccessful boot attempts.
[22:06:54][D][esp32.preferences:113]: Saving 1 preferences to flash...
[22:06:54][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[22:06:54][I][app:029]: Running through setup()...
[22:06:54][C][spi:023]: Setting up SPI bus...
[22:06:55][E][display:100]: Could not allocate buffer for display!
[22:06:55][C][esp32_ble:026]: Setting up BLE...
[22:06:55]
[22:06:55]Brownout detector was triggered
`

jonsenior79, 2023-02-18:
Tried it and get a power cycle due to brownout error. Which makes me wonder if I don't need to find a better power source than the USB hub that it's running from right now. I'll have a play with independant power for it.

INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[22:06:53] clock div:1
[22:06:53]load:0x3fff0030,len:1344
[22:06:53]load:0x40078000,len:13864
[22:06:53]load:0x40080400,len:3608
[22:06:53]entry 0x400805f0
[22:06:53]E (338) psram: PSRAM ID read error: 0xffffffff
[22:06:53][I][logger:258]: Log initialized
[22:06:53][C][ota:469]: There have been 0 suspected unsuccessful boot attempts.
[22:06:53][D][esp32.preferences:113]: Saving 1 preferences to flash...
[22:06:53][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[22:06:53][I][app:029]: Running through setup()...
[22:06:53][C][spi:023]: Setting up SPI bus...
[22:06:54][E][display:100]: Could not allocate buffer for display!
[22:06:54][C][esp32_ble:026]: Setting up BLE...
[22:06:54]
[22:06:54]Brownout detector was triggered
[22:06:54]
[22:06:54]ets Jun  8 2016 00:22:57
[22:06:54]
[22:06:54]rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
[22:06:54]configsip: 0, SPIWP:0xee
[22:06:54]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:06:54]mode:DIO, clock div:1
[22:06:54]load:0x3fff0030,len:1344
[22:06:54]load:0x40078000,len:13864
[22:06:54]load:0x40080400,len:3608
[22:06:54]entry 0x400805f0
[22:06:54]E (338) psram: PSRAM ID read error: 0xffffffff
[22:06:54][I][logger:258]: Log initialized
[22:06:54][C][ota:469]: There have been 1 suspected unsuccessful boot attempts.
[22:06:54][D][esp32.preferences:113]: Saving 1 preferences to flash...
[22:06:54][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[22:06:54][I][app:029]: Running through setup()...
[22:06:54][C][spi:023]: Setting up SPI bus...
[22:06:55][E][display:100]: Could not allocate buffer for display!
[22:06:55][C][esp32_ble:026]: Setting up BLE...
[22:06:55]
[22:06:55]Brownout detector was triggered

jonsenior79, 2023-02-18:
Also, apologies for closing the issue. I clicked on the wrong thing and apparently can't take it back!