esphome/issues #4200: matrix_keypad: requires that pins support output when not needed

Meetsch, 2023-02-18:
### The problem

Using the matrix_keypad with ESP32 pints 34~39 (input only) it complains that the pin "does not support output pin mode".
But input only should be sufficient when the keypad has no LEDs (as the vast majority of keypads)

### Which version of ESPHome has the issue?

2023.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

matrix_keypad

### Example YAML snippet

```yaml
#Keypad
#https://esphome.io/components/matrix_keypad.html
matrix_keypad:
  id: keypad
  rows:
    - pin: 34
    - pin: 35
    - pin: 32
    - pin: 33
  columns:
    - pin: 25
    - pin: 26
    - pin: 27
    - pin: 14
  keys: "123A456B789C*0#D"
  has_diodes: false
#https://esphome.io/components/key_collector.html
key_collector:
  - id: pincode_reader
    source_id: keypad
    min_length: 4
    max_length: 4
    end_keys: "#"
    end_key_required: true
    back_keys: "D"
    clear_keys: "C"
    allowed_keys: "0123456789"
    timeout: 10s
    on_progress:
      - logger.log:
          format: "input progress: '%s', started by '%c'"
          args: [ 'x.c_str()', "(start == 0 ? '~' : start)" ]
    on_result:
      - logger.log:
          format: "input result: '%s', started by '%c', ended by '%c'"
          args: [ 'x.c_str()', "(start == 0 ? '~' : start)", "(end == 0 ? '~' : end)" ]
    on_timeout:
      - logger.log:
          format: "input timeout: '%s', started by '%c'"
          args: [ 'x.c_str()', "(start == 0 ? '~' : start)" ]
```

### Anything in the logs that might be useful for us?

```txt
matrix_keypad: [source /config/esphome/esphome-alarm-keypad.yaml:109]
  - id: keypad
    rows: 
      - 
        GPIO34 (34-39) does not support output pin mode.
        pin: 34
      - pin: 35
      - pin: 32
      - pin: 33
    columns: 
      - pin: 25
      - pin: 26
      - pin: 27
      - pin: 14
    keys: 123A456B789C*0#D
    has_diodes: False
```

ssieb, 2023-02-18:
I suppose it should be specified in the docs, but the rows require output.  The columns can be input only, but they require pullups which those pins don't have either.  You would have to add external pullups to use those pins.

Meetsch, 2023-02-18:
do you have a link to the doc?

ssieb, 2023-02-18:
You have the link in your yaml file.  I said that it "should" have been specified in the docs, but it isn't.  I will see about adding that information now that I see it could matter.

Meetsch, 2023-02-18:
yes, it was not obvious, at least for me...
but your small hint paved me in the right direction and now all is working fine. thanks for the great job on this integration !

closed 2023-02-22 by nagyrobi