esphome/issues #4203: Compilation for Pico Fails on Raspberry Pi 4

bwims, 2023-02-18:
### The problem

This problem is essentially the same as https://github.com/esphome/issues/issues/3854 except that I am running 64-bit OS on a Raspberry Pi 4.

I am trying to install ESPHome on a Pico W, and it is failing to install the toolchain.

### Which version of ESPHome has the issue?

v2023.2.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

RP2040

### Board

pico W

### Component causing the issue

Compilation

### Example YAML snippet

```yaml
esphome:
  name: test
  friendly_name: test

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "haJALLqkGytxTBq0yyfU+wgpMzBRD9FR0mk7BeZ5HFk="

ota:
  password: "d013f248668e5d59dc9671092b707cbc"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"
    password: "H2LM00ADsdki"
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/test.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing test (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
Tool Manager: Installing earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.221223
INFO Installing earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.221223
Error: Could not find the package with 'earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.221223' requirements for your system 'linux_aarch64'
```

cazeaux, 2023-02-19:
I have the same issue, but running esphome on home assistant using proxmox (with haos 9.5) so on standard linux x86 64.
Same version of ESPHOME.

I have the following output:

```
INFO Reading configuration /config/esphome/esphome-web-08faa0.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-08faa0 (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
Tool Manager: Installing earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.221223
INFO Installing earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.221223
Error: Could not find the package with 'earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.221223' requirements for your system 'linux_x86_64'
```

bwims, 2023-02-19:
It looks like [this](https://github.com/maxgerhardt/platform-raspberrypi/commit/b9928a2d113f095ccea2540f6fb060d665243063)
is related to my problem, but it seems to have been applied and should have fixed it by now, but it hasn't.

Could somebody PLLLLLLLEEEEEEAAAASSSSSEEEEE give me a clue?

Thanks!

B.

bwims, 2023-02-20:
Looks like the answer was to remove the ".git" from  
platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

After that everything compiles.   It fails to upload, but that's something to do with the reset, I guess.