esphome/issues #4204: can't ping or visit device web server, but logging in ESPHome is normal

mac-city, 2023-02-18:
### The problem

this issue is now over a year old, and is the same case with all my ESPHome devices.  I can no longer ping or reach the web server, yet device logs show up normally and it works fine in HA as well.  this is just a minor irritation

### Which version of ESPHome has the issue?

2023.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP8266

### Board

esp01_1m

### Example YAML snippet

```yaml
## LOHAS RGBW Smart Bulb with ESPHome

esphome:
  name: lohas19
  platform: ESP8266
  board: esp01_1m
  includes:
    - copychan.h

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: redacted
    gateway: redacted
    subnet: redacted
    dns1: redacted

# Enable logging and direct connect to change firmware 
web_server:
  port: 80
  
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "redacted"

# Enable MQTT if you are using it over the HA API
#mqtt:
#  broker: '192.168.1.x'
#  username: 'MQTT_User'
#  password: 'MQTT_Password'

ota:
  
my9231:
 data_pin: GPIO13  
 clock_pin: GPIO15  
 num_channels: 6
 num_chips: 2 
 
output:
  - platform: my9231
    id: output_blue
    channel: 3
  - platform: my9231
    id: output_red
    channel: 5
  - platform: my9231
    id: output_green
    channel: 4
  - platform: my9231
    id: output_cw1
    channel: 1
  - platform: my9231
    id: output_cw2
    channel: 2
  - platform: custom
    type: float
    lambda: |-
      auto *copy = new CopyOutput();
      copy->channel_a = id(output_cw1);
      copy->channel_b = id(output_cw2);
      return {copy}; 
    outputs:
    - id: cw

light:
  - platform: rgbw
    name: "Lohas19"
    default_transition_length: 0s
    red: output_red
    green: output_green
    blue: output_blue
    white: cw
```

adamsd5, 2024-02-18:
I'm having this issue.  First time esphome user

adamsd5, 2024-02-18:
[I][logger:351]: Log initialized
[C][ota:473]: There have been 0 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[C][wifi:038]: Setting up WiFi...
[C][wifi:051]: Starting WiFi...
[C][wifi:052]:   Local MAC: <redacted>
[D][wifi:455]: Starting scan...
[D][wifi:470]: Found networks:
[I][wifi:514]: - '<redacted>' [redacted]▂▄▆█
[D][wifi:515]:     Channel: 6
[D][wifi:516]:     RSSI: -30 dB
[I][wifi:514]: - '<redacted>' [redacted]▂▄▆█
[D][wifi:515]:     Channel: 1
[D][wifi:516]:     RSSI: -86 dB
[D][wifi:519]: - [redacted] [redacted]▂▄▆█
[D][wifi:519]: - [redacted] [redacted]▂▄▆█
[D][wifi:519]: - [redacted] [redacted]▂▄▆█
[D][wifi:519]: - [redacted] [redacted]▂▄▆█
[D][wifi:519]: - [redacted] [redacted]▂▄▆█
[D][wifi:519]: - [redacted] [redacted]▂▄▆█
[D][wifi:519]: - [redacted] [redacted]▂▄▆█
[D][wifi:519]: - [redacted] [redacted]▂▄▆█
[D][wifi:519]: - [redacted] [redacted]▂▄▆█
[D][wifi:519]: - [redacted] [redacted]▂▄▆█
[I][wifi:300]: WiFi Connecting to '<redacted>'...
[I][wifi:587]: WiFi Connected!
[C][wifi:405]:   Local MAC: <redacted>
[C][wifi:410]:   SSID: [redacted]
[C][wifi:411]:   IP Address: 192.168.200.5
[C][wifi:413]:   BSSID: [redacted]
[C][wifi:414]:   Hostname: 'esp32-device-1'
[C][wifi:416]:   Signal strength: -31 dB ▂▄▆█
[C][wifi:420]:   Channel: 6
[C][wifi:421]:   Subnet: 255.255.255.0
[C][wifi:422]:   Gateway: 192.168.200.1
[C][wifi:423]:   DNS1: 192.168.200.1
[C][wifi:424]:   DNS2: 0.0.0.0
[D][wifi:596]: Disabling AP...
[C][web_server:123]: Setting up web server...
[C][ota:097]: Over-The-Air Updates:
[C][ota:098]:   Address: esp32-device-1.local:3232
[C][ota:101]:   Using Password.
[C][api:025]: Setting up Home Assistant API server...
[I][app:062]: setup() finished successfully!
[I][app:102]: ESPHome version 2023.12.9 compiled on Feb 18 2024, 17:17:47
[C][wifi:573]: WiFi:
[C][wifi:405]:   Local MAC: <redacted>
[C][wifi:410]:   SSID: [redacted]
[C][wifi:411]:   IP Address: 192.168.200.5
[C][wifi:413]:   BSSID: [redacted]
[C][wifi:414]:   Hostname: 'esp32-device-1'
[C][wifi:416]:   Signal strength: -33 dB ▂▄▆█
[C][wifi:420]:   Channel: 6
[C][wifi:421]:   Subnet: 255.255.255.0
[C][wifi:422]:   Gateway: 192.168.200.1
[C][wifi:423]:   DNS1: 192.168.200.1
[C][wifi:424]:   DNS2: 0.0.0.0
[C][logger:439]: Logger:
[C][logger:440]:   Level: DEBUG
[C][logger:441]:   Log Baud Rate: 115200
[C][logger:443]:   Hardware UART: UART0
[C][captive_portal:088]: Captive Portal:
[C][web_server:168]: Web Server:
[C][web_server:169]:   Address: esp32-device-1.local:80
[C][mdns:115]: mDNS:
[C][mdns:116]:   Hostname: esp32-device-1
[C][ota:097]: Over-The-Air Updates:
[C][ota:098]:   Address: esp32-device-1.local:3232
[C][ota:101]:   Using Password.
[C][api:139]: API Server:
[C][api:140]:   Address: esp32-device-1.local:6053
[C][api:142]:   Using noise encryption: YES

Also, I cannot telnet to port 80.  The device does show in my router, but without the ".local" extension.  The name (without .local) resolves to the right IP address.

adamsd5, 2024-02-18:
FWIW, I resolved my issue by changing the WIFI network to use my "IoT" network instead of "Guest".  The guest gave a 192.168.200.X address and the IoT gave 192.168.1.X... I presume this was the key issue since my HA device is on 192.168.1.x.  That, or maybe the guest network didn't allow services to run listening on ports.