esphome/issues #4205: Delay in controlling BLE switch due to gattc queue full

narenarora, 2023-02-19:
### The problem

I am using a esp32 board with esphome and switchmate config to control a switchmate toggle via BLE. Although it works, i always have a delay in the "turn on" or "turn off" commands due to continuous errors indicating gattc queue is full. Not sure how to fix this. Can one of you please help ? 

PS : I have no idea of the code for either project. I am just finding my way by searching others fixes.

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Board

generic esp32dev

### Component causing the issue

not sure

### Example YAML snippet

```yaml
substitutions:
  name: esp32-ble-proxy-1
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false

  includes:
    - switchmate.h

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32_ble_tracker:
  scan_parameters:
    interval: 2000ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true
  cache_services: true

api:
logger:
ota:
improv_serial:

# SwitchMate-ESPHome Config (requires switchmate.h)
ble_client:
  # You can find these through the Switchmate app
  - mac_address: <mac_addr>
    id: bedroom_light_esp

custom_component:
- lambda: |-
    return {
      new SwitchmateController("Lights (Controller)", false, 1000, bedroom_light_esp)
    };
[... 244 lines cut ...]

[21:34:59][D][esp-idf:000]: E (190109) BT_APPL: bta_gattc_enqueue(), the gattc command queue is full.

[21:35:03][D][esp-idf:000]: E (194102) BT_APPL: bta_gattc_enqueue(), the gattc command queue is full.

[21:35:03][D][esp-idf:000]: E (194105) BT_APPL: bta_gattc_enqueue(), the gattc command queue is full.

[21:35:04][D][esp-idf:000]: E (195102) BT_APPL: bta_gattc_enqueue(), the gattc command queue is full.

[21:35:06][D][esp-idf:000]: E (197102) BT_APPL: bta_gattc_enqueue(), the gattc command queue is full.

[21:35:07][D][esp-idf:000]: E (198102) BT_APPL: bta_gattc_enqueue(), the gattc command queue is full.

[21:35:08][D][esp-idf:000]: E (199102) BT_APPL: bta_gattc_enqueue(), the gattc command queue is full.

[21:35:10][D][esp-idf:000]: E (201108) BT_APPL: bta_gattc_enqueue(), the gattc command queue is full.

[21:35:11][D][esp-idf:000]: E (202107) BT_APPL: bta_gattc_enqueue(), the gattc command queue is full.

[21:35:12][D][esp-idf:000]: E (203102) BT_APPL: bta_gattc_enqueue(), the gattc command queue is full.

[21:35:14][D][esp-idf:000]: E (205101) BT_APPL: bta_gattc_enqueue(), the gattc command queue is full.

[21:35:14][D][esp-idf:000]: E (205108) BT_APPL: bta_gattc_enqueue(), the gattc command queue is full.
```

distante, 2023-12-16:
@narenarora did you found out a fix?

narenarora, 2023-12-19:
> @narenarora did you found out a fix?

I just updated the ESPHome version and re-installed it on my ESP32. That fixed the issue for me.
I don't have the delay , or these messages in the logs anymore.
Although, the battery % seems to be stuck at 11% for me. That doesn't get updated. I have run through a couple batteries, and its still stuck at 11%.

Just FYI - I have updated the ESPHome version a couple times post this, and still no issues.
Currently I'm on 2023.11.3 and its still working okay for me.

closed 2023-12-19 by narenarora