esphome/issues #4208: ESP32-C3 with IRremoteESP8266 problem

phoenixxie0, 2023-02-19:
### The problem

ESP32-C3 with IRremoteESP8266 problem.
We are already aware of the issue that occurs when compiling with the crankyoldgit/IRremoteESP8266 @ 2.7.12 version in ESP32-C3.
`/data/esp-ktyw-dt-1780/.piolibdeps/esp-ktyw-dt-1780/IRremoteESP8266/src/IRrecv.cpp: In member function 'uint32_t IRrecv::ticksLow(uint32_t, uint8_t, uint16_t)':
/data/esp-ktyw-dt-1780/.piolibdeps/esp-ktyw-dt-1780/IRremoteESP8266/src/IRrecv.cpp:926:8: error: no matching function for call to 'max(int32_t, int)'
       0));
        ^
`
But When we set -IRremoteESP8266 @ 2.8.4, it still downloads version 2.7.12 for compilation, as shown below.
`esphome:`
 ` name: esp-ktyw-dt-1780`
`  platformio_options:`
`    board_build.flash_mode: dio`
`    board_build.mcu: esp32c3`
`  libraries:`
`    # use the git version of a library used by a component`
`    - IRremoteESP8266@^2.8.4`

`Library Manager: Installing crankyoldgit/IRremoteESP8266 @ 2.7.12
INFO Installing crankyoldgit/IRremoteESP8266 @ 2.7.12
Downloading  [##----------------------------------]    7%  00:01:23`

And build FAILED with the same problem
/data/esp-ktyw-dt-1780/.piolibdeps/esp-ktyw-dt-1780/IRremoteESP8266@2.7.12/src/IRrecv.cpp:926:8: error: no matching function for call to 'max(int32_t, int)'
       0));

### Which version of ESPHome has the issue?

v2023.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.2.5

### What platform are you using?

ESP32

### Board

ESP32-C3

### Component causing the issue

IRremoteESP8266

### Example YAML snippet

```yaml
esphome:
  name: esp-ktyw-dt-1780
  platformio_options:
    board_build.flash_mode: dio
    board_build.mcu: esp32c3
  libraries:
    # use the git version of a library used by a component
    - IRremoteESP8266@^2.8.4

esp32:
  board: esp32-c3-devkitm-1
  variant: ESP32C3
  framework:
    type: arduino
    version: latest
    platform_version: '6.0.0'

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "123"
    password: "123123123"

captive_portal:

# Example configuration entry
web_server:
  port: 80

i2c:
   - id: bus
     sda: GPIO5
     scl: GPIO4
     scan: true

# Example configuration entry
remote_transmitter:
  pin: GPIO10
  carrier_duty_percent: 50%

climate:
  - platform: heatpumpir
    protocol: hitachi
    horizontal_default: auto
    vertical_default: auto
    max_temperature: 30
    min_temperature: 16       # adjust to match your AC unit!
    name: "Living Room AC"

sensor:
  - platform: sht3xd
    i2c_id: bus
    temperature:
      name: "Living Room Temperature"
    humidity:
      name: "Living Room Humidity"
    address: 0x44
    update_interval: 5s
```

### Anything in the logs that might be useful for us?

```txt
Library Manager: Installing crankyoldgit/IRremoteESP8266 @ 2.7.12
INFO Installing crankyoldgit/IRremoteESP8266 @ 2.7.12
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%
Library Manager: IRremoteESP8266@2.7.12 has been installed!
INFO IRremoteESP8266@2.7.12 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40400.0
INFO Installing platformio/tool-scons @ ~4.40400.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-scons@4.40400.0 has been installed!
INFO tool-scons@4.40400.0 has been installed!
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Library Manager: Installing IRremoteESP8266 @ ^2.8.4

Library Manager: IRremoteESP8266@2.8.4 has been installed!
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- IRremoteESP8266 @ 2.8.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- HeatpumpIR @ 1.0.20
|   |-- IRremoteESP8266 @ 2.8.4
|-- IRremoteESP8266 @ 2.7.12
```

phoenixxie0, 2023-02-19:
The additional log
Library Manager: IRremoteESP8266@2.8.4 has been installed!
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- IRremoteESP8266 @ 2.8.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- HeatpumpIR @ 1.0.20
|   |-- IRremoteESP8266 @ 2.8.4
|-- IRremoteESP8266 @ 2.7.12

/data/esp-ktyw-dt-1780/.piolibdeps/esp-ktyw-dt-1780/IRremoteESP8266@2.7.12/src/IRrecv.cpp: In member function 'uint32_t IRrecv::ticksLow(uint32_t, uint8_t, uint16_t)':
/data/esp-ktyw-dt-1780/.piolibdeps/esp-ktyw-dt-1780/IRremoteESP8266@2.7.12/src/IRrecv.cpp:926:8: error: no matching function for call to 'max(int32_t, int)'
       0));
        ^

ferbulous, 2023-11-15:
Hi @phoenixxie0 were you able to resolve this?

phoenixxie0, 2023-11-17:
@ferbulous Nope.

predam, 2024-01-13:
Workaround:

```yml
esphome:
  name: ir-controller
  friendly_name: IR controller
  libraries:
    # use the git version of a library used by a component
    - IRremoteESP8266=https://github.com/predam/IRremoteESP8266.git#v2.8.6
```