esphome/issues #4209: ld2410 example values

n1kDon, 2023-02-19:
### The problem

The example threshold values of the # Example configuration entry are very uncommon, with them there is no use of the still_energy value etc. They should be changed to the standart values as described on "Configuration variables":

ld2410:
  timeout: 5s
  max_move_distance : 4.5m
  max_still_distance: 4.5m
  g0_move_threshold: 50
  g0_still_threshold: 0
  g1_move_threshold: 50
  g1_still_threshold: 0
  g2_move_threshold: 40
  g2_still_threshold: 40
  g3_move_threshold: 40
  g3_still_threshold: 40
  g4_move_threshold: 40
  g4_still_threshold: 40
  g5_move_threshold: 40
  g5_still_threshold: 40
  g6_move_threshold: 30
  g6_still_threshold: 15
  g7_move_threshold: 30
  g7_still_threshold: 15
  g8_move_threshold: 30
  g8_still_threshold: 15

### Which version of ESPHome has the issue?

2023.2.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

-

### What platform are you using?

ESP8266

sebcaps, 2023-03-02:
Indeed they are uncommon and must be updated to comply with your needs & setup.
Next version of the component will remove this config and allow config directly from HA.
Hope it helps.

regevbr, 2023-03-03:
> Indeed they are uncommon and must be updated to comply with your needs & setup. Next version of the component will remove this config and allow config directly from HA. Hope it helps.

see https://github.com/esphome/esphome/pull/4434 which you can start using right away