esphome/issues #4211: Dev addon was not published in over a month

regevbr, 2023-02-19:
### The problem

Docs state that the dev addon for HA is published daily, but it was not updated in over a month. This makes testing with the latest dev branch impossible.
Trying to specify a different fork in the configuration instead, fails:

```
Obtaining file:///esphome
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
    Preparing wheel metadata: started
    Preparing wheel metadata: finished with status 'done'
ERROR: Exception:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/base_command.py", line 223, in _main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/req_command.py", line 180, in wrapper
    return func(self, options, args)
  File "/usr/lib/python3/dist-packages/pip/_internal/commands/install.py", line 346, in run
    requirement_set = resolver.resolve(
  File "/usr/lib/python3/dist-packages/pip/_internal/resolution/resolvelib/resolver.py", line 101, in resolve
    r = self.factory.make_requirement_from_install_req(
  File "/usr/lib/python3/dist-packages/pip/_internal/resolution/resolvelib/factory.py", line 306, in make_requirement_from_install_req
    cand = self._make_candidate_from_link(
  File "/usr/lib/python3/dist-packages/pip/_internal/resolution/resolvelib/factory.py", line 155, in _make_candidate_from_link
    self._editable_candidate_cache[link] = EditableCandidate(
  File "/usr/lib/python3/dist-packages/pip/_internal/resolution/resolvelib/candidates.py", line 324, in __init__
    super(EditableCandidate, self).__init__(
  File "/usr/lib/python3/dist-packages/pip/_internal/resolution/resolvelib/candidates.py", line 144, in __init__
    self.dist = self._prepare()
  File "/usr/lib/python3/dist-packages/pip/_internal/resolution/resolvelib/candidates.py", line 222, in _prepare
    dist = self._prepare_distribution()
  File "/usr/lib/python3/dist-packages/pip/_internal/resolution/resolvelib/candidates.py", line 335, in _prepare_distribution
    return self._factory.preparer.prepare_editable_requirement(self._ireq)
  File "/usr/lib/python3/dist-packages/pip/_internal/operations/prepare.py", line 579, in prepare_editable_requirement
    req.check_if_exists(self.use_user_site)
  File "/usr/lib/python3/dist-packages/pip/_internal/req/req_install.py", line 433, in check_if_exists
    existing_dist = get_distribution(self.req.name)
  File "/usr/lib/python3/dist-packages/pip/_internal/utils/misc.py", line 549, in get_distribution
    pkg_resources.working_set.require(req_name)
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 791, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (zeroconf 0.47.1 (/usr/local/lib/python3.9/dist-packages), Requirement.parse('zeroconf==0.47.3'), {'esphome'})
```

### Which version of ESPHome has the issue?

dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.3

### What platform are you using?

ESP32

regevbr, 2023-02-19:
I just uninstalled and reinstalled and set up the `dev` branch and it worked so maybe just a glitch in the Matrix :-)

regevbr, 2023-02-21:
I just restarted the addon to get the latest dev and it failed again on same error. Seems there is an issue when trying to redownload... uninstalling fixed the issue again

jesserockz, 2023-06-22:
Sorry for no reply. 

The dev addon is not exactly "published", its an addon that is built locally on your instance and the base docker image used when it build is rebuilt daily. So you have to go into the addon page and hit `rebuild` to get the latest image.