esphome/issues #4214: New setting `friendly_name` not applying as a prefix

thibmaek, 2023-02-20:
### The problem

I removed all the manual interpolation from my sensors because of the new `friendly_name` setting in 2023.2

Updated my devices to 2023.2 but now it just removed the prefix, instead of prefixing with it like the changelogs says:

> This name will be used for the config entry, the device name, and will be automatically prefixed to all of the entities where needed by Home Assistant

You can see the old entities and the new ones after the update here:

[image]

My device config looks like this:

```yaml
substitutions:
  device_name: 'ikea-vindriktning-bureau-thib'
  friendly_name: 'Air Quality Bureau Thib'

esp8266:
  board: d1_mini

packages:
  ikea_vindriktning: !include ./lib/ikea_vindriktning.yaml
```

I then use a shared config called `./lib/ikea_vindriktning.yaml`, that looked like this for 2022.12:

```yaml
globals:
  - id: warn_threshold
    type: int
    initial_value: '35'
  - id: danger_treshold
    type: int
    initial_value: '55'

uart:
  rx_pin: D2
  baud_rate: 9600

binary_sensor:
  # Ref: https://blissair.com/what-is-pm-2-5.htm
  - platform: template
    name: "${friendly_name} Safety"
    device_class: safety
    lambda: |-
      if (id(pm2_5_sensor).state < id(danger_treshold)) {
        return false;
      } else {
        return true;
      }

sensor:
  - platform: pm1006
    pm_2_5:
      name: "${friendly_name}"
      id: pm2_5_sensor

text_sensor:
  - platform: template
    name: "${friendly_name} Status"
    icon: mdi:air-filter
    lambda: |-
      if (id(pm2_5_sensor).state < id(warn_threshold)) {
        return {"safe"};
      } else if(id(pm2_5_sensor).state < id(danger_treshold)) {
        return {"risky"};
      } else {
        return {"dangerous"};
      }
```

And now looks like this for 2023.2:

```yaml
globals:
  - id: warn_threshold
    type: int
    initial_value: '35'
  - id: danger_treshold
    type: int
    initial_value: '55'

uart:
  rx_pin: D2
  baud_rate: 9600

binary_sensor:
  # Ref: https://blissair.com/what-is-pm-2-5.htm
  - platform: template
    name: "Safety"
    device_class: safety
    lambda: |-
      if (id(pm2_5_sensor).state < id(danger_treshold)) {
        return false;
      } else {
        return true;
      }

sensor:
  - platform: pm1006
    pm_2_5:
      name: "PM2.5"
      id: pm2_5_sensor

text_sensor:
  - platform: template
    name: "Status"
    icon: mdi:air-filter
    lambda: |-
      if (id(pm2_5_sensor).state < id(warn_threshold)) {
        return {"safe"};
      } else if(id(pm2_5_sensor).state < id(danger_treshold)) {
        return {"risky"};
      } else {
        return {"dangerous"};
      }
```

### Which version of ESPHome has the issue?

2023.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP8266

### Board

D1 Mini

### Example YAML snippet

```yaml
globals:
  - id: warn_threshold
    type: int
    initial_value: '35'
  - id: danger_treshold
    type: int
    initial_value: '55'

uart:
  rx_pin: D2
  baud_rate: 9600

binary_sensor:
  # Ref: https://blissair.com/what-is-pm-2-5.htm
  - platform: template
    name: "Safety"
    device_class: safety
    lambda: |-
      if (id(pm2_5_sensor).state < id(danger_treshold)) {
        return false;
      } else {
        return true;
      }

sensor:
  - platform: pm1006
    pm_2_5:
      name: "PM2.5"
      id: pm2_5_sensor

text_sensor:
  - platform: template
    name: "Status"
    icon: mdi:air-filter
    lambda: |-
      if (id(pm2_5_sensor).state < id(warn_threshold)) {
        return {"safe"};
      } else if(id(pm2_5_sensor).state < id(danger_treshold)) {
        return {"risky"};
      } else {
        return {"dangerous"};
      }
```

trvrnrth, 2023-02-27:
Are you definitely setting the [config](https://esphome.io/components/esphome.html#configuration-variables) and not just a substitution? 

Bear in mind that HA will hide the prefix on the device page based on the device name.

thibmaek, 2023-02-27:
Auch. That was indeed it. Sorry for not checking enough, could've spotted this.

closed 2023-02-27 by thibmaek