esphome/issues #4216: Pico W test installation compiles, but fails to load.

bwims, 2023-02-20:
### The problem

I have a Docker installation of ESPHome running on a RPi 4.  
A Pico W is attached to a usb port.

After successfully compiling the standard test script provided, the system fails to upload.

### Which version of ESPHome has the issue?

v2023.2.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

RP2040

### Board

pico w

### Component causing the issue

Upload

### Example YAML snippet

```yaml
esphome:
  name: test
  friendly_name: test

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    # NB the .git on the following URL left off deliberately, otherwise it fails.
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "haJALLqkGytxTBq0yyfU+wgpMzBRD9FR0mk7BeZ5HFk="

ota:
  password: "d013f248668e5d59dc9671092b707cbc"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"
    password: "H2LM00ADsdki"
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/test.yaml...
[... 342 lines cut ...]
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda      8:0    0 111.8G  0 disk
├─sda1   8:1    0   128M  0 part /boot
└─sda2   8:2    0 111.7G  0 part /
sdb      8:16   1   128M  0 disk
└─sdb1   8:17   1   128M  0 part

<<<during>>>

root@DietPi:~# lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda      8:0    0 111.8G  0 disk
├─sda1   8:1    0   128M  0 part /boot
└─sda2   8:2    0 111.7G  0 part /

<<<after>>>

root@DietPi:~# lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda      8:0    0 111.8G  0 disk
├─sda1   8:1    0   128M  0 part /boot
└─sda2   8:2    0 111.7G  0 part /
sdb      8:16   1   128M  0 disk
└─sdb1   8:17   1   128M  0 part
```

MorganMLGman, 2023-02-28:
I have the same issue, tried two linux machines with the same results. 

ESPHome version: 2023.2.4

Logs:
`morgan@morgan-rpi:~/esphome $ esphome run pico_weather.yaml --device /dev/sdb1
INFO Reading configuration pico_weather.yaml...
WARNING Found 'hVC*2H67$RUd2e@J' (see wifi->ap->password) which looks like a substitution, but 'RUd2e' was not declared
WARNING Found 'hVC*2H67$RUd2e@J' (see wifi->ap->password) which looks like a substitution, but 'RUd2e' was not declared
WARNING 'pico_weather': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Generating C++ source...
INFO Compiling app...
Processing pico_weather (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
 - framework-arduinopico @ 1.20604.0 (2.6.4)
 - tool-rp2040tools @ 1.0.2
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 1.0.0
|   |-- Updater @ 1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- LittleFS @ 0.1.0
|   |   |-- PicoOTA @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |-- MD5Builder @ 1.0.0
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- lwIP_CYW43 @ 1
|   |   |-- SPI @ 1.0
|   |-- SPI @ 1.0
|-- LEAmDNS @ 1.2
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- WiFi @ 1.0.0
|   |   |-- Updater @ 1.0
|   |   |   |-- MD5Builder @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |   |   |-- PicoOTA @ 1.0.0
|   |   |   |   |-- LittleFS @ 0.1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- lwIP-Ethernet @ 1
|   |   |   |-- lwIP_CYW43 @ 1
|   |   |   |   |-- SPI @ 1.0
|   |   |   |-- SPI @ 1.0
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|-- Updater @ 1.0
|   |-- MD5Builder @ 1.0.0
|   |-- LittleFS @ 0.1.0
|   |-- PicoOTA @ 1.0.0
|   |   |-- LittleFS @ 0.1.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- MD5Builder @ 1.0.0
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
RAM:   [===       ]  26.4% (used 69220 bytes from 262144 bytes)
Flash: [=====     ]  47.5% (used 496616 bytes from 1044480 bytes)
Building .pioenvs/pico_weather/firmware.bin.signed
====================================================================== [SUCCESS] Took 6.89 seconds ======================================================================
INFO Successfully compiled program.
Processing pico_weather (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
 - framework-arduinopico @ 1.20604.0 (2.6.4)
 - tool-mklittlefs-rp2040-earlephilhower @ 5.100300.230216 (10.3.0)
 - tool-openocd-rp2040-earlephilhower @ 5.100300.230216 (10.3.0)
 - tool-rp2040tools @ 1.0.2
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 1.0.0
|   |-- Updater @ 1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- LittleFS @ 0.1.0
|   |   |-- PicoOTA @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |-- MD5Builder @ 1.0.0
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- lwIP_CYW43 @ 1
|   |   |-- SPI @ 1.0
|   |-- SPI @ 1.0
|-- LEAmDNS @ 1.2
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- WiFi @ 1.0.0
|   |   |-- Updater @ 1.0
|   |   |   |-- MD5Builder @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |   |   |-- PicoOTA @ 1.0.0
|   |   |   |   |-- LittleFS @ 0.1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- lwIP-Ethernet @ 1
|   |   |   |-- lwIP_CYW43 @ 1
|   |   |   |   |-- SPI @ 1.0
|   |   |   |-- SPI @ 1.0
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|-- Updater @ 1.0
|   |-- MD5Builder @ 1.0.0
|   |-- LittleFS @ 0.1.0
|   |-- PicoOTA @ 1.0.0
|   |   |-- LittleFS @ 0.1.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- MD5Builder @ 1.0.0
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
RAM:   [===       ]  26.4% (used 69220 bytes from 262144 bytes)
Flash: [=====     ]  47.5% (used 496616 bytes from 1044480 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, picoprobe, picotool, raspberrypi-swd
CURRENT: upload_protocol = picotool
Looking for upload port...
Using manually specified: /dev/sdb1
Forcing reset using 1200bps open/close on port /dev/sdb1
Uploading .pioenvs/pico_weather/firmware.elf
rp2040load 1.0.1 - compiled with go1.15.8
.....................
*** [upload] Error 1
====================================================================== [FAILED] Took 17.31 seconds ======================================================================`

MorganMLGman, 2023-02-28:
[image]

I think you need to manualy copy this file to rasberry pico w 

I had the same problem, that my pico w wouldn't flash with the same error.

I tried connecting pico w with `BOOTSEL` button pressed and then manualy copied this `.uf2` file to it. After a while the pico was connected to my wifi.

bwims, 2023-03-01:
Thanks!  I won't be able to try this today, but will asap.

ex-fil, 2023-11-27:
I solved a issue similar to this one by running this docker command:

~~~bash
docker run --privileged -v $(pwd):/data -v /dev:/dev -v /media:/media --rm -it <my_pio_img>
~~~

Probably you have to specify also media as volume.

bwims, 2023-11-28:
> ```shell
> ```shell
> docker run --privileged -v $(pwd):/data -v /dev:/dev -v /media:/media --rm -it <my_pio_img>
> ```
> 
> 
>     
>       
>     
> 
>       
>     
> 
>     
>   
> ```

Thanks for that.  I'm not sure what it does though.   I must admit that I haven't been able to get back to the pico project since, so my thoughts are a bit rusty.

How does this help to load the compiled image onto the pico and run it?

Thanks

B.

bdraco, 2026-03-12:
Closing — this is the same PlatformIO/picotool serial upload issue as #4070, not an ESPHome bug. For initial flashing, hold BOOTSEL and copy the .uf2 file manually, then use OTA for subsequent updates. The Docker container also needs proper device/volume mappings to access USB devices.