esphome/issues #4219: PCA9685 not working after 2023.2.5 Update

total1ty, 2023-02-20:
### The problem

I have a ESP8266 with two PCA9685 attached via I2C, one at default address 0x40 and one at 0x41.
This worked flawlessly until yesterday when I rolled out the 2023.2.5 Update.

Now, the PCA9685 at 0x40 default address doesn't work anymore, as in its outputs stay off no matter what I do.
There are no indications in the log or elsewhere that the PCA should not work: it gets initialized correctly like the other one, the entities come online, I can even switch the lights in HA frontend. ESPHome Log also correctly shows the Lights switching on/off. But all the outputs stay off all the time.
This only affects the 0x40 PCA9685, the other one at address 0x41 works fine.

### Which version of ESPHome has the issue?

2023.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP8266

### Board

nodemcu v2

### Component causing the issue

pca9685

### Example YAML snippet

```yaml
(I truncated the config to a few outputs for readability)

substitutions:
  node_name: fnv2-studio-ceiling
  node_name_display: "Studio Ceiling"

esphome:
  name: ${node_name}

esp8266:
  board: nodemcuv2

logger:

api:
  encryption:
    key: !secret api_key

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  ap:
    ssid: "(F) ${node_name_display}"
    password: !secret wifi_fallback_password

captive_portal:

time:
  - platform: homeassistant
    id: esptime

i2c:
  sda: D2
  scl: D1
  scan: true

pca9685:
  - id: pca9685_1
    frequency: 500
    address: 0x40
  - id: pca9685_2
    frequency: 500
    address: 0x41

output:
  - platform: pca9685
    id: 'pca1_ch0'
    pca9685_id: pca9685_1
    channel: 0
    
  - platform: pca9685
    id: 'pca1_ch1'
    pca9685_id: pca9685_1
    channel: 1

  - platform: pca9685
    id: 'pca2_ch0'
    pca9685_id: pca9685_2
    channel: 0
    
  - platform: pca9685
    id: 'pca2_ch1'
    pca9685_id: pca9685_2
    channel: 1

light:
# this light would not work
  - platform: monochromatic
    name: "${node_name_display} Spot 1"
    output: pca1_ch0
# this light would work fine
  - platform: monochromatic
    name: "${node_name_display} Spot 2"
    output: pca2_ch0
```

### Anything in the logs that might be useful for us?

```txt
Log Output shows working and non-working lights switching on and off correctly:

[19:31:21][D][light:035]: 'Studio Ceiling Spot 1' Setting:
[19:31:21][D][light:046]:   State: ON
[19:31:21][D][light:084]:   Transition length: 1.0s
[19:31:22][D][light:035]: 'Studio Ceiling Spot 2' Setting:
[19:31:22][D][light:046]:   State: ON
[19:31:22][D][light:084]:   Transition length: 1.0s
```

Ferao, 2023-02-25:
I have exactly the same issue, let me know if help is needed for testing

total1ty, 2023-02-26:
I've been doing some further testing. When I remove all entries using the 2nd PCA, the first PCA works fine. Which further confirms this being a software problem rather than a hardware problem

total1ty, 2023-02-26:
Today I also tested changing the order in which the two PCA9685 get defined.

It seems like the PCA defined second is the one working. I don't have more than 2 available at the moment, but I presume something gets "overwritten", and so only the last defined PCA is working.

I don't think anything was changed in the PCA9685 module, so I guess it's a bug in the way ESPHome compiles the stuff?

Also, when I hit "Validate", the config looks fine, which also rules out an error in the package/substitutions process -> suggesting that the issue lies deeper

total1ty, 2023-02-27:
Temporary workaround until this is fixed, add this to your config:
```
external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome.git
      ref: 2022.12.8
    components: [ pca9685 ]
```

vmotoska, 2023-02-28:
If I understand it correctly the problem is the reset functionality of the pca. As reset is beeing sent  to 0x00 which resets all the devices. So during initialization of the second device the first one gets also reseted, which leaves the device in low power mode so ( oscillator off ).

Swwils, 2023-03-17:
It looks like in the cpp its being reset using the general call address (0x00), which results in resetting all PCA9685 devices on the I2C bus.

To fix this, could you modify the code to reset each PCA9685 device individually by writing to their MODE1 register?

closed 2023-04-17 by ssieb

dizherui, 2023-05-25:
It seems that no one has come to solve this problem? I have used several PCA9685 modules, and now I dare not upgrade the firmware.

total1ty, 2023-07-06:
it's fixed, see https://github.com/esphome/issues/issues/4186#issuecomment-1622808256