esphome/issues #4220: Wiegand tag scanned: invalid parity (and tag not scanned into HA)

Quarco, 2023-02-20:
### The problem

Installed Wiegand tag reader (FSTech 13,56 MHz).

Problems are: 

- when a tag is scanned HA reports an "Invalid Parity
- it seems the on_tag statement is not executed 

[image]

Home Assistant 2023.2.5
Supervisor 2023.01.1
Operating System 9.5
Frontend-versie: 20230202.0 - latest

### Which version of ESPHome has the issue?

2023.2.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP32

### Board

esp32-dev + external antenna

### Component causing the issue

Wiegand keypad / tag scanner

ssieb, 2023-02-20:
Yes, this is a known issue.  Somehow I broke the parity checking just before it was merged.  I thought it was tested, but apparently not.  I will fix it soon.

ssieb, 2023-02-21:
Try adding this to your yaml and see if that fixes it:
```yaml
external_components:
  - source: github://ssieb/esphome@wiegand
    components: [ wiegand ]
    refresh: 1min
```

Quarco, 2023-02-22:
Yes, that seems to remove the parity-error!

(MARCO = for my debugging / yaml)
[image]

[image]

(Seems to be a delay from about 1 sec)

closed 2023-02-22 by jesserockz

marcolinux46, 2023-04-28:
> Yes, that seems to remove the parity-error!
> 
> (MARCO = for my debugging / yaml) [image]
> 
> [image]
> 
> (Seems to be a delay (after scan-tag) of about 1 sec)

@Quarco do You remember the procedure to convert the reader in wiegand 26/34 mode? 
Today i opened an issue case for Unknown bit problem with the reader.

Quarco, 2023-04-28:
The fontsize of the manual was so tiny, I had to take photo's of it :-)
If you need more, let me know!

[image]

marcolinux46, 2023-04-28:
@Quarco thank you for your quick reply,

your information paper is same as mine.

So if i undestood:
1 - power-off the reader , short-circuit RESET and power-on until consequential bip.
2 - Power-off again and connect D0 or D1 to ground and power-on
3 - Wait until the green light starts to blink and during blink disconnect D0 or D1 and ground immediately.
4 - Click any button only one time (what do they mean for "click once to complete the conversion"? )

This is the right procedure?

Thank you for your support

Quarco, 2023-04-28:
I did follow the procedure and it wasn't that hard after all. Had it changed the first time I tried it.
Tried both modes trying to fix my problem in february. Immediately the debug / log in HA showed 26 / 34 .

Just give it a try, i'm sure you'll get it :-)

marcolinux46, 2023-04-28:
> I did follow the procedure and it wasn't that hard after all. Had it changed the first time I tried it. Tried both modes trying to fix my problem in february. Immediately the debug / log in HA showed 26 / 34 .
> 
> Just give it a try, i'm sure you'll get it :-)

Thank you for your reply.
"Immediately the debug / log in HA showed 26 / 34" --> During the boot procedure?

Only one question....after reset i have to reboot the reader?

Thank you

Quarco, 2023-04-28:
Reboot can't do any harm :-)
After the procedure the log in HA should show 34 or 26 in the logs.

marcolinux46, 2023-04-29:
> Reboot can't do any harm :-) After the procedure the log in HA should show 34 or 26 in the logs.

Procedure done again.....Now everything is working.
I connected also the ESP GND (thanks to @ssieb) that was missed yesterday.

@Quarco Thank you for your support

laxfiend77, 2023-07-16:
I'm having the invalid parity issue currently with one of my esphome setups. Admittedly this is my first go round with these. I've tried pasting what you said into my yaml....assuming it's supposed to go into the esp home yaml for that reader and not my homeassistant config yaml or somewhere else.

ssieb, 2023-07-16:
@laxfiend77 this was fixed a long time ago.  Do you have the latest esphome version?

zipposi, 2023-08-04:
I'm having the same issue. Don't know exactly since when, but a few versions eatrlier was functioning

```
[07:51:03][D][wiegand:073]: received 26-bit tag: 10663461
[07:51:03][W][wiegand:075]: invalid parity
```

ssieb, 2023-08-04:
What esphome version do you have?  Does the number match the tag?  Try swapping the data lines.