esphome/issues #4222: Unable to use mac_address and lambda together

steku, 2023-02-21:
### The problem

Both of these actions work individually but when I try to do them together it will not compile.

### Which version of ESPHome has the issue?

2023.2.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

esp32_ble_tracker

### Example YAML snippet

```yaml
esp32_ble_tracker:
  scan_parameters:
    duration: 60s
    active: false
  on_ble_advertise:
    - mac_address: EB:82:34:XX:XX:XX
      then:
        - lambda: |-
            std::string name = x.get_name();
            if (esphome::str_startswith(name, "level:")) {
              name.erase(0,7);
              std::string value = esphome::str_until(name.c_str(), ' ');
              id(propane_level_tank_1).publish_state(atof(value.c_str()));
            }
    - lambda: |-
        parseAdvertisement(x);
```

### Anything in the logs that might be useful for us?

```txt
Failed config

esp32_ble_tracker: [source /config/esphome/esphome-master.yaml:39]
  scan_parameters: 
    duration: 60s
    active: False
  on_ble_advertise: 
    - 
      Unable to find action with the name 'mac_address'.
```

ssieb, 2023-02-21:
You can't combine the two forms.  You either filter by mac address (multiple times if you want) or you have a general lambda.  You can't have both.

jesserockz, 2023-06-22:
You have to be consistent with the structure of the items in the list
```yaml
esp32_ble_tracker:
  scan_parameters:
    duration: 60s
    active: false
  on_ble_advertise:
    - mac_address: EB:82:34:XX:XX:XX
      then:
        - lambda: |-
            std::string name = x.get_name();
            if (esphome::str_startswith(name, "level:")) {
              name.erase(0,7);
              std::string value = esphome::str_until(name.c_str(), ' ');
              id(propane_level_tank_1).publish_state(atof(value.c_str()));
            }
    - then:
        - lambda: |-
            parseAdvertisement(x);
```