esphome/issues #4223: Socket operation failed: BAD_INDICATOR errno=11 After updating to 2023.2.3

KidA001, 2023-02-21:
### The problem

I upgraded multiple ESP (both 8266 and 32's) to the newest `2023.2.3` version. One of my Wemos D1 Mini's started throwing this error when I connect to the logs: `Socket operation failed: BAD_INDICATOR errno=11 After updating to 2023.2.3`

The device no longer shows up in HomeAssistant

### Which version of ESPHome has the issue?

2023.2.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

API

### Example YAML snippet

```yaml
esphome:
  name: well-pump-switch-v2

esp8266:
  board: d1_mini

# Enable logging, api, ota, captive_portal
logger:
api:
  encryption:
    key: !secret encryption_key
ota:
  password: !secret ota_password
captive_portal:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "WellPumpFallbackAp"
    password: !secret ap_pw

switch:
  - platform: gpio
    name: Well Pump Switch v2
    pin: GPIO5
```

### Anything in the logs that might be useful for us?

```txt
[08:30:12][I][app:102]: ESPHome version 2023.2.3 compiled on Feb 21 2023, 08:29:20
[08:30:12][C][wifi:504]: WiFi:
[08:30:12][C][wifi:362]:   Local MAC: 40:91:51:4C:F2:2C
[08:30:12][C][wifi:363]:   SSID: [redacted]
[08:30:12][C][wifi:364]:   IP Address: 192.168.20.184
[08:30:12][C][wifi:365]:   BSSID: [redacted]
[08:30:12][C][wifi:367]:   Hostname: 'well-pump-switch-v2'
[08:30:12][C][wifi:369]:   Signal strength: -40 dB ▂▄▆█
[08:30:12][C][wifi:373]:   Channel: 6
[08:30:12][C][wifi:374]:   Subnet: 255.255.255.0
[08:30:12][C][wifi:375]:   Gateway: 192.168.20.1
[08:30:12][C][wifi:376]:   DNS1: 192.168.20.1
[08:30:12][C][wifi:377]:   DNS2: 0.0.0.0
[08:30:12][C][logger:293]: Logger:
[08:30:12][C][logger:294]:   Level: DEBUG
[08:30:12][C][logger:295]:   Log Baud Rate: 115200
[08:30:12][C][logger:296]:   Hardware UART: UART0
[08:30:12][C][switch.gpio:076]: GPIO Switch 'Well Pump Switch v2'
[08:30:12][C][switch.gpio:098]:   Restore Mode: restore defaults to OFF
[08:30:12][C][switch.gpio:031]:   Pin: GPIO5
[08:30:12][C][captive_portal:088]: Captive Portal:
[08:30:12][C][mdns:108]: mDNS:
[08:30:12][C][mdns:109]:   Hostname: well-pump-switch-v2
[08:30:12][C][ota:093]: Over-The-Air Updates:
[08:30:12][C][ota:094]:   Address: well-pump-switch-v2.local:8266
[08:30:12][C][ota:097]:   Using Password.
[08:30:12][C][api:138]: API Server:
[08:30:12][C][api:139]:   Address: well-pump-switch-v2.local:6053
[08:30:12][C][api:141]:   Using noise encryption: YES
[08:30:23][D][api:102]: Accepted 192.168.10.100
[08:30:24][D][api:102]: Accepted 192.168.10.100
[08:30:24][D][api:102]: Accepted 192.168.10.100
[08:30:24][D][api:102]: Accepted 192.168.10.100
[08:30:24][D][api:102]: Accepted 192.168.10.100
[08:30:27][D][api:102]: Accepted 192.168.10.100
[08:30:27][D][api:102]: Accepted 192.168.10.100
[08:30:27][D][api:102]: Accepted 192.168.10.100
[08:30:27][D][api:102]: Accepted 192.168.10.100
[08:30:27][W][api.connection:071]: 192.168.10.100: Socket operation failed: BAD_INDICATOR errno=11
[08:30:27][W][api.connection:071]: 192.168.10.100: Socket operation failed: HANDSHAKESTATE_SETUP_FAILED errno=11
[08:30:27][D][api.connection:918]: Home Assistant 2023.2.5 (192.168.10.100): Connected successfully
[08:30:27][D][api:102]: Accepted 192.168.10.100
[08:30:27][D][api.connection:918]: Home Assistant 2023.2.5 (192.168.10.100): Connected successfully
[08:30:27][D][api.connection:918]: Home Assistant 2023.2.5 (192.168.10.100): Connected successfully
[08:30:28][D][api.connection:918]: Home Assistant 2023.2.5 (192.168.10.100): Connected successfully
[08:30:28][D][api.connection:918]: Home Assistant 2023.2.5 (192.168.10.100): Connected successfully
[08:30:28][D][api.connection:918]: Home Assistant 2023.2.5 (192.168.10.100): Connected successfully
[08:30:28][D][api.connection:918]: Home Assistant 2023.2.5 (192.168.10.100): Connected successfully
[08:30:28][D][api.connection:918]: Home Assistant 2023.2.5 (192.168.10.100): Connected successfully
[08:30:33][D][api:102]: Accepted 192.168.10.100
[08:30:33][W][api.connection:071]: 192.168.10.100: Socket operation failed: BAD_INDICATOR errno=11
[08:31:33][D][api:102]: Accepted 192.168.10.100
[08:31:33][W][api.connection:071]: 192.168.10.100: Socket operation failed: BAD_INDICATOR errno=11
[08:32:33][D][api:102]: Accepted 192.168.10.100
[08:32:33][W][api.connection:071]: 192.168.10.100: Socket operation failed: BAD_INDICATOR errno=11
[08:33:33][D][api:102]: Accepted 192.168.10.100
[08:33:33][W][api.connection:071]: 192.168.10.100: Socket operation failed: BAD_INDICATOR errno=11
[08:34:33][D][api:102]: Accepted 192.168.10.100
[08:34:33][W][api.connection:071]: 192.168.10.100: Socket operation failed: BAD_INDICATOR errno=11
[08:35:03][I][ota:113]: Boot seems successful, resetting boot loop counter.
[08:35:33][D][api:102]: Accepted 192.168.10.100
[08:35:33][W][api.connection:071]: 192.168.10.100: Socket operation failed: BAD_INDICATOR errno=11
```

### Additional information

I have tried the following to no avail:

- re-installing the compiled code via OTA on the device
- re-installing the ESPHome via USB on the device
- updating the OTA password and installing it on the device via OTA
- cleaning build files and re-installing
- removing the device from HA (now it no longer shows up and cannot re-add)

Any help appreciated

ssieb, 2023-02-21:
That looks like HA is a little confused.  That's a lot of simultaneous connection attempts.
Did you try restarting HA?  And you can always add the device manually even if it isn't auto-detected.

KidA001, 2023-02-21:
well....a restart did it. didn't realize it had anything to do with HA. Thank you!

closed 2023-02-21 by KidA001

piffio, 2023-03-22:
Just had the same issue and this thread saved me lots of hours of troubleshooting

Manhem, 2023-03-29:
I have the same problem with my ESP32.
Nothing of the above is helping and now I'm out of idears.

The only thing I get to work is to do a totaly new configuration and upload it.
But if I disconnect the ESP32 for one hour or so, then I have the same problem again.

ssieb, 2023-03-29:
@Manhem , did you check the HA logs?

Manhem, 2023-03-30:
I reinstall everything yesterday.
It works so far even if the was no sound. It sounds terribly. Couldn't hear anything.
I shall try more today and see what happends.

genestealer, 2023-04-04:
I had the same error message, turns out I had not entered the new esphome api encryption key in home assistant, in the /config/integrations page.

MasterCATZ, 2023-04-21:
I have been having this 

```[W][api.connection:075]:` 192.168.2.14: Socket operation failed: BAD_INDICATOR errno=11```
since being forced to change from using "password: !secret api_password"
```
# Enable Home Assistant API
api:
    encryption:
    key: !secret key
```

ActuariesPy, 2023-04-22:
I have exactly the same. Don't know if it's because of recents upgrades but my esphome device are all working but i can't add news because of this error :  Socket operation failed: BAD_INDICATOR errno=11

Jooel1023, 2023-04-22:
I have exactly the same. Don't know if it's because of recents upgrades but my esphome device are all working but i can't add news because of this error : Socket operation failed: BAD_INDICATOR errno=11

NeoMatrixJR, 2023-04-24:
> I have been having this
> 
> `` [W][api.connection:075]:` 192.168.2.14: Socket operation failed: BAD_INDICATOR errno=11 `` since being forced to change from using "password: !secret api_password"
> 
> ```
> # Enable Home Assistant API
> api:
>     encryption:
>     key: !secret key
> ```

This doesn't look right... Someone correct me if I'm wrong, but I thought each device was supposed to have a different key.  This would make them all the same.  Also, your tabbing is wrong, so it's not likely getting the key right.  Should be
api:
  encryption:
    key: "somebase64key"

Jooel1023, 2023-04-24:
In my case, i just uninstalled EspHome and reinstalled and now everything is fine.

smoke007, 2023-05-04:
> try restarting HA

I just updated a device from the deprecated password option and switched to the new [encryption](https://esphome.io/components/api.html#configuration-variables) option and started seeing this event too.  As @ssieb recommended, restarting HA resolved this issue.  BTW, I tried restarting just ESPHome first, but that wasn't enough.

Now I'll continue updating the rest of my devices since my test one is updated successfully.  :)  Thanks

Ritchie3, 2023-06-19:
After updating your ESPHome device from the password to the api option,
go to  Settings->Devices,
a new Device is found. Provide it with the api key. Save.

this solved it for me. no restart needed.