esphome/issues #4224: Reading IR-Sensor via UART broken since v2023.2

BiasF, 2023-02-21:
### The problem

I read my Powermeter via UART_BUS with an IR Write/Read head. Since ESPHome 2023.02 it's not working anymore.
No IR Signals are visible in the debug log, also If I send IR codes via my TV Remote control.

I downgraded to ESPHome 2022.11 and it's now working again.

### Which version of ESPHome has the issue?

2023.02. (Maybe since 2022.12, I skipped that version)

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.02.

### What platform are you using?

ESP32

### Board

TTGO T-18

### Component causing the issue

TTL IR Read/Write head for Powermeter

### Example YAML snippet

```yaml
uart:
  id: uart_bus
  rx_pin: GPIO13
  baud_rate: 9600
  data_bits: 8
  parity: NONE
  stop_bits: 1
  debug:
```

### Anything in the logs that might be useful for us?

```txt
Nothing related in the log, also in verbose mode.
```

### Additional information

Downgrade to older Version fix the issue

matjahs, 2023-02-23:
Experiencing the same issue with a M5stack Atom Lite and a pm1006 on GPIO33.
I can see it initialize in de logs but after that, nothing. Not even with log level VERY_VERBOSE.

BiasF, 2023-03-16:
Any news here? 
Is this issue addressed in the ESPHome 2023.3?

BiasF, 2023-07-06:
Still not fixed in 2023.7 ... I had to upgrade because since homeassistant 2023.7 communication with ESPHome  2022.11 stoped working.