esphome/issues #4226: Waveshare E-paper Images rendered as black squares

JeroenTuinstra, 2023-02-21:
### The problem

Since version 2023.2.3 (not sure it started there, because I only update my firmware when necessary), images are rendered as black squares. These are PNG with a transparency layer. Before they would render correctly.

So the image is loaded as such: 

```
image:
  - file: "images/restafval2.png"
    id: Lrestafvalpic
    type: TRANSPARENT_BINARY
```
Then it is called upon like such in the display section:
```
it.image(255, 450, id(Lrestafvalpic));
```
This generates a black square.
This is the actual image used: 
[image]
This is the result:
[image]

### Which version of ESPHome has the issue?

2023.2.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.2.3

### What platform are you using?

ESP32

### Board

Firebeetle32

### Component causing the issue

display/it.image

### Example YAML snippet

```yaml
image:
  - file: "images/restafval2.png"
    id: Lrestafvalpic
    type: TRANSPARENT_BINARY

display:
  - platform: waveshare_epaper
    cs_pin: 22
    dc_pin: 21
    busy_pin: 5
    reset_pin: 13
    model: 7.50in-hd-b
    update_interval: never
    reset_duration: 2ms
    rotation: 90°
    id: epaper_display
    lambda: |-
      it.image(255, 450, id(Lrestafvalpic));
```

### Anything in the logs that might be useful for us?

```txt
No errors in the log.
```

JeroenTuinstra, 2023-02-21:
When replaced by a JPG image it works fine. So it seems that rendering of PNG images is broken. 

```
image:
  - file: "images/restafval2.jpg"
    id: Lrestafvalpic
```
This the actual JPG picture: 
[image]

And the result is: 
[image]

jdituicha, 2023-02-27:
I also have the same problem with an OLED display since the update to v2023.2.4.

guillempages, 2023-03-10:
@JeroenTuinstra @jdituicha  I was having the same issue, but seems like changing the image type from "TRANSPARENT_BINARY" to "TRANSPARENT_IMAGE" fixes the problem for me (also using PNGs)
I still have to test some more, but this looks promising; you might want to check that as well

jdituicha, 2023-03-25:
@guillempages tanks you for the references. "TRANSPARENT_BINARY" to "TRANSPARENT_IMAGE"  is my solutions. : )

guillempages, 2023-03-25:
@jdituicha, think of this as a temporary solution/workaround though; once my PR es phome/esphome#4600 gets merged, the TRANSPARENT_IMAGE type might get removed again (under discussion)

closed 2023-05-21 by jesserockz