esphome/issues #4227: Web Server table sliders too small

Uksa007, 2023-02-22:
### The problem

Hi,

I'm using the V2 Web Server, my sliders in the "Actions" column are too small and not usable:
Absorption time, Absorption voltage and Charging current max

It seems the table is not allocating enough width to last  "Actions" columns so the sliders are not fully visible, and are unable to to be used to change the value.

The "State" Column is too wide due to some larger values further down(not show in picture)
Is there any way to reduce the width of "State" and or "Name" columns to allow the "Actions" Column to be wider?

Please see attached screenshot!

### Which version of ESPHome has the issue?

2023.2.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

NA

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

### Component causing the issue

Web Server

### Example YAML snippet

```yaml
web_server:
  port: 80
  auth:
    username: !secret web_server_username
    password: !secret web_server_password
  local: true
  ota: true
```

### Anything in the logs that might be useful for us?

```txt
NA
```

### Additional information

[image]

Uksa007, 2023-06-22:
Anyone have any suggestions?

jesserockz, 2023-06-22:
Seems fine for mine:
[image]

Uksa007, 2023-06-22:
> Seems fine for mine:

Add some more text(more characters) to Senors name and state and you will see the issue present itself.