esphome/issues #4229: Issue with mqtt.publish_json

zoic21, 2023-02-22:
### The problem

Wrong json result (not always) with mqtt.publish_json

### Which version of ESPHome has the issue?

2023.2.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

Olimex ESP32-GATEWAY

### Component causing the issue

mqtt

### Example YAML snippet

```yaml
substitutions:
  radiation_limit: "30"
  pm1_limit: "15"
  pm2_5_limit: "20"
  pm10_limit: "30"

esphome:
  name: esp-weather-outdoor

esp32:
  board: esp32dev
  framework:
    type: arduino

globals:
  - id: pm1_alert
    type: int
    restore_value: no
    initial_value: '0'
  - id: pm2_5_alert
    type: int
    restore_value: no
    initial_value: '0'
  - id: pm10_alert
    type: int
    restore_value: no
    initial_value: '0'

# Enable logging
logger:
  level: DEBUG

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: on
  ap:

captive_portal:

mqtt:
  broker: !secret mqtt_ip
  username : !secret mqtt_username
  password : !secret mqtt_password

time:
  - platform: sntp
    servers: !secret ntp_ip

web_server:
  port: 80
  auth:
   username: admin
   password: !secret web_server_password
    
uart:
  rx_pin: GPIO15
  tx_pin: GPIO16
  baud_rate: 9600
  id: pms_uart

i2c:
  sda: 14
  scl: 13
  scan: True
  frequency: 100kHz
  
sensor:
  - platform: tsl2561
    name: "Ambient Light"
    address: 0x39
    update_interval: 50s

  - platform: bme280
    temperature:
      name: "Temperature"
      oversampling: 2x
    pressure:
      name: "Pressure"
      oversampling: 2x
    humidity:
      name: "Humidity"
      oversampling: 2x
    address: 0x76
    update_interval: 60s

  - platform: template
    name: "Radiation année"
    unit_of_measurement: 'mSv/a'
    lambda: return id(radiation).state * 0.008766;
    update_interval: 60s

  - platform: pulse_counter
    id: radiation
    pin:
      number: 36
    name: "Radiation"
    unit_of_measurement: 'nSv/h'
    accuracy_decimals: 0 
    count_mode:
     rising_edge: DISABLE
     falling_edge: INCREMENT
    filters:
      - offset: -12.0 # J305ß Geiger Mueller tube background noise 0.2 pulses / sec x 60 sec = 12 CPM (Counts per Minute)
      - multiply: 8.12037037037
      - sliding_window_moving_average:
          window_size: 30
          send_every: 1
    on_value:
        then:
          - if:
              condition:
                lambda: return id(radiation).state > ${radiation_limit};
              then:
                - mqtt.publish_json:
                    topic: ntfy/maison
                    payload: |-
                      root["tags"] = "skull,radioactive";
                      root["title"] = "Alerte";
                      root["message"] = ((std::string) "Détection de radiation : " + to_string((int) id(radiation).state)+ (std::string) " nSv/h").c_str();
  - platform: pmsx003
    type: PMSX003
    update_interval: 60000ms
    uart_id: pms_uart
    pm_1_0:
      name: "PMS <1.0µm"
      id : pm_1_0
      filters:
        - delta : 1.0
      on_value:
        then:
          - if:
              condition:
                and:
                  - lambda: return id(pm_1_0).state > ${pm1_limit};
                  - lambda: return id(pm1_alert) == 0;
              then:
                - mqtt.publish_json:
                    topic: ntfy/maison
                    payload: |-
                      root["tags"] = "warning,cloud";
                      root["title"] = "Alerte";
                      root["message"] = ((std::string) "Détection d'une forte concentration PM1 (pollution voiture, industrie...) : " + to_string((int) id(pm_1_0).state)+ (std::string) " µg/m³").c_str();
                - globals.set:
                    id: pm1_alert
                    value: '1'
              else:
                - if:
                    condition:
                      lambda: return id(pm1_alert) == 1;
                    then:
                      - globals.set:
                          id: pm1_alert
                          value: '0'
    pm_2_5:
      name: "PMS Matter <2.5µm"
      id : pm_2_5
      filters:
        - delta : 1.0
      on_value:
        then:
          - if:
              condition:
                and:
                  - lambda: return id(pm_2_5).state > ${pm2_5_limit};
                  - lambda: return id(pm2_5_alert) == 0;
              then:
                - mqtt.publish_json:
                    topic: ntfy/maison
                    payload: |-
                      root["tags"] = "warning,cloud";
                      root["title"] = "Alerte";
                      root["message"] = ((std::string) "Détection d'une forte concentration PM2.5 (particules de combustion, composés organiques...) : " + to_string((int) id(pm_2_5).state)+ (std::string) " µg/m³").c_str();
                - globals.set:
                    id: pm2_5_alert
                    value: '1'
              else:
                - if:
                    condition:
                      lambda: return id(pm2_5_alert) == 1;
                    then:
                      - globals.set:
                          id: pm2_5_alert
                          value: '0'
    pm_10_0:
      name: "PMS Matter <10.0µm"
      id : pm_10_0
      filters:
        - delta : 1.0
      on_value:
        then:
          - if:
              condition:
                and:
                  - lambda: return id(pm_10_0).state > ${pm10_limit};
                  - lambda: return id(pm10_alert) == 0;
              then:
                - mqtt.publish_json:
                    topic: ntfy/maison
                    payload: |-
                      root["tags"] = "warning,cloud";
                      root["title"] = "Alerte";
                      root["message"] = ((std::string) "Détection d'une forte concentration PM10 (poussière, pollen, moisissure...) : " + to_string((int) id(pm_10_0).state)+ (std::string) " µg/m³").c_str();
                - globals.set:
                    id: pm10_alert
                    value: '1'
              else:
                - if:
                    condition:
                      lambda: return id(pm10_alert) == 1;
                    then:
                      - globals.set:
                          id: pm10_alert
                          value: '0'

switch:
  - platform: gpio
    name: "PMS"
    pin: 
      number: GPIO02
      mode: OUTPUT
    restore_mode: ALWAYS_ON
    id: pms_set
```

### Additional information

Here received data on mqtt : 

````
0320|[2023-02-22 17:29:54][DEBUG] : Received message on topic : ntfy/maison => {"tags":"skull,radioactive","title":"Alerte","message":"{\"tags\":\"skull,radioactive\",\"title\":\"Alerte\",\"message\":\"{\"t"}
0446|[2023-02-22 17:30:54][DEBUG] : Received message on topic : ntfy/maison => {"tags":"skull,radioactive","title":"Alerte","message":"{\"tags\":\"skull,radioactive\",\"title\":\"Alerte\",\"message\":\"{\"t"}
0774|[2023-02-22 17:32:07][DEBUG] : Received message on topic : ntfy/maison => {"tags":"skull,radioactive","title":"Alerte","message":"{\"tags\":\"skull,radioactive\",\"title\":\"Alerte\",\"message\":\"{\"t"}
0881|[2023-02-22 17:33:07][DEBUG] : Received message on topic : ntfy/maison => {"tags":"skull,radioactive","title":"Alerte","message":"{\"tags\":\"skull,radioactive\",\"title\":\"Alerte\",\"message\":\"{\"t"}
0997|[2023-02-22 17:34:07][DEBUG] : Received message on topic : ntfy/maison => {"tags":"skull,radioactive","title":"Alerte","message":"{\"tags\":\"skull,radioactive\",\"t"}
````

We can see it's seem there is json inside json  but not always the same

oxan, 2023-02-23:
Your code is broken; remove the `c_str()` call and it should work fine.

zoic21, 2023-02-24:
Hello,

You have right... I don't known why I add this function... And since it's works sometime I don't think my code is wrong.... Sorry for the issue.

closed 2023-02-24 by zoic21