esphome/issues #4231: Nextion tft_upload not working

OptimusGREEN, 2023-02-22:
### The problem

The tft_upload is causeing the display/esp32 to reset rather than uploading the new tft file.
This used to work fine.

### Which version of ESPHome has the issue?

2023.2.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP32

### Board

esp32-devkit/d1mini32

### Component causing the issue

nextion

### Example YAML snippet

```yaml
substitutions:
  device_name: tft-upload-test
# TFT
  url_tft: "http://[redacted]:8123/local/tft/tft_upload_test_success.tft"
  rx: GPIO1
  tx: GPIO3
  baud: '115200'

packages:
  wifi: !include common/wifi-nextion-dev.yaml
  time: !include common/time.yaml

esphome:
  name: $device_name
  platform: esp32
  board: wemos_d1_mini32
  arduino_version: dev
  platformio_options:
    build_flags:
      - -DCONFIG_ARDUINO_LOOP_STACK_SIZE=32768
  on_boot:
    - priority: -110
      then:
        - lambda: |-
            id(nextiondisplay).send_command_printf("wup=0");
            id(nextiondisplay).send_command_printf("thup=1");
            id(nextiondisplay).set_backlight_brightness(1.0);

ota:
captive_portal:
logger:
  level: VERY_VERBOSE
api:
  services:
    - service: update_nextion
      then:
        - lambda: "id(nextiondisplay)->upload_tft();"

[... 182 lines cut ...]
[20:15:41][VV][nextion:303]: print_queue_members_ size 0
[20:15:41][D][nextion:438]: Got touch page=0 component=1 type=PRESS
[20:15:41][D][binary_sensor:036]: 'KTS TFT Upload': Sending state ON
[20:15:41][VV][nextion:781]: Loop End
[20:15:41][VV][nextion:298]: this->command_data_ e length 7
[20:15:41][VV][nextion:200]: print_queue_members_ (top 10) size 0
[20:15:41][VV][nextion:201]: *******************************************
[20:15:41][VV][nextion:214]: *******************************************
[20:15:41][VV][nextion:303]: print_queue_members_ size 0
[20:15:41][D][nextion:438]: Got touch page=0 component=1 type=RELEASE
[20:15:41][D][binary_sensor:036]: 'KTS TFT Upload': Sending state OFF
[20:15:41][D][nextion_upload:169]: Connected
[20:15:41][D][nextion_upload:175]: Requesting URL: http://[redacted]:8123/local/tft/tft_upload_test_success.tft
WARNING nextion-dev-test.optinet: Connection error occurred: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for nextion-dev-test.optinet
WARNING Disconnected from API
INFO Successfully connected to nextion-dev-test.optinet
[20:15:55][VV][scheduler:207]: Running interval 'update' with interval=5000 last_execution=3600 (now=8600)
[20:16:00][VV][scheduler:207]: Running interval 'update' with interval=5000 last_execution=8600 (now=13600)
[20:16:05][VV][scheduler:207]: Running interval 'update' with interval=5000 last_execution=13600 (now=18600)
```

### Additional information

I have been using this feature with my proper nextion setup but it has stopped uploading so i made this basic single button one to eliminate the possibility of it being that there was too much going on.

OptimusGREEN, 2023-04-08:
Does anyone have any info or solution on this? I'm literally having to take my screen off the wall, insert an sd card to update my tft.

Any help would be great.

J0RD4N300, 2023-04-12:
Can't help the issue directly but it's been a problem for roughly 6 months now (if your issue is the same as mine which i believe it is.)  I believe it happens when a certain function is called relating to the device wifi it crashes the ESP and it reboots. I don't know enough about coding to try and fix it myself but that's what I seemed to track down when I looked into it.

closed 2023-10-23 by jesserockz