esphome/issues #4232: M5 Stack 4 relais

KoenBreeman, 2023-02-22:
### The problem

Gents 
I have a 4 relais from M5Stack. I've started to write a custom component. I'm almost there, but the build generates errors. Is there someone who can help me?

### Which version of ESPHome has the issue?

2023.2.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

M5Atom

### Component causing the issue

M5 stack 4 relais

### Example YAML snippet

```yaml
M5Stack_4_relais:
  Relais_1:
    name: Relais 1
    id: Relais_1
  Relais_2:
    name: Relais 2
    id: Relais_2
  Relais_3:
    name: Relais 3
    id: Relais_3
  Relais_4:
    name: Relais 4
    id: Relais_4
  Relais_all:
    name: Relais all
    id: Relais_all
```

### Anything in the logs that might be useful for us?

```txt
Folder structure:
M5Stack_4_relais
-M5Stack_4_relais.h
-M5Stack_4_relais.cpp
-__init__.py
--switch (folder)
---M5Stack_switch.cpp
---M5Stack_switch.h
---switch.py

M5Stack_4_relais.h
/*
*******************************************************************************
* Copyright (c) 2021 by M5Stack
*                  Equipped with Atom-Lite/Matrix sample source code
* Visit for more information: https://docs.m5stack.com/en/unit/4relay
*
* Product: Unit 4RELAY.
* Date: 2022/7/22
*******************************************************************************
  Please connect to port ,Control 4 relays and demonstrate the asynchronous
[... 374 lines cut ...]
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- Wire @ 2.0.0
Compiling .pioenvs/coming-soon-dmx/src/esphome/components/M5Stack_4_relais/M5Stack_4_relais.cpp.o
src/esphome/components/M5Stack_4_relais/M5Stack_4_relais.cpp:1:10: fatal error: M5stack_4_relais.h: No such file or directory

**************************************************************************
* Looking for M5stack_4_relais.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:M5stack_4_relais.h"
* Web  > https://registry.platformio.org/search?q=header:M5stack_4_relais.h
*
**************************************************************************

 #include "M5stack_4_relais.h"
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pioenvs/coming-soon-dmx/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/coming-soon-dmx/src/esphome/components/api/api_server.cpp.o
*** [.pioenvs/coming-soon-dmx/src/esphome/components/M5Stack_4_relais/M5Stack_4_relais.cpp.o] Error 1
Compiling .pioenvs/coming-soon-dmx/src/esphome/components/api/list_entities.cpp.o
========================== [FAILED] Took 9.58 seconds ==========================
```

ssieb, 2023-02-22:
Github issues are not for support requests.  Ask on the esphome discord server or one of the forums.  It would also be better if you have the code published to a git repo or somewhere else where it's viewable in place.

nielsnl68, 2023-02-22:
And I have already a component for this on my github: https://github.com/nielsnl68/esphome-components

KoenBreeman, 2023-02-23:
Thnx Niels, but I see that you haven't fully implemented all the switch functions. So I will try and update them

nielsnl68, 2023-02-26:
when you want, please create a PR on my github and i will merge it right away.

KoenBreeman, 2023-06-30:
Old, see this pull request: https://github.com/esphome/esphome/pull/4786

KoenBreeman, 2023-06-30:
Old, see this pull-request

closed 2023-06-30 by KoenBreeman