esphome/issues #4233: pmsx003, D1 mini (esp8266) and debug logging causes reboots

MallocArray, 2023-02-22:
### The problem

When using a D1 Mini based on ESP8266 with a PMS5003 and ssd1306_i2c OLED, the default settings for the logger and update_interval causes the device to reboot frequently, typically 30 min - 3 hours but sometimes shorter and sometimes longer.

If I do not use the PMS5003 sensor at all, it appears stable.
If I change the `update_interval:` to something higher, such as 120s from the default of 1s, reboots are less frequent, sometime days, but still reboots.
If I change the logging to INFO, it appears to be stable over several days
```
logger:
  level: INFO
```
I did try only lowering the logging on sensor and leaving the default, but still encountered reboots
```
logger:
  logs:
    pmsx003: INFO 
```

Also seeing the same behavior on a separate device with the Ikea sensor pm1006 with a SGP30 and no OLED with default logging, although much less frequent, about every day, but this pm1006 sensor only receives data every 20 seconds by default compared to every 1 second on the 5003.  The SGP30 is also very chatty on the Debug logs

There has been discussion on the AirGradient forum about it, and the Arduino Core SoftwareSerial package has been a culprit from reboots with debug logging, but from what I can tell the default ESPHome package uses a version older than SoftwareSerial 7
https://forum.airgradient.com/t/airgradient-rebooting-itself/603/48
https://forum.airgradient.com/t/airgradient-pro-with-esphome-display-keeps-turning-back-on/641

Ultimately, it appears that reducing the logging, either by changing from DEBUG to INFO or by increasing the update_interval seems to help, but would prefer a permanent fix.

### Which version of ESPHome has the issue?

2022.2.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

D1 Mini

### Component causing the issue

logger pmsx003

### Example YAML snippet

```yaml
esphome:
  name: ag-pro

esp8266:
  board: d1_mini
  # framework:
  #   platform_version: 4.1.0

logger:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

uart:
  # - rx_pin: D4
  #   tx_pin: D3
  #   baud_rate: 9600
  #   id: senseair_s8_uart

  - rx_pin: D5
    tx_pin: D6
    baud_rate: 9600
    id: pms5003_uart

i2c:
  sda: D2
  scl: D1

sensor:
  - platform: pmsx003
    type: PMSX003
    pm_2_5:
      name: "Particulate Matter <2.5µm Concentration"
      id: pm2
      filters:
        - sliding_window_moving_average:
            window_size: 30
            send_every: 30
    uart_id: pms5003_uart

  # - platform: senseair
  #   # https://esphome.io/components/sensor/senseair.html
  #   co2:
  #     name: "SenseAir S8 CO2 Value"
  #     id: co2
  #   update_interval: 30s
  #   uart_id: senseair_s8_uart

  - platform: uptime
    name: Uptime Sensor
    id: agpro_uptime

  - platform: sgp30
    eco2:
      name: "eCO2"
      id: eco2
      accuracy_decimals: 1
      filters:
        - sliding_window_moving_average:
            window_size: 30
            send_every: 30
    tvoc:
      name: "TVOC"
      id: tvoc
      accuracy_decimals: 1
      filters:
        - sliding_window_moving_average:
            window_size: 30
            send_every: 30
    eco2_baseline:
      name: "eco2_baseline"
      id: eco2_baseline
    tvoc_baseline:
      name: "tvoc_baseline"
      id: tvoc_baseline
    store_baseline: true

font:
  - file: './fonts/liberation_sans/LiberationSans-Regular.ttf'
    id: font1
    size: 21

display:
  - platform: ssd1306_i2c
    model: "SH1106 128x64"
    id: oled_display
    reset_pin: D0
    address: 0x3C
    lambda: |-
      it.printf(0, 00, id(font1), "PM2.5: %.0fppm", id(pm2).state);
      it.printf(0, 21, id(font1), "CO2: %.0fppm", id(eco2).state );
      it.printf(0, 42, id(font1), "TVOC: %.0fppb", id(tvoc).state);
```

ssieb, 2023-02-22:
> ### Which version of ESPHome has the issue?
> 
> 2022.2.2

Is that really the esphome version you're using?

MallocArray, 2023-02-22:
@ssieb No, sorry, I manually typed instead of copy and paste.  Corrected to 2023.2.2

ssieb, 2023-02-22:
Can you get serial logs from the crash, preferably with a decoded stack trace, which requires it being plugged into the device running esphome (not over web serial)?

MallocArray, 2023-02-22:
Is there a document for how to do that? I can plug it into the device running ESPHome, although it is in a Docker container.

ssieb, 2023-02-22:
If you can get the serial port recognized inside the container, the dashboard will let you get the logs that way.

MallocArray, 2023-02-24:
Of course, after I post this all of my devices have been up and stable for more than 24 hours, even if I didn't update their configuration.  I'm still keeping an eye on it to collect logs

MallocArray, 2023-02-28:
I had a power outage yesterday and when all of my devices rebooted due to that, I'm seeing more random reboots.  Trying to capture logs when it happens.

A user on the AirGradient forum posted a good summary of 2 different issues with the SoftwareSerial library, and it appears that what ESPHome uses will be pre-6.15.1 which has a bug relating to the last packet being 0xFF

https://forum.airgradient.com/t/airgradient-pro-with-esphome-display-keeps-turning-back-on/641/23
https://forum.airgradient.com/t/airgradient-pro-with-esphome-display-keeps-turning-back-on/641/25

This may be what we are running into

SergiuToporjinschi, 2023-03-12:
I have the same issue with an Esp8266-12e, just that is not able to start at all with 
```  
logger:   
  level: DEBUG
```
Reading those links, that MallocArray posted, I've checked and yes I'm using the same version 
after switching logs to NONE or INFO is working

MallocArray, 2025-07-04:
Unable to confirm if this is still occurring with current versions of ESPHome

closed 2025-07-04 by MallocArray