esphome/issues #4235: ESPHome does not reconnect Wifi after wireless install of newly compiled firmware

jurviz, 2023-02-23:
### The problem

I'm currently testing ESPHome with Raspberry Pi Pico W without anything connected.
After install of new compiled YAML file the board boots, but does not reconnect to the Wifi network. It reconnects after powering off and on.
This behavior is reproducible.
I've tested with two different boards.

### Which version of ESPHome has the issue?

2023.2.4, 2022.12.8

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

RP2040

### Board

Raspberry Pi Pico W

### Example YAML snippet

```yaml
esphome:
  name: sensor-bme280-bh1750-1

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xDZZmKXi8y+cNJBBI94HHdyTJJ1ZKDq2ifN+ulJzigg="

ota:
  password: "f33b156868b2be62be4820dfc7c871ea"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: ".fritz.box"
  fast_connect: True
#  use_address: 192.168.178.99

# Optional manual IP
  manual_ip:
    static_ip: 192.168.178.99
    gateway: 192.168.178.1
    subnet: 255.255.255.0
    dns1: 192.168.178.1

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Sensor-Bme280-Bh1750-1"
    password: "1od85rPEzA5q"

# Example configuration entry
sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
[D][sensor:126]: 'WiFi Signal Sensor': Sending state -58.00000 dBm with 0 decimals of accuracy
[W][api.connection:083]: ESPHome Logs 2023.2.4 (192.168.178.56): Connection closed
[D][ota:147]: Starting OTA Update from 192.168.178.56...
[D][ota:316]: OTA in progress: 0.2%
[D][ota:316]: OTA in progress: 15.4%
[D][ota:316]: OTA in progress: 31.3%
[D][ota:316]: OTA in progress: 47.1%
[D][ota:316]: OTA in progress: 63.1%
[D][ota:316]: OTA in progress: 79.1%
[D][ota:316]: OTA in progress: 95.0%
[I][ota:349]: OTA update finished!
[I][app:134]: Rebooting safely...

Error reading from serial device
```

jurviz, 2023-02-23:
It reconnects after about 20 minutes without intervention.

MallocArray, 2023-02-24:
Sounds like it may be similar to https://github.com/esphome/issues/issues/4053 where I'm seeing a D1 Mini giving Auth Expired messages right after flashing OTA, but after some amount of time it will eventually reconnect

emefff, 2023-02-25:
This sounds very familiar to me. My only RP2040 reboots in irregular intervals also. I posted my findings weeks ago in [(https://github.com/esphome/issues/issues/3791)] but didn't get any meaningful answers.

I think at the moment, RP2040 is just not up to par with other EPSs in terms of integration in ESPHome.

jurviz, 2023-03-01:
I did some more tests and found that setting 
``` 
# Enable Home Assistant API
api:
  reboot_timeout: 1min
```
reduces the time to reconnect to 1 minute. 
Disabling the timeout with 0s leads to complete hangup, needs cycling power to recover.

emefff, 2023-03-28:
Hello,
thank you, I tried that. A few days ago, I had to reflash the Pico via USB. It hasn't been onlie for days. After successful flashing of ESPHome 2023.3.1 this Pico worked quite well until the next morning. Since then, it is unreachable again, it never went online. 
[image]

DennisYildirim, 2023-09-23:
Well it does not work when using esp32 c3 wemos mini. Not sure why people are simply closing each topic as soon as they get a shot but not confirm if it hit or not. too much hyperactivity everywhere...