esphome/issues #4236: ESP32 device can't update from 2022.12.8 to later versions OTA

PrayerfulDrop, 2023-02-23:
### The problem

I continue to receive notifications that my ESP32 Bluetooth Proxy needs to be updated from 2022.12.8 to the latest esphome version (as of posting this 2023.2.4.  When attempting to flash the ESP32 device either through Home Assistant notifications or through the ESPHome add-on the ESP32 becomes nonresponsive and does not accept the update.  This then requires a reinstall of the bluetooth proxy software from https://esphome.github.io/bluetooth-proxies/ URL which sets the device back to 2022.12.8. 

### Which version of ESPHome has the issue?

2023.X.X

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.X.X

### What platform are you using?

ESP32

### Board

nodecmu

### Component causing the issue

bluetooth proxy

### Example YAML snippet

```yaml
standard Bluetooth proxy install from https://esphome.github.io/bluetooth-proxies/
```

### Anything in the logs that might be useful for us?

```txt
nothing in the logs
```

mrand, 2023-03-05:
Possible dup of #4146?

asboldyrev, 2023-05-24:
I have a similar problem.  I haven't been able to update the add-on for several months now.  I am using Raspberry Pi.