esphome/issues #4237: Unable to update ESP32 from 2023.2.3 to 2023.2.4

xfader2018, 2023-02-23:
### The problem

Hello, I'm unable to update met ESP32 devices to 2023.2.4.
My other ESP8266 updates fine.
For both ESP32 devices this is the error I get:

```
INFO Reading configuration /config/esphome/carcam.yaml...
WARNING GPIO0 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing carcam (board: esp32cam; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
Linking /data/carcam/.pioenvs/carcam/firmware.elf
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.
*** [/data/carcam/.pioenvs/carcam/firmware.elf] Error 1
========================= [FAILED] Took 86.36 seconds =========================
```

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP32

### Board

mrover dev kit

ssieb, 2023-02-23:
That's almost certainly an out of memory issue.  What are you running this on?  I guess you were just lucky before or other things weren't taking as much memory.

xfader2018, 2023-02-23:
> That's almost certainly an out of memory issue. What are you running this on? I guess you were just lucky before or other things weren't taking as much memory.

One is very default BT tracker and the other is default esp32cam but nothing specials.

ssieb, 2023-02-23:
I mean the system running HA and esphome.

xfader2018, 2023-02-23:
> I mean the system running HA and esphome.

HA running on proxmox with more than enough memory. Server has 64GB and HA got 4GB and doesn't even reach 1GB.

ssieb, 2023-02-23:
You checked the RAM usage at that link point?  Does it crash right away or does it run for a couple of seconds?  4GB should be enough, but I don't know what else would cause that problem.

xfader2018, 2023-02-23:
You were right. I had some weird memory leak where proxmox server was high memory. Even with a max ZFS arc of 4GB.
Anyway, after reset the compile went with issue.
Thanks for help.

closed 2023-02-23 by ssieb