esphome/issues #4238: Compiling fails in Home Assistant when using platform: adc

kolchy, 2023-02-24:
### The problem

I have an Atom Lite that I deployed using Home Assistant running on a RPi. It was last deployed with ESPHome version 2022.11.1. I recently tried to deploy using the latest version (2023.2.4) and when I try to deploy, the compilation fails:

```
INFO Reading configuration /config/esphome/atom-garden-tanks.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing atom-garden-tanks (board: m5stack-atom; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/atom-garden-tanks/.pioenvs/atom-garden-tanks/src/esphome/components/adc/adc_sensor.cpp.o
In file included from src/esphome/components/adc/adc_sensor.h:4,
                 from src/esphome/components/adc/adc_sensor.cpp:1:
src/esphome/core/hal.h:14:10: fatal error: esp_attr.h: No such file or directory

******************************************************************
* Looking for esp_attr.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_attr.h"
* Web  > https://registry.platformio.org/search?q=header:esp_attr.h
*
******************************************************************

 #include <esp_attr.h>
          ^~~~~~~~~~~~
compilation terminated.
Compiling /data/atom-garden-tanks/.pioenvs/atom-garden-tanks/src/esphome/components/api/api_connection.cpp.o
Compiling /data/atom-garden-tanks/.pioenvs/atom-garden-tanks/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/atom-garden-tanks/.pioenvs/atom-garden-tanks/src/esphome/components/api/api_pb2.cpp.o
*** [/data/atom-garden-tanks/.pioenvs/atom-garden-tanks/src/esphome/components/adc/adc_sensor.cpp.o] Error 1
In file included from src/esphome/components/socket/socket.h:5,
                 from src/esphome/components/api/api_frame_helper.h:13,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/socket/headers.h:116:10: fatal error: sys/ioctl.h: No such file or directory
 #include <sys/ioctl.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [/data/atom-garden-tanks/.pioenvs/atom-garden-tanks/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from src/esphome/components/socket/socket.h:5,
                 from src/esphome/components/api/api_frame_helper.h:13,
                 from src/esphome/components/api/api_frame_helper.cpp:1:
src/esphome/components/socket/headers.h:116:10: fatal error: sys/ioctl.h: No such file or directory
 #include <sys/ioctl.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [/data/atom-garden-tanks/.pioenvs/atom-garden-tanks/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
In file included from src/esphome/components/api/proto.h:4,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
src/esphome/core/log.h:14:10: fatal error: esp_err.h: No such file or directory

*****************************************************************
* Looking for esp_err.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_err.h"
* Web  > https://registry.platformio.org/search?q=header:esp_err.h
*
*****************************************************************

 #include <esp_err.h>
          ^~~~~~~~~~~
compilation terminated.
*** [/data/atom-garden-tanks/.pioenvs/atom-garden-tanks/src/esphome/components/api/api_pb2.cpp.o] Error 1
========================== [FAILED] Took 4.58 seconds ==========================
```

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP32

### Board

m5stack-atom

### Component causing the issue

adc

### Example YAML snippet

```yaml
esphome:
  name: atom-garden-tanks
  board: m5stack-atom
  platform: esp32

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXX"

ota:
  password: "YYY"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

sensor:
  - platform: adc
    pin: 33
    name: "Garden Tanks (V)"
    update_interval: 30s
    accuracy_decimals: 3
    unit_of_measurement: "V"
```

tracestep, 2023-03-02:
Possibly related to #4264

kolchy, 2023-03-03:
Thanks @tracestep 🙏 Based on the details in #4264 I uninstalled and reinstalled the ESPHome integration and everything started working again...

closed 2023-03-03 by kolchy