esphome/issues #4241: Sonoff THR320 with THS01 Sensor - DHT failed

erdembey, 2023-02-24:
### The problem

I believe yaml and logs are enough, long story short;
I am using Sonoff THR320 with THS01 Sensor.
I have working HA setup for my home automation.
I've tried several tutorials but it is not working.

Any recommendations ?

Thanks

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP32

### Board

Sonoff THR320

### Component causing the issue

THS01

### Example YAML snippet

```yaml
esphome:
  name: "thr320"
  friendly_name: "THR320"
  on_boot:
    - priority: 110
      then:
      - switch.turn_on: ${name}_sensor_power

esp32:
  board: nodemcu-32s

switch:
  - platform: gpio
    pin: 27
    id: ${name}_sensor_power
    restore_mode: always_on

sensor:
  - platform: dht
    pin: 25
    model: SI7021
    temperature:
      name: "${friendly_name} Temp"
      id: "${name}_temp"
    humidity:
      name: "${friendly_name} Hum"
      id: "${name}_humi"
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
[00:35:08][W][dht:169]: Requesting data from DHT failed!
[00:35:08][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[00:35:08][D][sensor:127]: 'THR320 Temp': Sending state nan °C with 1 decimals of accuracy
[00:35:08][D][sensor:127]: 'THR320 Hum': Sending state nan % with 0 decimals of accuracy
```

erdembey, 2023-02-27:
Update: Solution provided in the link below worked for me
https://github.com/esphome/issues/issues/4189#issuecomment-1445023142

closed 2023-02-27 by erdembey

erdembey, 2023-03-18:
> If you're using some external code, then it's not actually fixed.

Unfortunately it's not fixed in latest update ... I have to use the solution provided in the link manually.

kristjanbjarni, 2023-03-25:
This is also broken for me in version 2023.3.1 but used to work in older version. Can somebody reverse the new broken code or use the fix provided in the comment?

erdembey, 2023-03-26:
> This is also broken for me in version 2023.3.1 but used to work in older version. Can somebody reverse the new broken code or use the fix provided in the comment?

There are 2 solutions provided for this issue in the same thread;

https://github.com/esphome/issues/issues/4189#issuecomment-1445023142

and

kristjanbjarni, 2023-03-26:
> > This is also broken for me in version 2023.3.1 but used to work in older version. Can somebody reverse the new broken code or use the fix provided in the comment?
 
> There are 2 solutions provided for this issue in the same thread;
> 
> [#4189 (comment)](https://github.com/esphome/issues/issues/4189#issuecomment-1445023142)
> 
> and
> 
> [#4189 (comment)](https://github.com/esphome/issues/issues/4189#issuecomment-1468872470)

Yes thank you I used the fix with external_components and it works, but this really needs to be included in the master release. Is somebody working on a PR for this?

erdembey, 2023-03-27:
> Yes thank you I used the fix with external_components and it works, but this really needs to be included in the master release. Is somebody working on a PR for this?

We have a new release now but it's not included again.
I believe nobody working on PR for this. @Beungoud ?

erdembey, 2023-04-20:
We have a new release now but it's not included again.
Btw, i've added Beungoud fix as external component but not working.

kristjanbjarni, 2023-04-20:
> We have a new release now but it's not included again. Btw, i've added Beungoud fix as external component but not working.

@erdembey Are you saying it stopped working again even with Beungoud fix? This is not good. I think I might have to start looking into moving all my devices to Tasmota instead of Esphome because this is just getting ridiculous with every single Esphome update breaking something.

erdembey, 2023-04-20:
> @erdembey Are you saying it stopped working again even with Beungoud fix? This is not good. I think I might have to start looking into moving all my devices to Tasmota instead of Esphome because [...]

Yes, it's just stopped working even though i've added external_component for fixing the issue. 
Latest message from device (same components, different model);

`[21:53:27][W][dht:174]: Requesting data from DHT failed!
[21:53:27][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[21:53:27][D][sensor:110]: 'THR316D': Sending state nan °C with 1 decimals of accuracy
[21:53:27][D][sensor:110]: 'THR316D': Sending state nan % with 0 decimals of accuracy`

kristjanbjarni, 2023-04-20:
> Yes, it's just stopped working even though i've added external_component for fixing the issue. Latest message from device (same components, different model);
>
> `[21:53:27][W][dht:174]: Requesting [...]

So you get no readings? I was getting NaN values occasionally but I just added this filter for NaN values and it was ok.

``` yaml
sensor:
  - platform: dht
    model: SI7021
    pin: GPIO25
    temperature:
      name: "${devicename} Temperature"
      id: temperature
      filters:
        - filter_out: nan
```

erdembey, 2023-06-22:
It seems that the modifications proposed in this thread still not applied in the latest beta versions.