esphome/issues #4243: SGP30 sensor invalid values after upgrade 2023.2.4

rdone2018, 2023-02-25:
### The problem

After upgrade ESPHOME to version 2023.2.4 - all my Air sensors readings goes minimum values.
Normal values eCO2 before update - between 800-1500ppm

Sensor 1
[10:25:51][D][sgp30:282]: Got eCO2=400.0ppm TVOC=1.0ppb
[10:25:51][D][sgp30:157]: Baseline reading not available for: 41687s
[10:25:52][D][sgp30:282]: Got eCO2=400.0ppm TVOC=4.0ppb
[10:25:52][D][sgp30:157]: Baseline reading not available for: 41686s

Sensor 2
[10:26:42][D][sgp30:282]: Got eCO2=400.0ppm TVOC=4.0ppb
[10:26:42][D][sgp30:156]: Baseline reading not available for: 41029s
[10:26:43][D][sgp30:282]: Got eCO2=400.0ppm TVOC=5.0ppb
[10:26:43][D][sgp30:156]: Baseline reading not available for: 41028s

Sensor 3
[10:27:20][D][sgp30:282]: Got eCO2=410.0ppm TVOC=14.0ppb
[10:27:20][D][sgp30:156]: Baseline reading not available for: 42608s
[10:27:21][D][sgp30:282]: Got eCO2=420.0ppm TVOC=13.0ppb
[10:27:21][D][sgp30:156]: Baseline reading not available for: 42607s

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP8266

### Board

D1 mini + ESP32

### Component causing the issue

SGP30

jshank, 2023-08-21:
This is still happening for me. Any updates?

Globolo001, 2024-06-04:
idk if it’s just an software or if I am stupid, but I get this error. Just installed it, and not working. Might be a different or this issue.
After 12 hours waiting still no good baseline and reporting same values. 
```
[00:23:34][D][sensor:094]: 'VOCs': Sending state 0.00000 ppb with 0 decimals of accuracy
[00:23:34][D][sgp30:158]: Baseline reading not available for: 43190s
[00:23:35][D][sgp30:292]: Got eCO2=400.0ppm TVOC=0.0ppb
[00:23:35][D][sensor:094]: 'eCO2 Value': Sending state 400.00000 ppm with 0 decimals of accuracy
```

boggiano, 2025-03-10:
> idk if it’s just an software or if I am stupid, but I get this error. Just installed it, and not working. Might be a different or this issue. After 12 hours waiting still no good baseline and [...]

I don't know for the values since I have the same problem but the problem about the baseline, quite sure, it depends on the reboot of the device (it needs to be added to HA or it reboots itself thinking about a ongoing problem).

Globolo001, 2025-03-12:
It worked after a while doing nothing. Idk what changed but the sensors are reading values.