esphome/issues #4244: WS2812 with esp-edf

JakobTewes, 2023-02-25:
### The problem

Heyhey,

because of an addon, I need to use esp-idf framework also I need support for WS2812 LEDs.
As both neopixelbus and fastled do not work, I´m now stuck ;-(.
Can someone recommend me a solution for using WS2812 with esp-idf framework?

`esp32:`
`  board: az-delivery-devkit-v4`
`  framework:`
`    type: esp-idf`
`    version: latest`
`light:`
`## No work with esp-idf`
`#  - platform: fastled_clockless`
`#    id: "${devicename}_strip"`
`#    name: "${friendly_name} LED Strip"`
`#    icon: mdi:led-strip-variant`
`#    chipset: WS2811`
`#    pin: GPIO18`
`#    num_leds: 8`
`#    rgb_order: GRB`
`## No work with esp-idf`
`#  - platform: neopixelbus`
`#    id: "${devicename}_strip"`
`#    name: "${friendly_name} LED Strip"`
`#    icon: mdi:led-strip-variant`
`#    type: GRB`
`#    variant: 800KBPS`
`#    pin: GPIO18`
`#    method:`
`#      type: esp32_i2s`
`#    num_leds: 8`
`#    internal: true`
`#    default_transition_length: 3s`

Thanks and regards

Jakob

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP32-IDF

### Board

az-delivery-devkit-v4

### Component causing the issue

neopixelbus fastled

### Example YAML snippet

```yaml
esp32:
  board: az-delivery-devkit-v4
  framework:
    type: esp-idf
    version: latest
## No work with esp-idf
#  - platform: fastled_clockless
#    id: "${devicename}_strip"
#    name: "${friendly_name} LED Strip"
#    icon: mdi:led-strip-variant
#    chipset: WS2811
#    pin: GPIO18
#    num_leds: 8
#    rgb_order: GRB
## No work with esp-idf
#  - platform: neopixelbus
#    id: "${devicename}_strip"
#    name: "${friendly_name} LED Strip"
#    icon: mdi:led-strip-variant
#    type: GRB
#    variant: 800KBPS
#    pin: GPIO18
#    method:
#      type: esp32_i2s
#    num_leds: 8
#    internal: true
#    default_transition_length: 3s
```

TheRemote, 2023-03-02:
I'm stuck on this as well because ESP32-C3 still boot loops and won't connect to WiFi unless you use ESP-IDF.  This worked fine for all of my sensors but I can't control the NeoPixels on my buttons.

mattkasa, 2023-05-19:
@JakobTewes @TheRemote I think this might be resolved by @jesserockz with https://github.com/esphome/esphome/pull/4708 and included in `2023.5.0`.

I'm using this config successfully:

```yaml
light:
  - platform: esp32_rmt_led_strip
    id: led_ring
    name: "${friendly_name} Light"
    pin: GPIO19
    num_leds: 12
    rmt_channel: 0
    rgb_order: GRB
    chipset: ws2812
```

JakobTewes, 2023-05-20:
Can confirm. Works like a charm. Thanks so much!

closed 2023-10-21 by ssieb