esphome/issues #4247: Issue with Sprinkler Controller component when last zone only selected

GWH74, 2023-02-25:
### The problem

When using a pump and only the last valve is enabled pump is not turned off after last zone run duration has ended. 
Another problem is listed below.

Setup:
Sprinkler Controller component on an ESP32 8 channel relay board.
1st relay is setup as pump ( master solenoid) 
Then next 7 relays are used for zone 1 - 7
Enable all zones via vales enable_switch, run through a full cycle. (note yaml below is only set for run duration of 5s for testing).
Disable zones 1-6 via enable_switch so that only zone 7 is enabled. 
Call service start_full_cycle.
Zone 7 will run (as well as pump switch) and once zone 7 turns off after run duration the pump switch will stay on. 
Need to turn off main_switch (calls sprinkler.shutdown) to turn off pump.
This has been confirmed on other systems.

Now... If you only run 1 zone that is not the last zone (1-6), like zone 2, then it behaves correctly.
BUT now if you un-enable the above zone 2 and now only enable zone 7 then when you run start_full_cycle  the above zone 2 will run again and NOT zone 7. Then when that zone  turns off the pump will stay on until the sprinkler.shutdown is call.
The below log file shows this. Disable zones 1-6 but then when the start_full_cycle is called zone 2 was ran because it was the single zone I ran before only enabling only zone 7.

 

### Which version of ESPHome has the issue?

2023.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP32

### Board

Generic 8 way relay board with ESP32 module

### Component causing the issue

Sprinkler Contoller

### Example YAML snippet

```yaml
sprinkler:
  - id: lawn_sprinkler_ctrlr
    main_switch: "Lawn Sprinklers"
    auto_advance_switch: "Lawn Sprinklers Auto Advance"
    reverse_switch: "Lawn Sprinklers Reverse"
    valve_open_delay: 1s
    valves:
      - valve_switch: "Zone 1"
        enable_switch: "Enable Zone 1"
        pump_switch_id: sprinkler_master
        run_duration: 5s
        valve_switch_id: zone_1
      - valve_switch: "Zone 2"
        enable_switch: "Enable Zone 2"
        pump_switch_id: sprinkler_master
        run_duration: 5s
        valve_switch_id: zone_2
      - valve_switch: "Zone 3"
        enable_switch: "Enable Zone 3"
        pump_switch_id: sprinkler_master
        run_duration: 5s
        valve_switch_id: zone_3
      - valve_switch: "Zone 4"
        enable_switch: "Enable Zone 4"
        pump_switch_id: sprinkler_master
        run_duration: 5s
        valve_switch_id: zone_4
      - valve_switch: "Zone 5"
        enable_switch: "Enable Zone 5"
        pump_switch_id: sprinkler_master
        run_duration: 5s
        valve_switch_id: zone_5
      - valve_switch: "Zone 6"
        enable_switch: "Enable Zone 6"
        pump_switch_id: sprinkler_master
        run_duration: 5s
        valve_switch_id: zone_6
      - valve_switch: "Zone 7"
        enable_switch: "Enable Zone 7"
        pump_switch_id: sprinkler_master
        run_duration: 5s
        valve_switch_id: zone_7

switch:
  - platform: gpio
    id: sprinkler_master
    pin: GPIO32
  - platform: gpio
    id: zone_1
    pin: GPIO33
  - platform: gpio
    id: zone_2
    pin: GPIO25
  - platform: gpio
    id: zone_3
    pin: GPIO26
  - platform: gpio
    id: zone_4
    pin: GPIO27
  - platform: gpio
    id: zone_5
    pin: GPIO14
  - platform: gpio
    id: zone_6
    pin: GPIO12
  - platform: gpio
    id: zone_7
    pin: GPIO13
```

### Anything in the logs that might be useful for us?

```txt
[16:02:38][D][switch:017]: 'Enable Zone 1' Turning OFF.
[16:02:38][D][switch:056]: 'Enable Zone 1': Sending state OFF
[16:02:38][D][switch:017]: 'Enable Zone 2' Turning OFF.
[16:02:38][D][switch:056]: 'Enable Zone 2': Sending state OFF
[16:02:39][D][switch:017]: 'Enable Zone 3' Turning OFF.
[16:02:39][D][switch:056]: 'Enable Zone 3': Sending state OFF
[16:02:39][D][switch:017]: 'Enable Zone 4' Turning OFF.
[16:02:39][D][switch:056]: 'Enable Zone 4': Sending state OFF
[16:02:40][D][switch:017]: 'Enable Zone 5' Turning OFF.
[16:02:40][D][switch:056]: 'Enable Zone 5': Sending state OFF
[16:02:40][D][switch:017]: 'Enable Zone 6' Turning OFF.
[16:02:40][D][switch:056]: 'Enable Zone 6': Sending state OFF
[16:02:42][D][switch:013]: 'Lawn Sprinklers' Turning ON.
[16:02:42][D][sprinkler:1062]: Starting valve 1 for 5 seconds, cycle 2 of 1
[16:02:42][D][switch:013]: 'zone_2' Turning ON.
[16:02:42][D][switch:056]: 'zone_2': Sending state ON
[16:02:42][D][switch:013]: 'sprinkler_master' Turning ON.
[16:02:42][D][switch:056]: 'sprinkler_master': Sending state ON
[16:02:42][D][switch:056]: 'Lawn Sprinklers': Sending state ON
[16:02:42][D][switch:056]: 'Zone 2': Sending state ON
[16:02:47][D][switch:017]: 'zone_2' Turning OFF.
[16:02:47][D][switch:056]: 'zone_2': Sending state OFF
[16:02:47][D][switch:056]: 'Zone 2': Sending state OFF

**** I have to manually turn it off or valve 1 (pump) will start on.

[16:02:52][D][switch:017]: 'Lawn Sprinklers' Turning OFF.
[16:02:52][D][switch:056]: 'Lawn Sprinklers': Sending state OFF
[16:02:54][D][switch:017]: 'sprinkler_master' Turning OFF.
[16:02:54][D][switch:056]: 'sprinkler_master': Sending state OFF
[16:02:54][D][sprinkler:1084]: All valves stopped, including pumps
```

### Additional information

This has also been tested on a co-workers system and he has the same issue. We have not shared our code, just we have both followed the guide lines and examples.

kbx81, 2023-05-11:
Can you confirm if this is still an issue in the latest release version of ESPHome? Thanks!

GWH74, 2023-05-17:
Sorry for the late reply. Can confirm that it is working as expected and fault as described above has been fixed

kbx81, 2023-05-17:
Great, thanks for confirming! 🍻

closed 2023-05-17 by kbx81