esphome/issues #4249: Differential control with two adjustable sensors, is that possible?

judie07, 2023-02-26:
### The problem

### A question/problem
Not 100% sure this is a bug, but i read through the important information in climate action vs climate mode. For a differential control, a second temperature is necessary, but since this must come from a sensor, the "default_target_temperature" that can be set via the frontend is not targeting.

„Differential control = keeping the temperature constant between two sensors.“

What I have already tried (without success) ->
default_target_temperature: _${default_target_temp}_
or
default_target_temperatur: _!lambda |-
                                   return id(setp).state;  ==> "This option is not templatable!"

**The background** is the Control of a Solar Pump which should enable a "sliding" operation of the Collector.

Keeping the Temperature Difference between e.g. TC and TR constant leads to a "gliding" operation of the Collector. Operation of the Collector. If TK decreases as a result of decreasing irradiation, the Difference between TK and TR also decreases. Difference between TK and TR decreases. The Controller then lowers the speed, which reduces the residence time of the time of the medium in the Collector and thus increases the difference TK - TR.

### Which version of ESPHome has the issue?

v2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.2.5

### What platform are you using?

ESP8266

### Board

WeMosD1 R2

### Component causing the issue

Climate Component - PID Climate

### Example YAML snippet

```yaml
....
captive_portal:
############################################
substitutions:
  devicename: solartec
  default_target_temp: '22 °C'
### Steuerung PWM Ausgang D5    
output:
  - platform: esp8266_pwm
    pin: D5
    frequency: 500Hz
    id: ventcontrol
    inverted: True
#### PID-Regelung Solar-S1 auf RL-S2
climate:
  - platform: pid
    visual:
      min_temperature: 5 °C
      max_temperature: 95 °C
      temperature_step: 1 °C 
    name: "PID Solar Controller"
    id: pid_solar
    sensor: sols1
    default_target_temperature: ${default_target_temp}
    heat_output: ventcontrol
    control_parameters:
      kp: 0.49460
      ki: 0.00487
      kd: 12.56301
      output_averaging_samples: 5      
      derivative_averaging_samples: 5  
    deadband_parameters:
      threshold_high: 0.5°C     
      threshold_low: -0.5°C
sensor:
### Sensors to Read PID information into Home Assistant ###
  - platform: homeassistant
    name: "Solar S1"
    entity_id: sensor.kollekt
    id: sols1
```

ssieb, 2023-02-26:
I don't really understand your description but there's no bug here.  You're trying to set the **default** temperature which is only useful on the initial flash.  It's not something you would keep changing.  The substitution should work, but just use a number.  If you need more help, then ask on the discord server or one of the forums.

closed 2023-02-26 by ssieb

judie07, 2023-02-27:
Hello Samuel,

in my opinion it is a point that should be discussed here! A change to the parameter setting, where else?
The statement of you is not correct - there are very well use cases in which it is **useful** to automatically track the setpoint temperature. I have already recorded such a case. Again, it is about the constant temperature between two sensors with a continuously changing process value (solar collector).
It is important to understand how a differential control works and if it does not work or is not wanted, then it is just as but do not write it is not **useful!**

ssieb, 2023-02-27:
I said it's not useful to change the **default** temperature.  If you want to discuss how to do a differential control (whatever that is), then there are places to ask that.  It is **not** an esphome bug, so this is not an appropriate place.

judie07, 2023-02-27:
**ESPHOME "Extension/improvement/addition".**
Okay, I didn't see it that way. Nevertheless the request to check if the "default_target_temperature" value can be changed via a template.

ssieb, 2023-02-27:
You've still missed the point about **default**.  The answer is no and that can't be changed because it doesn't even make sense.