esphome/issues #4251: MAX98357 Not Working With Touch Sensor

brooksben11, 2023-02-26:
### The problem

Multiple folks are reporting issues using the MAX98357 audio board when a touch sensor is also configured. Media player works fine for folks as does including the "esp32_touch:" component. However, as soon as an actual touch sensor is specified, the audio becomes choppy and slow/delayed. Multiple touch pins have been tried with no success.

https://community.home-assistant.io/t/max98357-does-not-work-with-a-touch-sensor/536928

https://community.home-assistant.io/t/mediaplayer-and-esp32-touch-conflict/539997

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

MAX98357, i2s_audio, Touch Sensor

disconn3ct, 2023-03-20:
I found a workaround. Lowering `measurement_duration` to 2ms seems to work for audio (including streaming radio for a few minutes) and still respond reasonably well to touch after resetting the thresholds.

Reproducing requires an enabled touch pin, not just the hub, and doesn't seem to care about `sleep_duration`. It also doesn't care about actual touches, so it should be possible to reproduce without opening existing devices.

disconn3ct, 2023-03-20:
Related to #3060 (the stale bot blindly closed)

brooksben11, 2023-03-24:
@disconn3ct Just tried adding the touch sensor back to my project using a measurement_duration of 0.5ms. Still unusable for me playing a local mp3 file, although it is LESS choppy and slowed down than using the default. Bumping it back to the 2ms you originally mentioned definitely makes it worse. Not sure why radio streaming is working for you.

I also noticed this in my logs (don't think I looked before), but when the touch sensor is included I get this repeating continuously:

`[14:20:46][D][api.connection:920]: Home Assistant 2023.3.6 (192.168.2.84): Connected successfully
[14:20:46][D][api:102]: Accepted 192.168.2.84
[14:20:46][W][api.connection:083]: Home Assistant 2023.3.6 (192.168.2.84): Connection closed
[14:20:46][D][api.connection:920]: Home Assistant 2023.3.6 (192.168.2.84): Connected successfully
[14:20:47][D][api:102]: Accepted 192.168.2.84
[14:20:47][W][api.connection:083]: Home Assistant 2023.3.6 (192.168.2.84): Connection closed` 

Whereas it's just fine when the touch sensor is NOT included.

disconn3ct, 2023-06-23:
It stopped working overall and I had to disable the touch sensor.